ma1co/OpenMemories-Tweak #114: Request : enable longer exposure times than 30s in non bulb mode

benolry, 2016-09-21:
Is there a way to enable longer exposures than 30s in non bulb mode? Can the behaviour of when the camera saves full 14bit raw be influenced in addition? Thanks in advance,

Ben

kaidaning, 2016-09-26:
+1
It's found that the Bulb mode will cause 'star eater problem'. If  longer exposures than 30s in M mode can be enabled, it is a good news for astro-photographer. 
Thank you!

ScorpionCarcass, 2017-04-28:
+1. Just a note with 3.3 firmware they eat stars after 3.2 seconds. Even taking pictures of the city at night long than 3.2 seconds will eat the detail of the lights in the city. See here for examples:

http://community.sony.com/t5/Alpha-SLT-DSLR-Cameras/star-eater-algorithm-in-Bulb-mode-shots-please-offer-alternative/td-p/619350

jkbsahner, 2025-02-12:
Hey Guys, do you know if there has been any progress in finding out how to fix the star eater issues via firmware?