ma1co/OpenMemories-Tweak #119: Gui Problem on HX400V

hasi699, 2016-10-13:
I successfully installed the App, but I can only access the info tab. Its not possible to access all the other tabs using the arrow keys on the camera.

ma1co, 2016-11-04:
Can you try the [PMCADemo](https://github.com/ma1co/PMCADemo) app, especially the "Key events" option? What does it show when you press the arrow keys?

allexoll, 2016-11-04:
to solve this I donwloaded an older version, when there wasnt the Info tab, so I didnt need to press left/right, only up/down, and it worked!

hasi65, 2016-11-04:
Hi, i tried the PMCADemo app and the "Key events" option shows correctly all pressed keys including the arrow keys.

hasi65, 2016-11-04:
I reinstalled the app and now it works without problems. One has to press the UP key first and then all keys are working. You can close this issue. Thank you very much.

closed 2017-10-30 by ma1co