ma1co/OpenMemories-Tweak #130: RX100m5 4K 5min limit

avtoportret, 2016-11-10:
Can you please remove 5min limit for 4K recording in RX100m5?

neoshik, 2016-11-30:
yes, 4K 5 min limit is not available on rx100m5
https://github.com/ma1co/OpenMemories-Tweak/issues/127#issuecomment-263748846

peppersass, 2016-12-19:
There's also a 5-minute limit on 120p in both 60M and 100M in XAVC S file format. Would be great to remove that, too. Might be the same flag (or not.)

xezon, 2017-02-12:
I am owner of this camera and would like to add that feature.

ma1co, can you please give me quick guide of what steps are required to dump the memory and how to test a new custom build on the camera? I could try to find the new 4k limit address and add it.

bpasu, 2017-02-21:
I dumped the memory but cannot find where the flag is. It would be nice if  @lainy can help us pin point what he did for RX100M4.

xezon, 2017-02-21:
Hi bro. You can find the commit in the history:
https://github.com/ma1co/OpenMemories-Tweak/commit/1a96c703f5e8da63fd32beb68f64f5012c9778e3

This appears to be the address to a 2 byte value:
public static final int REC_LIMIT_4K = 0x003c04b6;

And the expected default value is 300 in seconds = 5 minutes
public static final int DEFAULT_LIMIT = 300;

bpasu, 2017-02-21:
@xezon I saw that but the Backup.bin filesize is just 754,462 byte. I'm a bit confused.

xezon, 2017-02-21:
@bpasu Is it possible that you just dumped the code page but not the heap? Considering the addresses for the 30 minute recording limit work on this camera, the 4k limit address should be around that address location I assume.

peppersass, 2017-02-21:
The property addresses don't refer to locations in the Backup.bin file. You have to copy the Backup.bin file to the SD card, transfer it to a PC, and use the print_backup command in fwtool.exe to map Backup.bin to a text list of property addresses. This will allow you to inspect properties near 0x003x04b6 for one containing the default value of 300. Once you know that, you should be able to use the bk.elf command in the camera android shell to patch the value to something larger until ma1co updates the OpenMemoriesTweak app.

**Question**: Did you dump memory to the SD card? If so, how did you do it? When I try to copy Backup.bin to the SD card I get an error that it's a read-only file system. Can't use chmod to change permissions on the SD card or any of its parent directories, either, though it does work for changing permissions on the /setting/Backup.bin and /setting/Backup.bak files. This is not the case on my  RX100M3. I can copy files to the SD card on that camera without changing permissions.

Is there a command or sequence of commands to make the SD card rw or to copy files to it?

lainy, 2017-02-21:
Back when I did the rx100m4 4k limit thing, I obtained backup.bin directly from the firmware update images using the fwtool to unpack them. As @xezon and @peppersass point out, you just need to find the one with a default value of 300.

In my case, I seem to recall I narrowed it down by doing a sort-of diff between the a7R2 and rx100m4 backup.bin (from unpacked firmware update) to rule out any common values first, since the a7R2 does _not_ have a separate 4k time limit. The output was just a CSV with columns `Address`, `a7r2 value`, `rx100m4 value`.

BTW you can also extract some of the names of backup.bin values from libObj.so in the firmware image. For example, sub_645740 in libObj.so seems to list a lot of them:
```asm
.text:00645740 sub_645740
.text:00645740
.text:00645740 var_8       = -8
.text:00645740
.text:00645740    PUSH     {R4,R5,R7,LR}
.text:00645742    SUB      SP, SP, #8
.text:00645744    MOV      R4, R0
.text:00645746    ADD      R7, SP, #8
.text:00645748    MOVS     R0, #0x30 ; unsigned int
.text:0064574A    BLX.W    operator new(uint)
.text:0064574E    MOVS     R3, #0
.text:00645750    STR      R3, [SP,#8+var_8]
.text:00645752    MOVS     R2, #5
.text:00645754    LDR      R3, =(aInvalidPara_92+0x1F - 0x64575C)
.text:00645756    LDR      R1, =0x1070015
.text:00645758    ADD      R3, PC  ; "STILL_FORMAT_SIZE"
.text:0064575A    MOV      R5, R0
.text:0064575C    BL       sub_425048
.text:00645760    STR.W    R5, [R4,#0x308]
.text:00645764    MOVS     R0, #0x30 ; unsigned int
.text:00645766    BLX.W    operator new(uint)
.text:0064576A    MOVS     R3, #2
.text:0064576C    STR      R3, [SP,#8+var_8]
.text:0064576E    MOVS     R2, #4
.text:00645770    LDR      R3, =(aInvalidPara_93+0x1F - 0x645778)
.text:00645772    LDR      R1, =0x1070012
.text:00645774    ADD      R3, PC  ; "STILL_FORMAT_ASPECT"
.text:00645776    MOV      R5, R0
...
```
I think this list was lazily extracted from the rx100m4 libObj.so, firmware V1.30:
```
01070015 STILL_FORMAT_SIZE
01070012 STILL_FORMAT_ASPECT
01070013 STILL_FORMAT_FILE_TYPE
01070014 STILL_FORMAT_QUALITY
010713F0 STILL_FORMAT_SIZE_DUALREC
010713EF STILL_FORMAT_QUALITY_DUALREC
01070007 AVIEW_TIME
01070052 RED_EYE_REDUCTION
010700B6 ANTI_BLINK_MODE
010700B7 BLINK_ALERT_MODE
01070103 DRIVE_MODE
01070101 DRIVE_BURST_SPEED
01070100 DRIVE_BURST_MAX
01070102 DRIVE_CAPTURE_STOP_ENABLE
0107145F SELF_TIMER_MODE_OFF
010708FF SELF_TIMER_INTERVAL
010700FD BRACKET_MODE
01070141 BRACKET_STEP
010700FF BRACKET_OTHER_STEP
010700FE BRACKET_ORDER
0107145C BRACKET_SELF_TIMER_MODE_OFF
01071466 BRACKET_SELF_TIMER_INTERVAL
0107010B PANORAMA_MODE
0107010C PANORAMA_SIZE
0107010A PANORAMA_DIRECTION
0107005C AF_LIGHT_MODE
01070150 STILL_COLOR_SPACE
01070111 SMILE_SHUTTER_MODE
01070110 SMILE_DETECTION_SENSITIVITY
-------- PSEUDO_REC_MODE
01070112 SOFT_SKIN_EFFECT_LEVEL
-------- HOTSHOE_STROBE
-------- HOTSHOE_LANC_STROBE
-------- CA_EXT_STROBE
-------- CA_EXT_STROBE_MODE
01070108 NR_LONG_EXP
01070107 NR_HIGH_ISO
01070113 STEADYSHOT_STILL
010713D5 STEADYSHOT_STILL_INFO
010713D4 STEADYSHOT_STILL_BODY_AXIS
010713D6 STEADYSHOT_STILL_LENS_FOCAL_LENGTH
01070104 DRO_LEVEL
01070105 HDR_LEVEL
010700C0 BACKGROUND_DEFOCUS_LEVEL
01070152 STROBE_EV_SHIFT
01070154 STROBE_EV_STEP
01070156 STROBE_MODE
-------- ONESHOT_3D_MODE
010701BC STILL_DATE_ADD
-------- STROBE_POPUP
-------- STROBE_CHARGE_ENABLE
-------- AUTOSHUTTER_MODE
01070040 STILL_DATE_TIME_STAMP_STYLE
-------- STILL_AUTO_DELETE_MODE
0107031B PANTILT_LIMIT
01070319 PANTILT_FREQ
0107031A PANTILT_ORDER
01070318 PANTILT_AUTOREVIEW
01070317 PANTILT_INTERACTIVE
-------- PANTILT_DEMO
-------- SELFPORTRAIT_NUMBER
-------- SELFPORTRAIT_TIMER
-------- SELFPORTRAIT_INTERVAL
-------- LENS_COVER_INFO
-------- STROBE_CONTROL_MODE
010708DF EV_SHIFT_AFFECT
-------- STROBE_E_MNT
01070664 FOCUS_RELEASE_PRIORITY
01070771 S1_AF
0107086E FRONT_E_SHUTTER_MODE
-------- MANUAL_STROBE_LEVEL
-------- FE_LOCK
01070A71 TOUCH
01070AED GAIN_UP_LONGFLASH
01070AF1 STILL_EE_FRAME_RATE
01070B2C SILENT_SHUTTER
01070F0D MOTION_SHOT
010713E3 COPYRIGHT_INFO_MODE
010713E4 COPYRIGHT_NAME
010713E2 ARTIST_NAME
01071456 FOCUS_RELEASE_PRIORITY_AF_C
```
I don't remember how I got these values though. The pattern was really consistent so it might have just been a regex over the disassembled code 😋

peppersass, 2017-02-22:
While waiting for an answer to my question about the "read-only file system" problem with the SD card, I found that Sony recently updated the RX100M5 firmware to v1.01. I downloaded and upacked that version, then ran one of the backup.bin files through the fwtool print_backup command. Then I searched for property 0x003c4b6. That location contains the value 0C.

I figure the property is a 2-byte field loaded with 12 0C (assuming 300 = 0x12C and low-hi format), so I don't think property 0x003c4b6 contains the 4K time limit in the RX100M5.

(Note: I guess the value could be stored in hi-low format, but there are no occurences of "0C 12".)

So I searched for "12 0C". There are 27 properties with that value, with 6 of them consisting of four bytes and 21 consisting of two bytes. 17 of the 2-byte properties with that value are in groups with the value 12 0C in several consecutive locations: two groups of six properties, one group of three properties and one group of two properties. That leaves four properties with "0C 12" that's not in consecutive locations.

Note that there could be multiple instances of the 300-second time limit. For example, it applies to all four 4K variations (30p @100M, 30p @ 60M, 24p @ 100M, and 24p @ 60M), as well as two of the HD formats (120p @100M and 120p @60). There could be a single limit property for all, or it could be two different limit properties or six different limit properties.

I guess I could go through the properties with bk.elf and try setting them to a higher value one at a time, but it would be better if someone could dump the full property list.

xezon, 2017-02-22:
For testing I suggest to set them to a lower limit. Is faster to test.
Maybe the 4K time limit was changed to a H:M:S format like the 30 minute limit.

peppersass, 2017-02-22:
Oh! I didn't know the 30-minute limit was in H:M:S format. Do you know the property address for that one? I'd like to see how they formatted it.

xezon, 2017-02-22:
https://github.com/ma1co/OpenMemories-Tweak/blob/master/app/src/main/java/com/github/ma1co/openmemories/tweak/BackupKeys.java

public static final BackupProperty.Byte REC_LIMIT_H = new BackupProperty.Byte(0x003c0373);
public static final BackupProperty.Byte REC_LIMIT_M = new BackupProperty.Byte(0x003c0374);
public static final BackupProperty.Byte REC_LIMIT_S = new BackupProperty.Byte(0x003c0375);

peppersass, 2017-02-22:
Thanks. Assuming a property sequence of 00:05:00, there are a number of candidate locations. I didn't count them, but it's probably on the same order as the number of locations containing 12 0c.

However, I noticed that the 30-minute timer is actually set to 00:29:50. So if the 4K limit has been changed to H:M:S, they might have done something sneaky like 00:04:59. That said, the 4K timer on the camera always expires at 5 minutes on the dot. But it's also possible that there's a lag before the camera actually shuts off the video.

Anyway, it'll take a lot of trial and error to check all the locations for both formats. It would be so much better if we could get the property location from the firmware. **lainy**, how did you "lazily" extract the property list from libObj.so?  Why isn't it complete?

I don't know the machine language shown in the code segment, so I can't tell what it's doing.

lainy, 2017-02-23:
@peppersass from the snippet, the code refers to a backup ID and a name:
```asm
.text:00645756    LDR      R1, =0x1070015
.text:00645758    ADD      R3, PC  ; "STILL_FORMAT_SIZE"
```
```asm
.text:00645772    LDR      R1, =0x1070012
.text:00645774    ADD      R3, PC  ; "STILL_FORMAT_ASPECT"
```
The pattern is `LDR R1, =BACKUPID` followed by `ADD R3, PC ; "BACKUPID_NAME"`
So I just dumped the disassembly and did a search & replace in a text editor.

It isn't complete because the subroutine I was looking at only had the names of certain backup IDs, and some of the IDs didn't fit the above pattern so they didn't show up after my lazy search & replace. I don't remember if I couldn't find the missing backup IDs in the code or if I ran out of time and didn't look any further, though.

peppersass, 2017-02-23:
How did you generate the disassembly of libObj.so?

Not knowing how to do that, I loaded libObj.so into a text editor and searched for REC_LIMIT. Not there. In fact, I did a grep-like text search on the entire unpacked firmware directory and only came up with one file that has the string REC_LIMIT. It's libmpr.so, the only other very large file in the directory with libObj.so. There's one line in it with REC_LIMIT:

REC_LIMIT_TIME formatType = REC_FORMAT_TYPE_SD_MPEG2   REC_LIMIT_TIME formatType = REC_FORMAT_TYPE_HD_AVC   REC_LIMIT_TIME formatType = REC_FORMAT_TYPE_MP4_AVC   REC_LIMIT_TIME formatType = REC_FORMAT_TYPE_AVI_DV   REC_LIMIT_TIME formatType = REC_FORMAT_TYPE_XAVC_AVC   REC_LIMIT_TIME formatType = REC_FORMAT_TYPE_WAVE_LPCM   REC_LIMIT_TIME formatType = NON

Without seeing the code, hard to say what's happening here. But it looks like there are separate recording limit times for each format. But there isn't a separate reference to XAVC-S or XAVC-S 4K format. Without the code I can't find the associated property addresses. Need to disassemble this file. Can you tell me how?

Note that I did the search on both the RX100M4 and RX100M5 unpacked firmware, with the same results. The above line is the same in both versions of the firmware and there are no other references to REC_LIMIT. The property address may have changed but I think everything else is the same as far as recording limits go.

BTW, initially I looked for REC_LIMIT_H for the HD recording time limit, per ma1co's code extract shown above, and found no occurrences. But when I checked the code I found that these are variables into which the H:M:S are read from a single property that has the time in seconds. As such, I believe they're ma1co's variable names and are not actual property names. He has the property addresses hard coded, so he doesn't need the names.

xezon, 2017-02-24:
Can someone please copy & paste the memory footprint around the suspected location?

lainy, 2017-02-24:
@peppersass I used [IDA Pro](https://www.hex-rays.com/products/ida/) to disassemble libObj.so, but there are free disassemblers available also...

peppersass, 2017-02-24:
> @peppersass I used IDA Pro to disassemble libObj.so, but there are free disassemblers available also...

@lainy What kind of CPU is it? (I assume the disassembler needs to know that...)

[Sorry for being such a doofus hacker. Have done a lot of programming in assembler, including multitasking, but that was 45 years ago on mainframes! Did a small PIC project about 15 years. :-) ]

lainy, 2017-02-24:
@peppersass Haha, no problem! It's an ARM CPU, I don't remember specifically but any ARM-aware disassembler will probably do the trick.

peppersass, 2017-02-24:
> Can someone please copy & paste the memory footprint around the suspected location?

@xezon If you mean from a memory dump, as I said earlier I can't write anything to the SD card on the RX100M5 (works fine on my RX100M3.) Keeps telling me its a read-only files system. Have tried to change permissions on directories, but get the same error. But I was able to do a chmod on Backup.bin and Backup.bak. Not clear why. Would really like to find a fix for this. Has Sony figured out that people are hacking the firmware?

bpasu, 2017-02-24:
I dump out libObj.so from my RX100V might be helpful for reference in the future. http://pastebin.com/3jA5H8Vz

peppersass, 2017-02-24:
@bpasu - Is this from the unpacked firmware file or did you get it from the camera? If the camera, how did you copy it to the SD card? When I try to do that I get "Read-only file system."

bpasu, 2017-02-25:
@peppersass I got it from the camera. Did what in the readme.md

dd if=/dev/nflasha of=/android/storage/sdcard0/DUMP.DAT bs=1M

Extracted it using fwtool and disassembly. It is kind of out of scope here. I cannot find any register helpful for hacking 4K limit though.

peppersass, 2017-02-25:
I've done some hunting in the firmware with the demo version of IDA. I'd like to copy some code excerpts here, but I don't know how to format them to look nice, like @lainy's above. Can someone tell me how to do that?

Here's what I've found so far:

1. I used the freeware utility StringFinder to scan the unpacked RX100M4 and RX100M5 firmware files for occurrences of the time limit properties 0x3c0373 and 0x3c04b6 referenced in OpenMemoriesTweak (specifying the property keys in little endian format: 73 03 3c 00 for 0x003c0373.)

2. The code that handles the general time recording time limit is in libmpr.so. The routine picks up the H:M:S values from properties 0x003c0373, 0x003c0374, and 0x003c0375. The default value is 00 1d 32, or 29 minutes and 32 seconds. The code is the same for the RX100M4 and RX100M5, except for code location and address references.

3. The code that handles the 4K limit property 0x03c04b6 referenced by OpenMemoriesTweak is also in libmprctrl.so, in a routine called GetRecLimitTimeForFV(void) -- but only in the RX100M4 firmware. I can't tell exactly what's going on, but the routine appears to call routines to get time limits for three formats: Getreclimitt_2, Getreclimitt_3, and Getreclimitt_4. These correspond with time limit variables defined in libmpr.so:

| libmprctrl.so | libmpr.so | Format Type |
| -------------- | :----------|:---------------|
| Getreclimitta | aRec_limit_time | SD_MPEG2 |
| Getreclimitt_0 | aRec_limit_ti_0 | HD_AVC
| Getreclimitt_1 | aRec_limit_ti_1 | AVI_DV
| Getreclimitt_2 | aRec_limit_ti_2 | XAVC_AVC
| Getreclimitt_3 | aRec_limit_ti_3 | WAVE_LPCM
| Getreclimitt_4.| aRec_limit_ti_4 | NON

There doesn't appear to be a time limit definition for the XAVC S 4K formats, though they might use the XAVC_AVC definition, with all the other HD format time limits being defined as HD AVC, even if they use XAVC-S format. There are also time limit check calls in libmpr.so that differ somewhat from the above:

| libmpr.so | Format Type |
| :-------------- | :----------|
| aRec_limit_chk_ | HD |
| aRec_limit_ch_0 | SD |
| aRec_limit_ch_1 | MP4AVC |
| aRec_limit_ch_2 | NONE |

4. As implied in #3, the GetRecLimitTimeForFV(void) is missing from libmprctrl.so in the RX100M5 firmware, which explains why the property can't be set by OpenMemoriesTweak. However, the RX100M5 includes a new routine called GetHeatLimitedMovieRecTime(bool *, unsigned int *). There are two values that look like they might be property keys (0x9245DC and 0x920111), but I can't find them in Backup.bin.

5. Both the RX100M4 and RX100M5 also have a routine in libmprctrl.so called GetRecLimitTime(void). It looks like the code is the same, except for code location and address references. This code loads from what looks like property key 0x3C03A5, which is a real property, but I don't see values for that property and the surrounding properties that look like the default time limits or close to them. Here are the some of the values in the vicinity:

```
id=0x003c03a5, size=0x0001, attr=0x00:
  03                                               .

id=0x003c03a6, size=0x0002, attr=0x00:
  d0 07                                            ..

id=0x003c03a7, size=0x0002, attr=0x00:
  dc 05                                            ..

id=0x003c03a8, size=0x0002, attr=0x00:
  e8 03                                            ..

id=0x003c03a9, size=0x0002, attr=0x00:
  f4 01                                            ..

id=0x003c03aa, size=0x0002, attr=0x00:
  b8 0b                                            ..

id=0x003c03ab, size=0x0002, attr=0x00:
  70 17                                            p.

id=0x003c03ac, size=0x0002, attr=0x00:
  28 23                                            (#

id=0x003c03ad, size=0x0002, attr=0x00:
  b8 0b                                            ..

id=0x003c03ae, size=0x0002, attr=0x00:
  70 17                                            p.

id=0x003c03af, size=0x0002, attr=0x00:
  28 23                                            (#

id=0x003c03b0, size=0x0002, attr=0x00:
  d0 07                                            ..

id=0x003c03b1, size=0x0002, attr=0x00:
  dc 05                                            ..
```
This routine appears to call routines to get time limits for three formats: Getreclimitt, Getreclimitt_0, and Getreclimitt_1. These correspond with time limit variables defined in libmpr.so, as shown above.

I can paste in the code if someone will tell me how to make it look nice.

baybora2017, 2017-03-27:
Any one figured it out yet ! Will be nice to record over 5min

zippytiff, 2017-09-01:
Hi Guys,

Is this a dead end ? or is there hope to a solution

If anybody wants me to do anything, please ask

ZT

peppersass, 2017-09-01:
I spent a lot of time trying to find the setting or settings that would override the limit, but failed. In the end I bought a used Atomos Ninja Flame external display/recorder. No recording limit and codecs for higher-res 4K compression like ProRes and DNxHR are supported, not to mention a much larger screen for recording and playback.

zippytiff, 2017-09-02:
That is a real shame :-( perhaps an app could be written which just records 4k vid after 4k vid ? then it just leaves the heating fix (easy enough I recon)

AshBS, 2018-02-25:
Has anyone else had another crack at this? 5 min limit is killing me!!

coolbutpointless, 2018-05-14:
I wish I understood how to do this stuff because this would be a fantastic feature to unlock.

Faeb35, 2019-01-14:
Would be excellent if someone would get out how to overwrite this limit!
Thanks anyway for creating such a great software!