ma1co/OpenMemories-Tweak #131: (QUESTION) SHOOTING UNLIMITED 4K int&EXT WITH A7SII?

Elsibki, 2016-11-15:
Hi,

I'm working on some pre-production for a feature film. We are thinking about using your amazing crack to improve our 3 sony A7Sii.

Still, I would like to know (it's more a confirmation) if the crack allows you to record unlimitedly in 4K XAVCS internal and 4K on external recorder. 

Anyway, thanks for this Tweak, it opens a promising way for this camera. (4:2:2 4K int would simply make this camera go into a whole new level)

Cheers,

Sam.

tatsin, 2016-11-16:
sorry for interrupting sam :) I have a question about the external power way ? can you show me any alternative ? thank you much ( I have a6000)

gavroche37, 2016-12-21:
Hi!

Unlimited (or <2 hours) internal 4k recording for A7sii would be much appreciated. Love your app!

gavroche37, 2016-12-21:
Hi!
 
Unlimited (or >2 hours) internal 4k recording for A7sii would be much appreciated. Love your app!

kmoreau893, 2016-12-22:
I believe that this already works. I have used it on a A7sii for 4K with external dummy battery and it works. I used a 128GB card and an Atomos Power Station Video to power it.

sleepinghouse, 2016-12-30:
The checkbox for Unlimited 4K recording is greyed out. 
Is there a workaround for this yet?

ma1co, 2018-10-26:
The 4K checkbox is only for the RX100M4. Disabling the standard recording limit on the A7S2 should also disable the 4K limit.

closed 2018-10-26 by ma1co