ma1co/OpenMemories-Tweak #133: [request] Temporarilly change Sony HX60 model to HX60V to access PMCA store?

dany123, 2016-11-24:
Hy,

I recently bought a Sony DSC-HX60 camera but sadly sony forgot to add this specific model to any app's compatibility list even though the DSC-HX60V model has 21 available apps and it has the exact same firmware and specs except GPS module. The camera even comes with the Smart Remote control app by default but I can't even update it to the latest version due to "Camera not supported" error. 

Note the remote control app Ver. 3.10 even mentions the DSC-HX60 model in the changelog so it is 100% supported just some dev forgot to add this model to the compatibility list (or it was intentional to force to buy the more expensive model?): https://www.playmemoriescameraapps.com/portal/usbdetail.php?eid=IS9104-NPIA09014_00-F00002 (click on "more" below the Update area) and they won't bother adding it as their official response (on the sony forums) is they simply don't support that model.

Even though I haven't seen this model  (DSC-HX60) being mentioned around here I can say that it is compatible with the PMCA-RE hack and I could install all the apps in this store and they actually work (just tested today the 30+ min unlock, the PCMCA Demo and Better Manual).

In **conclusion** I'm asking if it's possible to change the model that the camera sends to the official store when checking for apps (I'm hoping it only sends the model string, not the serial number too) so I (we) can access the apps meant for other models (at our own risk of course).

Also, alternatively, if someone could share the latest version of the Smart Remote Control apk (and/or the apk's for the FREE apps - i've seen people asking for the paid ones and get no replies but this is not the case) I would be really really happy.

Thanks,
Daniel

dany123, 2016-12-04:
Quick update: the Sony DSC-HX60 model is supported in European regions so i created a new account and selected region as UK/English and was able to install everything I needed. 

Why these differences in sony's website? No idea... I even emailed support about this and they said the model are identical minus the gps but the model is not supported... lol.

closed 2018-10-26 by ma1co

isopix, 2022-07-24:
I wonder if by changing some string we could unlock GPS in DSC-HX60(convert it to HX60V ^^) or maybe hardware is lacking some chip and it's impossible?

I have only DSCH60 (without GPS) to test.