ma1co/OpenMemories-Tweak #134: [Feature] - a6000: Enable video recording for the android / iOS remote app

syntaxer, 2016-11-26:
The video recording is available for a5100 in the remote app for android / iOS.
I don't know why they disabled it for a6000.
Is it possible to enable it, or at least to be able to press the record button on the camera and preview the screen with an android / ios device?
Currently if the remote app is started and you press the record button, it displays that the feature is disabled.
Thanks in advance!

LXR85, 2016-11-28:
+1 for Sony A7, please.

cfanyyx, 2017-02-02:
+100000000

Don-Agus, 2017-02-23:
also if posible would be great on a6000 or a7r

hugomfranca, 2017-02-24:
This should be pretty easy for programmers, since it's just another feature that's already there but locked. I  believe most people around here with an a6000 would gladly donate money for someone to make this happen. I would!

mungewell, 2017-10-23:
might be related to issue 219 request

The 'Remote API' has concept of 'getSupported...()' and 'getAvailable...()' functions, meaning that some settings change dependent on current state of camera. I believe that on the HX60 (for example) movie recording is only possible in some dial positions....
https://github.com/Bloodevil/sony_camera_api/blob/master/doc/hx60_movie.txt#L48

Regarding LiveView; above project implements Remote API in python running on PC. Might prove that the camera is capable of doing what you are asking of the Android app.

iBierkasten, 2018-02-19:
Would love to have this Feature.
I think everyone who is into Moviemaking with the a6000 need this Feature. It must be possible, as the a5100 is the same Camera.
I dont understand Why Sony doesnt allow live view on the a6000...

mungewell, 2018-02-20:
It is the code of the "Smart Remote" (on camera application installed from PlayMemories store) which prevents remote video, see:
https://github.com/ma1co/Sony-PMCA-RE/issues/67

On my A5000, changing the model name (to 'A5001') made it possible to use remote video recording. I have no idea why Sony chose to _actively_ block this, but it appears to function OK.

Technically "Live View" is the ability to see the view finder in photo or video mode, video recording is just a plus.

pynkstyle, 2018-02-20:
Could this be done with a6000, what we need to change the camera name to 6001 or this could be done somehow with open memories tweak?

iBierkasten, 2018-02-20:
Could you give us a quick step by step Guide how to Change the Model Name so we could give it a try on the a6000?

mungewell, 2018-02-21:
I thought the link was pretty descriptive.... The model name is stores within the Key "0x003e0005" (at least on the a5000).

I used Tweak to connect to my WiFi and start it's telnet server. The from my PC I telneted to the camera and used 'bk.elf' to change the contents of this key. I don't recall whether 'bk.elf' is pre-installed in the 'runtime' image, or whether I copied if from the 'loader' image.

There are other ways to adjust keys, you can also use PCMA-Console.py  ("bk r" and "bk w"), see here:
https://github.com/ma1co/Sony-PMCA-RE/commit/5f35301afade1ab17b512c1d1af22bb22639bcde
https://github.com/ma1co/Sony-PMCA-RE/issues/58

pynkstyle, 2018-02-21:
Thanks **mungewell** it is working also on a6000, changed to a5001 and now I have live view  and can record from phone, tablet and RM-LVR1
I will chane it to 6001 to look better, but later.
Thanks again for info

iBierkasten, 2018-02-21:
thanks @mungewell! I alsop changed the camera name to 5001, but i just have live view on phone. I still cant record. I do have still the Error Text on my cam which says that i cant use der playmemories app in this mode (movie mode). Whenever i trie to start recording from phone or from camera button, nothing happens.

syntaxer, 2018-02-21:
Can a feature be added to the OpenMemories-Tweak app to change the camera name?

mungewell, 2018-02-23:
@iBierkasten Checking that you downloaded a recent version of "SmartRemote" from the PlayMemories store, the pre-installed version (in Camera f/w) is pretty crappy.

iBierkasten, 2018-02-23:
@mungewell thank you so much! its finally working!
I was about to buy the a5100 for my gimbal because of this feature!
Thank you so much, you saved me a lot of money!

Muuuuuuuux, 2018-03-13:
I‘m so sorry for changing the language now, but this topic seems to be too difficult, for me and my bad english skills. But I need this function too.

@iBierkasten kannst du mir etwas detaillierter erklären wie das ganze funktioniert? Mit welchem Programm sendest du den Code aus dem anderen Post an die Kamera? Oder hab ich da was total falsch verstanden?

Danke, liebe Grüße!

iBierkasten, 2018-03-14:
Hey,
Ist eigentlich ganz simpel. Du lädst dir den Open tweaks jailbreak auf die Kamera, Beschreibung dazu findest du zb bei Chip, einfach bei Google eingeben.
Dann aktivierst du den Developer Mode, das wlan und die Telnet Funktion in diesem Programm auf der Kamera.
Dann musst du jenachdem unter Windows 10 Telnet noch aktivieren, wie das geht findet man auch bei Google. Ist nur Häkchen setzen.
Dann drückst du win Taste + r und gibst ein cmd, da dann einfach telnet und dahinter die ip der Kamera eingeben, schon bist du verbunden. Dann den Code aus dem andern Thread. Wobei der mit dem r für read, also auslesen steht und mit dem mit w also Write überschreibst du den. Danach zur Kontrolle noch mal den Code auslesen und dann mit Exit einfach Verbindung beenden.
WICHTIG ist dass du vorher die Smart Remote App updatest, musst du machen weil mit der vorinstallierten geht es nicht, die heißt Smart Remote eingebettet, du brauchst aber nur Smart Remote. Das musst du unbedingt vorher machen da du es nicht mehr laden kannst weil die Seite denkt es wäre für die Kamera nicht kompatibel, hast ja das Modell geändert.

Lg

Muuuuuuuux, 2018-03-14:
Du bist der Hammer @iBierkasten !!
Ich hab alles genauso gemacht wie beschrieben und die Verbindung zwischen Computer und Kamera funktioniert auch und ich kann auch den Datensatz der Kamera lesen, doch beim Schreiben bekomme ich immer einen „Write error“.. weißt du, was ich da machen kann?

Muuuuuuuux, 2018-03-14:
Okay, ich revidiere, musste bei unlock protection settings ein Haken setzen... und es funktioniert :))))
Gibt es noch eine Möglichkeit, dass die Videos auf dem Handy gespeichert werden? Bei den Bildern passiert das ja automatisch

syntaxer, 2018-03-27:
@mungewell Thanks it works just fine on a6000
I modified the hex to change the name to a6001:
`bk.elf w 0x003e0005 49 4C 43 45 2D 36 30 30 31`

voxivoid, 2018-04-29:
In case someone wants a full guide on how to change the model of Sony A6000 (based on @syntaxer's post):

1.  Install the OpenMemories: Tweak app on your camera and go to the Protection tab and select Unlock protected settings.
2. Run the pmca-console through the command line / terminal with the flag updatershell (i.e. `pmca-console-v0.17-win.exe updatershell`).
3. Then change the model name from ILCE-6000 to ILCE-6001 with the following command: `bk w 0x003e0005 49 4c 43 45 2d 36 30 30 31 00 00 00 00 00 00 00`.

If you want to reset the model name to ILCE-6000 just write `bk w 0x003e0005 49 4c 43 45 2d 36 30 30 30 00 00 00 00 00 00 00`.

dodolgarut1, 2018-05-02:
@voxivoid i've done it but when i tried to change the model back pmca says that it's an unknown device, what can i do now?

ma1co, 2018-05-02:
@dodolgarut1 Try `pmca-console updatershell -m ILCE-6000`

dodolgarut1, 2018-05-03:
@ma1co thanks man! although i ended up telnetting through my camera and use **bk.elf w 0x003e0005 49 4c 43 45 2d 36 30 30 30 00 00 00 00 00 00 00** thank you so much

rizkisap, 2018-07-09:
I've tried steps above

I have successfully installed openmemories tweak and enable all its features.
but I do not understand the next step. how to connect camera with telnet and how to use cmd.

Please help

rizkisap, 2018-07-09:
`C:\Users\ACER>pmca-console-v0.17-win.exe updatershell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model ILCE-6000

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-console.py", line 96, in <module>
  File "C:\projects\sony-pmca-re\pmca-console.py", line 82, in main
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 360, in updaterShellCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 399, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 373, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 527, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 492, in _sendWriteCommands
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 471, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 74, in sendSonyExtCommand
  File "C:\projects\sony-pmca-re\pmca\usb\driver\windows\msc.py", line 227, in sendReadCommand
  File "C:\projects\sony-pmca-re\pmca\usb\driver\windows\msc.py", line 206, in _sendScsiCommand
pywintypes.error: (31, 'DeviceIoControl', 'A device attached to the system is not functioning.')
[7820] Failed to execute script pmca-console

C:\Users\ACER>`

revoRay, 2018-07-14:
+1 SERIOUSLY

semmio, 2018-08-21:
I have no programming experience and I'm completely lost. I downloaded pmca-console-v0.17-win and launched it but it exits immediately =(

santiago4m, 2018-09-20:
Hello everyone, I just wanna thank each one of you for your contributions, I've no idea of programming but following your steps (auf auch deutsch musste ich ein bisschen reden;)) after 3 hours researching it all, I succeded, and now I can record with my a6000 from my iPhone, which is what i really needed!!! 
Thanks a lot!

EfelerGibi, 2018-09-25:
> I have no programming experience and I'm completely lost. I downloaded pmca-console-v0.17-win and launched it but it exits immediately =(

you need to open cmd/powershell by pressing shift in the same folder as pmca-console-v0.17-win and write `pmca-console-v0.17-win.exe updatershell` now it should launch

eynkel, 2018-10-22:
I sucessfully follwed all the steps, but the camera can't be controled by WiFi for videos.

MenicoDo, 2018-10-27:
I have tried the method descripted above but i am still not able to take videos.
I assume because I did not update the Smart Remote App on the camera. Unfortunately Sony won't let me log in for some reason.
Is it there any other way to update the app on the camera, like an offline apk with the version 4.3.1?

dodolgarut1, 2018-10-27:
> I have tried the method descripted above but i am still not able to take videos.
> I assume because I did not update the Smart Remote App on the camera. Unfortunately Sony won't let me log in for some [...]

You will need to revert back to the original version. The "6000" not the "6001", and then you can update the app, after that revert it back to "6001"

MenicoDo, 2018-10-27:
Yep, I already had it set back to ILCE-6000 this was not a problem.
I go on https://www.playmemoriescameraapps.com/portal/ and the camera is perfectly recognized, ILCE-6000 - 96MB available. 
I was reading in this thread I need to update to the latest Smart Remote App to be able to do the 6001 trick. 
The problem is that I cannot login at all with my account (it was just created, I confirmed it with my email and that's it).
The camera is it not connected I tried with 2 computers, my phone, the camera without the name hack and for all the cases tells me access denied with my credentials. 
Dunno if my ISP is blocking the portal.
I saw that **pcma-console** has the hability to download apps from the market (I tried that too with camera disconnected and my credentials but nothing). 
I was hoping there was a way to get the Smart Remote apk, so the login it is not needed and if i could install it with the apk install feature in **pmca-gui**.

dodolgarut1, 2018-10-27:
> Yep, I already had it set back to ILCE-6000 this was not a problem.
> I go on https://www.playmemoriescameraapps.com/portal/ and the camera is perfectly recognized, ILCE-6000 - 96MB available.
> I was [...]

So the problem is that you cannot login to the website right?
I dont think its your ISP is blocking the portal, if it is then you can use VPN
Or maybe create another account, shouldnt be that hard i guess,

And from my experience i also didnt update the app first before changing the version to "6001"
And then i revert it back to "6000" update the app from the website and change it back to "6001" 

Also i dont really know about downloading apps from the **pcma-console**
Sorry about that

MenicoDo, 2018-10-27:
Hey thanks for the quick reply. So if i use Opera with the VPN version and a new account I can log in succesfully so there must be something fishy.
The problem is that Opera does not support the camera download extension. So now i have an account that works but no browser to download it.

About pcma-console that I found it today.
Basically you do 
**pcma-console market**
and that asks you to put username and password (and i assume) eventually will let you download an apk. But unfortunately can't connect that (because again tells me access denied)

dodolgarut1, 2018-10-27:
> Hey thanks for the quick reply. So if i use Opera with the VPN version and a new account I can log in succesfully so there must be something fishy.
> The problem is that Opera does not support the [...]

You could use google chrome with vpn extension and download it from there
I use this [vpn](https://chrome.google.com/webstore/detail/just-proxy-vpn-hide-ip-%20/ojedkepkekklpjcgdfiahladdbopbooh?hl=en), its free and its working for me (not sponsored) :p

Thanks :smile:

MenicoDo, 2018-10-27:
> You could use google chrome with vpn extension and download it from there
> I use this [...]

Please allow me to buy a virtual beer, it worked!!!

I connected via VPN with chrome so the camera was recognized and I could use my account.
I update the Smart Remote App to the 4.3.1 (that has the support for QR code, mine was the stock one since I just bought the camera) did the name change and I recorded my first video with the smartphone remote. 
Thank you very much!

lydkid, 2018-11-19:
I would be ever so grateful if I could get some help. I'm not a programmer or anything but ended up on this forum after hours of trying to get video mode to work on my Sony a6000. So far I have downloaded the OPEN MEMORIES Tweak app to my camera and have the PMCA downloaded to my computer.

What are the next steps to changing the model name of A6000 to A6001?

I see that you all mentioned to input those codes..but I'm confused as to where and how?

Thank you guys!

riziks, 2018-12-03:
> In case someone wants a full guide on how to change the model of Sony A6000 (based on @mungewell's and @syntaxer's post):
> 
> 1. Install the **OpenMemories: Tweak** app on your camera and go to the **Protection** tab and select **Unlock protected settings**.
> 2. Run the **pmca-console** through the command line / terminal with the flag **updatershell** (i.e. `pmca-console-v0.17-win.exe updatershell`).
> 3. Then change the model name from **ILCE-6000** to **ILCE-6001** with the following command: `bk w 0x003e0005 49 4c 43 45 2d 36 30 30 31 00 00 00 00 00 00 00`.
> 
> If you want to **reset** the model name to **ILCE-6000** just write `bk w 0x003e0005 49 4c 43 45 2d 36 30 30 30 00 00 00 00 00 00 00`.

Thnx, do this with A7, just rename to 7S:  bk w 0x003e0005 49 4c 43 45 2d 37 53 00 00 00 00 00 00 00 00 00

ripleycs, 2018-12-16:
This worked for me!! Thanks!!!!!

digitalbusiness-io, 2018-12-21:
I'll pay someone $10 to show me how to do this quickly via Skype. I'm running Mac osX, sony a6000, want to record video/live view with my Android app. Please send me an email to kyle@digitalbusiness.io

aj005, 2019-02-15:
I got it work! it was pretty easy but did take some time as i'm not a programmer of any sort! 
But i'm super pumped! and I really appreciate all the hard work everyone put into this!! if anyone needs help i can do my best to try and help

cvessup, 2019-02-16:
I need help!! i successfully changed it to 6001 but it does not work. im wondering if i needed to update the smart remote, but when i tried switching back to 6000, im getting the unknown device:ILCE-6001. i don't know what to do next, Please email me c.vessup@gmail.com

EfelerGibi, 2019-02-16:
> I need help!! i successfully changed it to 6001 but it does not work. im wondering if i needed to update the smart remote, but when i tried switching back to 6000, im getting the unknown device:ILCE-6001. i don't know what to do next, Please email me [c.vessup@gmail.com](mailto:c.vessup@gmail.com)

Try `pmca-console updatershell -m ILCE-6000` this should work

cvessup, 2019-02-16:
I saw this on here and I tried this multiple times. No luck. Any other suggestions?

w-stoltz, 2019-02-27:
I have the model number changed to 6001 but on the iOS app I still only have photo shoot mode as an option, any ideas?

Dutze, 2019-03-01:
Hi guys,

I'm super glad, that I found this site here - but I have one question:

Is there touch focus or an autofocus button in the activated video mode on Sony A7? I couldn't find any clear info on that yet :)

Have a nice day, dear reader!

Dutze, 2019-03-31:
Hi guys :)

I have a A5100 and I'm trying to enable the touch-focus in video mode on the remote app.

For this I changed the model name of my camera to ILCE-6300 (because the A6300 should support this touch-focus feature with the remote app). But that hasn't changed anything on my remote app...

Any tips from you guys? Would be really nice! :)

louismellies, 2019-04-18:
Hello, my firmware version is 3.1, so, is it still possible to install openmemories tweak?
I have conect on the page of installation on Safari but the navigator does not detect my A6000.
I need help thanks!

vama2301, 2019-04-22:
Took me some working hours to figure it out through telnet and tweak as an amateur, but now it´s working on my Sony A7. Well worth the time, can finally use my phone as an external monitor. Thanks a lot lads.

vama2301, 2019-04-22:
> Hello, my firmware version is 3.1, so, is it still possible to install openmemories tweak?
> I have conect on the page of installation on Safari but the navigator does not detect my A6000.
> I need help thanks!

I recommend downloading the installer to your computer instead of using the web page. 
You can find it here: https://sony-pmca.appspot.com/plugin/install

Remember to plug in and turn on the camera before opening the program. Worked for me. Good luck.

basbo52, 2019-04-30:
> I have the model number changed to 6001 but on the iOS app I still only have photo shoot mode as an option, any ideas?
> 
> EDIT: I figured it out!!
> 
> Make sure your camera ID is 6000, update the smart remote from the default version by going into the app store for the camera itself (you need to make a sony ID and everything first)
> 
> THEN change to 6001, reload the app on your phone and it works

Hey,
Did this after having problems with it too but now I updated the remote app on my a6000 I cannot run the program in the cmd consol
I already tried deleting the Tweak app from the camera and putting it back but I still got an error message on my cmd consol that says the following  : 

_```
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model ILCE-6000

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-console.py", line 96, in <module>
  File "C:\projects\sony-pmca-re\pmca-console.py", line 82, in main
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 360, in updaterShellCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 373, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 527, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 501, in _sendWriteCommands
Exception: Firmware update error: Low battery
[28660] Failed to execute script pmca-console
```_

Please help me

Arnie97, 2019-05-01:
@basbo52 The error message was "Low battery", so just charge your battery before your operations.

permargarin, 2019-05-14:
Hi.
When I try to download and install the pmca-console-v0.17-win.exe file, my pc says its virus infected. What to do??

rsjunderground, 2019-05-16:
It worked just fine for me. but i would like to enable touch-focus in video mode for my a6000 in the  android app. Someone knows how to do it?

Pekkios, 2019-07-05:
Hi everyone and thank you in advance for your time. I installed OpenMemories Tweak app on my A6000. I selected 'Unlock protected settings'. I plugged my camera to the PC and turned it on. Then I launched the 'pmca-console-v0.17-win.exe updatershell' from the terminal and then the command 'bk w 0x003e0005 49 4c 43 45 2d 36 30 30 31 00 00 00 00 00 00 00'. After this I got a 'Success' on the terminal but the camera was stuck and I had to unplug it and to remove and reinsert the battery to turn it on again. After this procedure, the camera is still A6000. Tried this operation several times with no success. Can you help me?

yaz241113, 2019-07-16:
> Hi everyone and thank you in advance for your time. I installed OpenMemories Tweak app on my A6000. I selected 'Unlock protected settings'. I plugged my camera to the PC and turned it on. Then I [...]

I am having the same issue, someone please help :( been at this for hours

Pekkios, 2019-07-16:
> I am having the same issue, someone please help :( been at this for hours

Probably I used to forget to digit exit on the prompt after the procedure was succesful. Anyway with the new Android App "Imaging Edge" I am still unable to record even if the name changed correctly to A6001.

darkileaves, 2019-08-01:
Hello everyone, I was looking for the touch to focus feature on video mode and I didn’t seem to get it? I know this is a request by many, so is there something we can do to get it?

[50 more replies]