ma1co/OpenMemories-Tweak #138: OpenMemories not working on new a6500

Quorro, 2016-12-08:
HI, 

this is the first time i write in a GitHub.. 

i have tried to tweak a new Sony a6500 to remove 30 minute video limit and pal/ntsc warning..

Obviously i set the camera in MTP and downloader the plugin for safari and explorer.

the process works fine... and the log on the home page respond "finished task" 

the tweak find the cam and the trial number the firmware version etc.. but when process finished nothing appear on the application list on the camera.

How can i help to fix this?...

i have to dump the firmware of a6500 and upload ?.

thanks you. and really nice work.. 

how i can donate to the author of this useful god lantern graal tweak ?.

Djskapy.

Quorro, 2016-12-12:
OK guys.. thank you :). i have found a workaround... now works even on 6500...

sleepinghouse, 2016-12-30:
@Quorro Did you get it to record 4K longer than 30 minutes? The box in the app is greyed out.

Quorro, 2016-12-30:
@sleepinghouse 

Yes , it's a my mistake.. now works and the annoying pal/ntsc message don't appear.. BUT. now.. that i have removed the app "camera initialization" the cameras says that I'M shooting in PAL.. instead of NTSC hahahahaha ...now i'm forced to set NTSC hahaha...

sleepinghouse, 2016-12-30:
@Quorro Interesting. Are you able to record in 4K beyond the 30 minute limit? I still can't uncheck the box in the Open Memories Tweak.

Quorro, 2017-01-03:
Yes friend :)... don't worry for the greyed out option on 4k .. just enable the top option and you are able to record "time free". even on 4k. but .. remember to "re-enable" protection in developer tab when you have enabled you options.

closed 2017-10-30 by ma1co