ma1co/OpenMemories-Tweak #140: [Request] A6000: Allow camera operation while charging.

syntaxer, 2016-12-10:
Currently if you plug in a charger on an external battery pack, you cannot use the camera or even access the menu.
I tested an RX100 and noticed that you can use the camera while it is charging.
I don't know if it's possible, but I would like to use my a6000 while it is connected to a battery pack.

pjb108, 2016-12-12:
OMG YES +1
 this has been a killer on the a6000, battery life sucks

gamerpaddy, 2016-12-30:
if its not possible over software, i suggest you getting a battery dummy for NP-FW50 (around 15-20$)
and a 7.4v li-ion battery pack (commonly used in bike flashlights. must be a 2S / 2Cell li-ion/lipo.

cfanyyx, 2017-02-02:
+10086

Korny-like, 2017-03-05:
Like to have this feature!

byChillout, 2019-07-22:
THISSSSSSS <3