ma1co/OpenMemories-Tweak #143: PAL / NTSC selector not working NEX 5R

gamerpaddy, 2016-12-30:
Hello, i got NEX 5R EU version (with PAL) but i wanted to enable NTSC for 60p.
Its enabled in the Tweaks menu but the PAL / NTSC selector wont show up in any configuration menu.

Does this only work from NTSC to PAL? I need it the other way around.

NEX5R 
Version 1.03 
I would really apprechiate if someone could do that.

 Thanks.

gamerpaddy, 2016-12-31:
tweaklog output, seems to be my backup file is not supported or corrupted?
`[INFO][MainActivity] application start
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: NEX-5R
[INFO][InfoAdapter] Backup region: 
[INFO][InfoAdapter] Tweak app version: 0.9
[INFO][InfoAdapter] Java API version: 1.1
[INFO][onPause] .InfoActivity
[INFO][onResume] .ProtectionActivity
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][writeProtection] setting protection to true
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][writeProtection] success
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][BaseProperty.setValue] backup property 0x1070148 is protected
[ERROR][LanguageAdapter.getOffValue] error reading region: java.io.IOException: Unsupported backup version
	at com.github.ma1co.openmemories.tweak.BackupFile.checkData(BackupFile.java:16)
	at com.github.ma1co.openmemories.tweak.BackupFile.<init>(BackupFile.java:8)
	at com.github.ma1co.openmemories.tweak.Backup.readData(Backup.java:42)
	at com.github.ma1co.openmemories.tweak.RegionActivity$1.getOffValue(RegionActivity.java:91)
	at com.github.ma1co.openmemories.tweak.RegionActivity$1.getOffValue(RegionActivity.java:12)
	at com.github.ma1co.openmemories.tweak.BackupSwitchAdapter.setEnabled(BackupSwitchAdapter.java:70)
	at com.github.ma1co.openmemories.tweak.ItemActivity$SwitchItem$1.onClick(ItemActivity.java:50)
	at android.view.View.performClick(View.java:2485)
	at android.widget.CompoundButton.performClick(CompoundButton.java:99)
	at android.view.View$PerformClick.run(View.java:9080)
	at android.os.Handler.handleCallback(Handler.java:587)
	at android.os.Handler.dispatchMessage(Handler.java:92)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)
[INFO][onPause] .RegionActivity
[INFO][onResume] .VideoActivity
[INFO][BaseProperty.exists] backup property 0x3c04b6 does not exist
[INFO][onPause] .VideoActivity
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: NEX-5R
[INFO][InfoAdapter] Backup region: 
[INFO][InfoAdapter] Tweak app version: 0.9
[INFO][InfoAdapter] Java API version: 1.1
[INFO][onPause] .InfoActivity
[INFO][onResume] .DeveloperActivity
[INFO][onPause] .DeveloperActivity
[INFO][onResume] .ProtectionActivity
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][MainActivity] application end

[INFO][MainActivity] application start
[... 264 lines cut ...]
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .DeveloperActivity
[INFO][onPause] .DeveloperActivity
[INFO][onResume] .ProtectionActivity
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][guessProtected] backup property 0x10d008f written successfully (no protection)
[INFO][writeProtection] setting protection to true
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][writeProtection] success
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][MainActivity] application end

`

alexnesher, 2017-02-08:
I have the very same problem with my NEX-5R. Could it be possibly solved? Cause I always wanted that option

gamerpaddy, 2017-02-08:
i made a experimental firmware update which i havent tested, if you want to take the  risk i can send it to you. it shows as 1.04 but is 1.03 with a unified config (all language config copied over the chinese, european, us etc. one

theantnest, 2017-07-23:
Did anyone ever solve this? What happened with the experimental firmware?

gamerpaddy, 2017-07-23:
i installed that firmware, which basically was 1.02 with its version number changed to 1.04 to be able to install it.  someone said in 1.02 it worked, in 1.03 it didnt.     but it did not fix it, but hey, im the only one with version 1.04 hehe

drozdov1984, 2018-09-01:
Hello, you still have  experimental firmware for nex 5r?

drozdov1984, 2018-09-01:
in my camera there is no application menu because of another region, I thought that maybe your firmware will help

drozdov1984, 2018-09-01:
same problem
https://github.com/ma1co/OpenMemories-Tweak/issues/32

gamerpaddy, 2018-09-01:
https://www.file-upload.net/download-13298676/FirmwareData_NEX5RV104.dat.html
you could try that, its the same im using on my 5R right now. shows up as 1.06 (i have to increase this everytime because i cant downgrade.)

drozdov1984, 2018-09-02:
Yes, Im using a Nex 5R. Thanks

gamerpaddy, 2018-09-02:
no idea where to change the region, i just use the "universal" .bin for all other region bin's
i attached a 1.07 with small changes to the previous one (in the .sum file), but im not sure if it even works 
with this one im using 1.03 as base firmware. 
https://www.file-upload.net/download-13299181/FirmwareData_NEX5RV104.dat.html

drozdov1984, 2018-09-02:
thank you very much

Coderyne, 2020-01-27:
Can you upload the firmware again？

Liriel-666, 2022-08-25:
2022 and the error is not removed

lijl070, 2024-10-10:
2024 still happens

gamerpaddy, 2024-10-12:
cant be fixed

closed 2024-10-12 by gamerpaddy