ma1co/OpenMemories-Tweak #145: Reset E:91:01 Flash Error

peppersass, 2017-01-05:
Would it be possible to add reset of the E:91:01 Flash Error? I need it for the RX100M5, but it's an issue with all Sony cameras. This is a flag that gets set if the camera detects something wrong with the flash. Unfortunately, it gets set after a repair or after replacing the flash, and the user can't reset it. The service manual says it can only be reset with a special program.

If you do this, the option should be accompanied by a warning that resetting the flash error could damage the camera if there really is something wrong with the flash.

peppersass, 2017-01-18:
Further thoughts on  this request:

The E:91:01 flash error affects many Sony models.  In many cases it bricks the camera, resulting in an expensive repair bill.

I got the error after accidentally shorting the flash caps while repairing an RX100 M3. Replacing the boards associated with the flash and reinitializing didn't clear the error. The service manual indicates that the error also occurs when any part of the flash system is repaired or replaced, and once that happens it's permanent. The only way to reset is with the "Flash Error Repair Tool Ver_[].[].exe". Of course Sony doesn't make this program available. The only solutions are to spend $300 to get Sony to reset the error, or try replacing the MCU board for $160. However, the service manual says there are camera-specific adjustments that must be stored in the MCU and refers to special software and possibly service hardware for that.) Stock repair boards may or may not have the right adjustment parameters, so this is a riskier path.

Before incurring such costs, I'd like to try resetting the error. The question is where the code is stored. I suspect it's not in Backup.bin because reinitializing the camera doesn't clear it (I think all that does is copy Backup.bak to Backup.bin.) I've used your excellent instructions to explore the file system via Telnet, but there's nothing obvious that would indicate a file where the code might be stored, and no obvious service programs that might reset it. The code might be stored in the firmware flash memory.

My guess is that there's a way for a program to reset the E:91:01 error, either via the USB interface or via an app running on the camera. While I've got a fair amount of tech experience, ranging from mainframe assembler to Android apps, I don't have the expertise to reverse engineer the Sony firmware to figure out how to reset the error.

It's conceivable that forcing a firmware reload would clear the error, but the stock Sony firmware update program will not allow loading the same or an earlier version. In looking over your source code, I see that the firmware load option communicates the new version number to the camera, and the camera will respond with an error code if the version isn't right -- i.e., greater than the installed version. I suspect it might be possible to fool the camera by sending the current version number plus one, that would forever require fooling the camera to install Sony's future updates (if any -- there hasn't been one since 2014.) Have you been able to force a firmware load of the same version as that which the camera runs?

Do you have any suggestions on how to proceed?

ma1co, 2017-01-27:
If your assumption is correct and it is really a flag stored somewhere and not a defect the camera detects everytime it's booting, i'm pretty sure it would be stored in Backup.bin.

You're wrong about reinitializing the camera, this resets only a few properties in the Backup.bin, a lot of settings are preserved (stuff like the serial number). Backup.bak has the same contents as Backup.bin (the settings are mirrored in both files), it has nothing to do with reinitializing. A firmware update won't help either, as it mostly preserves the Backup as well.

So if you want to, you could post your Backup.bin here, maybe someone will find something (be aware that the file contains personal information like your serial number). I'm not too optimistic though, as there are thousands of settings...

peppersass, 2017-01-27:
Thanks for the reply! Very helpful. I've attached my Backup.bin file in a .zip in case someone can help.

Question for maco1: Is it possible to modify the Backup.bin file from the Telnet interface? What I'm thinking is if someone with an RX100M3 would post their Backup.bin file I could try replacing mine with it. Otherwise I guess I'd have write an app do it.

[BACKUP.zip](https://github.com/ma1co/OpenMemories-Tweak/files/736499/BACKUP.zip)

ma1co, 2017-01-28:
The string "E:91:01" is actually stored in the Backup (property 0x010703de). The default value seems to be "0x00" (as a string). Can you try to run something like this over telnet?
`bk.elf w 010703de 30 78 30 30`
I'd guess that this might change the error message displayed...

peppersass, 2017-01-29:
Thanks very much for this. It was helpful (I think).

I patched Backup.bin as you suggested and read the location back using the "r" command in bk.elf just to make sure. However, when I restarted the camera it displayed the E:91:01 error. Sure enough, when I read location 010703de with bk.elf, it was loaded with E:91:01 again.

Unfortunately, there's no way to know whether the patch reset the error and it is, in fact, a hardware error, or there's more to resetting the error than changing location 010703de.

I have a replacement MCU card coming on Monday and hopefully that will clarify the situation. I felt it made sense to spring for a new MCU in light of the low odds of finding a way to reset the flash error, plus the fact that after I changed the flash board the problem gradually progressed from the flash not working to not being able to take photos to the buttons not working at all in camera mode. It's conceivable that the firmware has some sort of timer that shuts down functions the longer the E:91:01 error persists, but that seems unlikely. It seems more likely that shorting the flash capacitors damaged the MCU and the damage is getting worse (I've seen solid state electronics do things like that after a lightning strikes.)

peppersass, 2017-01-30:
UPDATE: I installed the replacement MCU and the E:91:01 error disappeared and the flash works. Clearly, nothing was wrong with the flash hardware. I can take pictures too. 

However, the camera will not focus (neither with the shutter button nor in continuous focus mode.) The green focus guides do not appear.  I reinstalled the old MCU and does focus, but of course has the E:91:01 error. Doesn't appear to be anything wrong with the focus hardware. My conclusion is that the focus problem is due one of the following:

1. The new MCU is defective. Seems unlikely, but you never know.

2. There are camera-specific parameters in the new MCU that must be manually set or adjusted, and one of them has to do with focus. The service manual says that when replacing the MCU there are parameters that must be copied from the old MCU or manually adjusted using a program called Adjust Manual on the Adjust Station: product ID, USB Serial No., Lens carry data, AWB adjust, Light sensor adj, Eye sensor adj, EVF Data Input. The MAC address has to be changed to match the old MCU, too. None of these sounds specifically like focus adjustment, however, and I would think there would be reasonable defaults (for example, the new MCU has a serial number.)

To check this out, I tried copying Backup.bin from the new MCU to the SDCARD, then to the old MCU. The cp command completes without error, but I don't think the file actually changed. The file date didn't change and it it still has "E:91:01" at 010703de. Doesn't seem to be a permission problem, as Backup.bin has rw across the board. Does the firmware keep a copy of the parameters in memory and then write them to Backup.bin at shutdown?

I'd like to try copying all the files from /settings on the old MCU to the new MCU. Is there a way to do that?

Meanwhile, I'll contact the parts supplier and see if I can exchange the MCU.

FWIW, the Backup.bin file from the new MCU is attached.

[new BACKUP.zip](https://github.com/ma1co/OpenMemories-Tweak/files/740558/new.BACKUP.zip)

ScartFace, 2017-02-11:
Hi there,

I'm in almost the same situation with my RX100M3. Bought it with non-functional LCD and EVF but the rest was OK. So I replaced the LCD, ripped it's flex cable, also replaced that one and life could have been good again. But I touched the capacitor contacts too often and finally even accidentally shortcutted them with a screwdriver. In this state, the flash would not work anymore but I didn't have the error code. Then I replaced the flash capacitors together with the flex board that they are soldered on. After switching the camera on again, I now have this infamous E:91:01 flashing in the display. Good thing tough is, that the camera works fine otherwise as long as the flash unit is not popped out. When the flash is up, the camera refuses shooting.

I also have a copy of this Service Manual where it reads that Error E:91:01 needs to be reset using this "Flash Error Repair Tool". So I'd second that we only would have to reset the flag(s) that indicate that the flash unit is disabled for safety reasons.

I have no idea how to work with OpenMemories-Tweak yet, so could use some detailed instructions. I managed to install it, though and also connected to the cam via telnet. But what then?... How did you create that backup.zip and what is it good for?

Anyway, when you say that property 0x010703de contains this string "E:91:01", couldn't it be that this is only where the definition of the error code is? That said, changing this string wouldn't do anything to flash-disabled situation, it would probably only change the string that is shown in the display. 

Is there a way to compare a dataset of a healthy RX100M3 with our broken one? If both are freshly initialized, the only difference should be that flag.

peppersass, 2017-02-11:
I'm going to comment on the firmware issues in this post and the hardware issues in a subsequent post.

Yes, there's a way to compare the settings of the broken camera with the settings of one that works.

As I mentioned previously, I installed a new MCU board in my RX100M3. This allowed me to copy its Backup.bin file to my computer and do a diff (compare) with the Backup.bin from the old MCU using WinMerge. See instructions below for copying and decoding the file.

With both MCUs, I tried to use the exact same sequence to initialize the MCU, install OpenMemorieTweak, initiate Telnet and copy the Backup.bin file. This was so that there would be no differences related to different camera settings. Even with that precaution, there are quite a few differences. Some of them are probably ID-related, like serial number, MAC address, USB identifier, etc. Others are probably camera-specific adjustments, like the ones referred to in the service manual and listed in one of my previous posts. In some instances, the print_backup command in fwtool.py (see below) knows that a value has been changed from the default and displays the reset value. Often this is what I see in the same field in the new MCU's Backup.bin file.

I thought one way to try to reset the old MCU would be to go field by field, copying the value from the new MCU's Backup.bin to the old MCU's Backup.bin. Or, I could try copying values in the other direction to see which one provokes the E:91:01 error, The advantage of this approach is that it copies the adjustment values from the old MCU to the new MCU, which I'd ultimately like to do.

Unfortunately, I can't get any of the patches to stick. In a previous post in this thread, ma1co suggested I use the bk.elf w command to patch the "E:91:01" string. This worked, and I use the bk.elf r command to verify that the change took effect, but as soon as I exited Telnet mode or turned the camera off/on, the value changed back to "E:91:01". 

This is also true for every patch that I've tried. It seems I can't permanently change any of the values in Backup.bin.

I don't understand what bk.elf actually does. Seems like it might just patch working memory, not the actual Backup.bin file. I tried using the bk.elf sync command to see if that would sync the changes, but it didn't. As I mentioned in my previous post, it doesn't seem to be possible to replace Backup.bin with another version, even though it has rw permission across the board. I think that the shell is already in su mode because the prompt is #.

ma1co: can you shed some light on this? Is there a way to make a permanent change to Backup.bin? How does your app make the setting changes permanent?

Another issue is that the settings are stored in blocks, each with an address and a number of characters. Some are very long -- like 100 bytes. It's not clear to me whether I have to enter all the characters to change just one value.

In other words, it would be great to have some documentation on how bk.elf actually works!

**Instructions for copying and processing Backup.bin:**

When you connect to the camera via Telnet (I used putty.exe on the PC end), you're in an android command shell (which I believe is basically the same as the Linux command shell). You can get a list of commands by Googling "android commands". Once in the shell, you can use the cp command to copy Backup.bin or any other file to the SD card. For example:

cp /android/setting/Backup.bin /android/mnt/sdcard

Then all you have to do is remove the SD card, insert it into a PC/Mac card reader, and copy the file to the computer. I put it in a zip file on the PC because Github won't allow attaching a bin file to a message.

Next, download a copy of ma1co's fwtool.exe from here:

[https://github.com/ma1co/fwtool.py](url)

Get into a Windows cmd shell and use the fwtool print_backup command to decode Backup.bin:

fwtool print_baclup -f Backup.bin >outfile

This will create a file called outfile with the decoded version of the settings file.

peppersass, 2017-02-12:
On the hardware issues:

I would bet that replacing the flash flex board is similar to replacing the flash, as warned in the service manual: once you remove it, the camera goes into permanent E:91:01 mode and the only way to get it out of that mode is to reset the error.

In my case, shorting the caps caused the E:91:01 error right away. The camera worked perfectly, except for the flash. I don't recall if popping up the flash prevented taking pictures.

Like you, I replaced the caps and flex board. After replacement, I tested the original caps and they were OK. No way of knowing whether the flex board was damaged. Unfortunately, replacing the flex board seems to have put the camera in a worse state. In addition to the E:91:01, the camera immediately goes into autofocus mode when turned on, beeps, and stays there (green grid displayed) as if the shutter is being pressed halfway. It won't take photos. Also, the buttons and wheel on the back of the camera don't do anything in shoot mode (they work fine in playback mode.)

When I replaced the MCU board, the E:91:01 error went away and the flash works, I can take photos and the buttons/wheel on the back work. But now the camera will not autofocus. It's the opposite of the problem with the old MCU: pushing the shutter halfway down does not initiate focus. The camera won't focus in AF-C mode, either (which it does with the old MCU.) It does, however, do pre-focus when that option is enabled, so I think the actual focus mechanism is OK.

One theory is that the service manual is right and certain adjustment parameters must to be set in the camera before autofocus will work. I could verify that if I could copy the settings from the old MCU to the new MCU (which the service manual says you're supposed to do, but you need the Adjust Manual program to do it.) 

Another theory is that shorting the caps drained or damaged the small lithium battery that's attached to the shutter board, or perhaps damaged some components on that board. That board is only $40, but I hesitate to replace it because the service manual indicates that once it's removed it must be "adjusted". The presence of the lithium battery suggests there's a volatile memory on that board and that a parameter must be set in it. It's possible that without Sony's adjustment software and/or adjustment station, this can't be done.

Like I said, if I could copy the old MCU parameters to the new MCU (less the E:91:01 error), I could determine if something's wrong with the shutter board.

ScartFace, 2017-02-12:
Forget about the lithium battery, which is actually rechargeable. Its purpose is the same as in a PC, just to backup the internal clock and custom settings. I detached mine for some hours and the only effect is that the cam will ask you for date and time after it's re-assembled.

ScartFace, 2017-02-12:
By the way, for hardware you should compare prices. For example, if you mean the RL-1027 board to cost $40, it is only about $1 in this Denmark based online shop: http://uk.eetgroup.com/i/A2045281A-Sony-Mounted-C.BoardRl-1027. These are genuine parts from Sony Japan.

peppersass, 2017-02-12:
> By the way, for hardware you should compare prices. For example, if you mean the RL-1027 board to cost $40, it is only about $1 in this Denmark based online shop: http://uk.eetgroup.com/i/A2045281A-Sony-Mounted-C.BoardRl-1027. These are genuine parts from Sony Japan.

Have you actually obtained parts from this supplier? Near as I can tell, they only ship to Europe and Africa.

ScartFace, 2017-02-12:
> Have you actually obtained parts from this supplier? Near as I can tell, they only ship to Europe and Africa.

Yes, I'm from Germany. But I'm pretty sure there is competition where you live as well. 

Another idea; would it be possible to label the onboard firmware as Ver 1.1 so that the 1.2 FW update tool is happy to proceed?

peppersass, 2017-02-12:
There are several places to get Sony parts here, but they're all about the same price and more expensive than the Danish outlet.

I don't believe a firmware reload will work. First, my reading of the source to ma1co's GUI firmware loader suggests that a command is sent to the camera with the new firmware version, and the camera responds with the current firmware version and whether the new version is valid. It might work to patch the loader to tell the camera the new version is 1.3, but it might also be necessary to patch the firmware file. And it may be problematic finding the place or places to do that. And once you start that sequence, you have to do it for each update. Finally, ma1co said earlier in this thread that a firmware reload probably will not reset the flash error parameter.

peppersass, 2017-02-13:
I think I've had a breakthrough, of sorts. I copied both BACKUP.BIN and BACKUP.BAK from the new MCU to the old MCU. This eliminated the E:91:01 error. This means that there is indeed a property in the settings file that can be changed to reset the E:91:01 error.

The next step is to locate the property. As a test, I tried using bk.elf to patch a property that's one byte different in the old and new MCU files. Then I copied BACKUP.BIN to BACKUP.BAK. I copied the two files to the SDCARD so I could translate them on the PC. What I found is that BACKUP.BAK had the new value, but BACKUP.BIN did not. I was able to get both files to have the new value by using the bk.elf sync command. However, when I then read the property using bk.elf, the byte was back to its old value. I think what's happening is that it's being read from memory, and that the property is something being set each time the camera is turned on. Or something like that.

Unfortunately, changing the error message string in this manner didn't work. The property gets reset to the E:91:01 value when the camera restarts.

I'm going to try changing several other parameters so I can figure out exactly what's going on between BACKUP.BIN, BACKUP.BAK, bk.elf and memory. Once I know that, I should be able to change the properties that differ between the MCUs one at a time until I find the one that drives the E:91:01 error.
 
A faster approach would be to copy half the BACKUP.BIN file from the new MCU to the old MCU BAKCKUP.BIN and BACKUP.BAK, and if that doesn't reset the error copy the other half, and so on (i.e., do a binary search to find the property.) Unfortunately, I don't know how to map a location or set of locations in the binary file to the associated property address.

I should also mention that when I copied the setting files from the new MCU to the old MCU, I first changed the permissions on BACKUP.BIN to rwx across the board (chmod 777 backup.bin). I did this because a previous attempt had failed. However, that could be because I copied only BACKUP.BIN and not BACKUP.BAK as well.

Finally, when I copied the new MCU settings to the old MCU, the focus behavior changed from the old MCU problem (autofocus stuck on) to the new MCU problem (will not autofocus). However, the buttons on the rear still didn't work. I think this means the old MCU has one or more hardware problems, and the new MCU needs some parameters to enable autofocus. If I can find the E:91:01 reset, I may be able to fix the new MCU by copying the old MCU files to it and resetting the flash error.

peppersass, 2017-02-15:
Good news: **I found the property that resets the E:91:01 flash error!**

See instructions below for resetting the E:91:01 error **(USE AT YOUR OWN RISK!)**

I found the property address by trial and error. I used WinMerge to compare the backup.bin files from my old and new MCUs, then used bk.elf to patch the backup.bin and backup.bak files on my old MCU until the error went away. It took several hours because I started in the second half of backup.bin, which contains the string "E:91:01", reasoning that the flash error flag would be nearby. Also, there were more short and single-byte properties in the second half of the file. But it wasn't there. It was closer to the beginning of the file.

The property address is 0x000e0004. The old value (with flash error) was 00 00 00 00 01 00 00 00. The value that resets the error is 00 00 00 00 00 00 00 00. When I did the patch, the E:91:01 error went away. The string value "E:91:01" was still in property 0x010703de, so I patched that to the reset value of 30 78 30 30. Thereafter, the value remained the same (when the flash error was in effect, property 010703de would be reset to "E:91:01" when the camera rebooted.)

Here are instructions for resetting the flash error **(USE AT YOUR OWN RISK!)**:

1.  Set your Wifi Access Point parameters if you haven't done so already.
2. Install ma1co's PlayMemoriesTweak app.
3. Check the Enable Wifi and Enable Telnet commands in the Developer tab of PlayMemoriesTweak
4. Establish a Telnet connection to the camera (I use putty.exe on a PC for this.)
5. Enter the following commands (type the characters in bold):

/# **cd /setting**
(changes current directory to /setting)
/# **cp backup.bin /android/mnt/sdcard/backup.bin**
(saves a copy of backup.bin on the sdcard -- just in case)
/# **cp backup.bak /android/mnt/sdcard/backup.bak**
(saves a copy of backup.bak on the sdcard -- just  in case)
/ # **bk.elf r 000e0004**
(should display the contents of property 0x000e0004, which should be 00 00 00 00 01 00 00 00)
/ # **bk.elf w 000e004 00**
(writes all zeros to the property and displays the new value: 00 00 00 00 00 00 00 00)
/# **bk.elf r 000e0004**
(check that the property has indeed been set to 00 00 00 00 00 00 00 00)
/ # **bk.elf r 010703de**
(should display the contents of property 010703de, which should be 45 3A 39 31 3A 30 31)
/ # **bk.elf w 010703de 30 78 30 30**
(writes all zeros to the property and displays the new value: 30 78 30 30 00 00 00 00)
/# **bk.elf r 010703de **
(check that the property has indeed been set to 30 78 30 30 00 00 00 00)
/# **bk.elf s**
(This is the sync command. Should respond "DONE.". I believe this syncs memory with backup.bin and backup.bak.)
/# **reboot**
(restarts the camera. You may have to close the PC Telnet session for the reboot to complete.)

If something goes wrong, you can copy the unpatched backup.bin and backup.bak files from the SD card to the file system with these commands:

/# **cd /android/mnt/sdcard/**
(changes directory to the SD card)
/# **cp backup.bin /setting/backup.bin**
(copies backup.bin to /setting)
/# **cp backup.bak /setting/backup.bak**
(copies backup.bak to /setting)

ScartFace, let me know if this fixes your flash error.

The bad news is that it might remove the error, but it might not fix your camera. I did this procedure with my old MCU installed (the one that was in the camera when I shorted the flash capacitors.) While the patch did reset the E:91:01 error, the flash still did not work and the camera was still stuck in autofocus at startup and the rear button panel didn't work in shoot mode (but did work in playback mode.)

I then installed the new MCU and copied the patched versions of backup.bin and backup.bak from the SD card to the new MCU's file system using the Telnet interface. After rebooting, the camera works perfectly, including the flash (I've only tested the basic functions and need to do a thorough test, but at this point I'm very encouraged.)

My conclusions from this are as follows:

1. My old MCU board was damaged. I'm not sure if the damage was caused by the capacitors being shorted or this happened when I replaced the capacitors and flash flex board (some of the cables had to be removed from the MCU and perhaps the connectors or components on the board were damaged.) The symptoms were flash not working, autofocus stuck on and rear panel disfunctional.

2. All these things worked on the new MCU, but it would not autofocus until all the settings were copied from the old MCU. This makes sense because the service manual says when the MCU is replaced a program called Adjust Manual must be used to copy a bunch of parameters from the old MCU to the new MCU. It seems to imply that if those parameters aren't available (e.g., old MCU is dead), the Adjust Manual program (running on the "Adjust Station") is used to adjust these parameters. The manual also implies that Adjust Manual copies only the parameters that are needed. I didn't know the property addresses for those settings, but was able to finesse this by simply copying the entire backup.bin and backup.bak files from the old MCU (after restting the E:91:01 error.) Now the new MCU has the identity and all the settings of the old MCU.

I hope this is helpful for others.

ScartFace, 2017-02-16:
Confirmed that this procedure resets the error E:91:01! 

Unfortunately, the error was back after a while. When I flipped up the flash unit, it would should the flash charging symbol for some seconds, which seemed to fail. After that the E:91:01 was back.

Apparently the camera still has some hardware issues, may it be the capacitors' board, the flash unit or the MCU. Good to know how to reset the error and copy config data around!

A minor correction to your instructions. You are missing a 0 when you reset the error flag. It should read
bk.elf w 000e0004 00 

Many thanks to peppersass for the troubleshooting and of course to ma1co for providing the tools!

peppersass, 2017-02-16:
Thanks for confirming the reset procedure. Hopefully this will help others with the same problem.

Also, thanks for pointing out the error in my instructions. I have edited the line in question.

**And a big thank-you to ma1co for the custom apps and firmware tools that made this possible! Also a big thanks to the folks at nex-hack for their extensive work decoding the firmware.**

Did you also reset property 010703de? Maybe that's necessary, too.

Otherwise, I agree it sounds like a hardware issue. Does the camera work OK after the reset procedure if you don't attempt to use the flash?

My guess is that the capacitors aren't charging. You might be able to confirm this by doing the flash reset, turning on the camera, allowing time for the caps to charge (don't raise the flash), then shutting down and checking the caps with a volt meter (careful not to short the caps again!) It's not a definitive test because the error itself prevents charging. 

BTW, I constructed the flash discharge jig recommended by the service manual so I could swap MCUs without getting shocked or shorting the caps. Of course, it takes care when using the jig not to short the caps. Very poor design for service.

Are you sure that when you replaced the capacitors and flex board that you matched the polarity of each capacitor to the symbol on the flex board? There's an obvious "+" mark for one of the caps (and I think a "-" mark too), but the other one only has a somewhat obscure "-" mark. I checked and rechecked the polarity several times and still was worried that I didn't have it right.

Otherwise, I would suspect the MCU. The flash didn't work after I reset the old MCU, but I only tried it briefly and may not have waited long enough for the error to recur before replacing the MCU.

zenzarino, 2017-06-06:
hello everyone, 
i've a question. my rx100 gives the same error code and flash does not fire at all. I cannot event select any flash mode at all. anyway, do you think the patch will work or you think that even if works it will show the same error again soon?

haowulin, 2017-07-15:
I particularly signed-up the website to thank the solution/method provide here to solve the E:91:01 error.
The method indeed works!
I have a RX100m3 and 2~3 month ago, it showed E:91:01 error out of nowhere and the flash could not fire.
The rest of the functions are fine and I continuously use the camera since.
Today, I tried the instructions provided here and the error code was GONE and flash was BACK!
Everything works now and back to normal.
Thank you so much for providing the useful instructions.

VAArvind, 2018-05-18:
Non-programmer here, trying to solve the Error 91 issue with my RX100 Mk 3. I downloaded the app OpenMemories Tweak to camera, enabled wifi and telnet. I also downloaded Putty to my PC. Changed port number to 23 but Putty says "Unable to open connection. Host does not exit"  
 Could someone please guide me on how to establish a Telnet connection to my camera from my PC?

nc1016, 2018-06-03:
Wow! I can also confirm that this worked for me... thank you so much to all involved in providing this solution!  I got here because my lovely wife was trying to shoot picture of my daughter and I tubing down a river when she fell off a log and dumped our RX100 MK3 completely in the water.  After about a week in a bag of rice, all the condensation dried up from the lens elements and it came back to life but had the E:91:01 Flash Error and the flash was disabled.  After following this solution, the error is gone and the flash is working again!  

To answer @VAArvind , I also stumbled for a second when trying to establish the telnet connection, so here are a couple more details:

1. On camera: Menu > Toolbox > Tab 2 > Pwr Save Start Time - set to 30 Min so cam doesn't turn off
2. On camera: Menu > Apps > OpenMemories Tweak [OMT]
3. In OMT: Developer > Enable Wifi - wait for it to connect and display IP address (xxx.xxx.xxx.xxx)
4. In OMT: Developer > Enable Telnet - should display "telnetd running on port 23"
5. On computer: telnet xxx.xxx.xxx.xxx 23 - You need to give the IP address that was reported by 
OMT and the port number 23
6. On camera: reset Pwr Save Start Time

Hope this helps!

ger0nim077, 2018-10-01:
Thank you so much everyone!
Especially **@ma1co**  and  **@peppersass**  very big thank to you guys, you did a great job!!!!
I have the broken flash E:91:01 issue on my RX100Mk4.
I followed all the steps, the erasing of the error went well.
Unfortunately the flash won't work, and after a few mins the error returned. So i think it is indeed some broken hardware inside :)
I am thinking to give a try to @ScartFace website suggestion from Denmark (https://uk.eetgroup.com/i/A2045281A-Sony-Mounted-C.BoardRl-1027) , but can any one point me to the exact part number i have to order?

Thank you guys, you rock!

ger0nim077, 2018-10-01:
@VAArvind 
my camera didn't properly connect to wifi at the beginning...i had to leave it very very close to the router and it worked.
on camera you need to have the ip displayed on the screen, then you know for sure it is connected and afterwards telnet will work too for sure

marciman11, 2018-12-14:
Like @haowulin I just signed up to GitHub to say thank you. I have no further idea of programming but this thread made it possible to me to bypass Sonys planned obsolescence. I love you guys so much. Thanks!

cyoungii, 2019-02-06:
Signed up for an account just to say thanks for the solution! Fixed my flash, and removed the error!

Drillionaire, 2019-02-06:
Thanks @peppersass for the research and thorough instructions! Successfully removed the error on RX100Mk3 and restored full flash functionality.

antonetwo, 2019-11-02:
hi guys im an absolute nube on this site and A non programmer , i just wat to fix my camera that has the  E:91:01  error. mine is a  DSC - WX350. I can follow clear instuctions well , if someone could take the time to share a list of links of progarmes i need to make this happen , also clear instructions on what to do i would be forever in you dept and extreemly greatful... Thanks in advance

ducatiroger, 2019-12-14:
Thank you @peppersass for figuring this out.  I have a rx100 v that **was** displaying the very annoying E91.01 error.  Your work and guide allowed me to clear the error.  Thanks also to @ma1co for the tweak tool.

kirguis, 2021-01-11:
Hi ! just signed up for to say tank you from Argentina , i bought a rx100 m3 very cheap from a guy who dropped the camera and the viewfinder  and flash unit stop working ,  the screen showed the image inverted  and shows the famous  E:91:01 error ,  the flex that connects the flash and viewfinder unit was loose with the droop , i rconnect the flex and the viewfinder  and the camer works fine except for the flash error , i follow the procedure from peppersass and it works !! the error has disappeared and the flash unit come back to life !! thank you so much !!

rxfan100, 2021-01-26:
I too can confirm ,works on rx100M3,thank you so much ,I only do hardware repairs.
I do have a question (as old as this topic is) how do I accomplish with the RX100 1 what is done here with the M3?.
The M1 cannot have any app's installed at all.
The solution here so far is good for rx 1 thru 5,is there a possibility for one for the Mark 1 model? (which has same "lock-out" issue.
Thank you so much.

magickone, 2021-02-05:
I have RX100M5A with the same problem E91:01 what to do?

Doster321, 2021-02-19:
Thank you! I got error E:91:01 after I'd used interval Time Lapse shooting with flash for about 10 min, 30 shots. I checked fuses on a mainboard, they were OK.  How to do this is another matter, and it's not difficult for DIY people.   The above mentioned action, on the site, worked for my RX100 M3 ! I admire you,  peppersass  and ScartFace.
I am a non-programer. It was my first activity with Telnet etc, so here are some advice for dilettantes like me, trying to cope with the problem.  
First of all  signs /#  before commands are not parts of the command,  just leave them (they are not in bold).                                                                                                                   
Secondly-  You can not use in Putty  shortcuts  Ctrl C  and Ctrl V.  Copy a command  by right clicking and choose 'copy'  then paste to Putty just right clicking a blinking green point .  Yes- only right click!                                                                                                                   
Thirdly  - during the procedure a camera is not connected to a PC, it works with wifi.
I've done all of the mistakes, so it took me 2 days to complete the demanded actions.  Hope it helps you.

totolight, 2021-11-21:
Hi,
Many thanks for this thread. I was wondering if the same logic could apply to other types of error.
My SONY RX100 M3 boots with the error "E:62:20", then retracts the lens, and then tells me "Turn Power off then ON".
Either the camera goes to the thrash, either I find a way to force this message to go away so I can use the camera
Any suggestions ?

rxfan100, 2021-11-22:
I can say I have tried it with an RX100 M4 and it worked BUT as for YOUR "lens error" it will NOT make it go away.
A lens error is a hardware error (I have repaired many),either the gears or front cover which covers lens or ribbon cable inside of lens goes bad,if you arent experienced with lens assemblies I suggest buying a known good used lens or a new one,they dont cost much but is a little tricky installing,step 1 after opening camera to replace lens is to discharge the flash capacitors or 1 touch and you get zapped or burned  badly OR the flash board goes bad OR (if lucky) nothing goes bad BUT you end up with the error code of this thread with NO flash and NO working veiwfinder hence having to re-flash camera.
Once flash capacitors are discharged fully no error will happen.
Search the web of making a discharging tool if you are going to try this.

totolight, 2021-11-22:
The first error "Turn Power off then ON" appeared after the camera fell on my wooden floor. It did not fall from very high fortunately ... 
Then, I tried to solve the problem by opening up the camera and releasing pressure on the lens (following some random youtube videos...)
I did get zapped when I touched the bottom of the camera, nothing too bad.

Would you mind sharing with me how to get rid of this "Turn Power off then ON" ("E:62:20" ?) error ? Everything above with telnet I understood how it works.
Because I would like to use the camera on a tripod as a webcam, and I really do not mind losing some functionalities. I'm fairly sure that if I can focus the lens I would get a good image. (I'm a camera engineer by the way, however my work speciality is in theoretical performance calculations ^^)
I just want the camera to leave me alone and let me power it on ...

rxfan100, 2021-11-23:
Sorry,it's the way many new products are programmed, one error will disable the unit,take it up with Sony ha ha.
In my opinion none of the RX models were meant for extended video recording,for that there is a sister model "ZV-1"  which still works as a camera but is mainly for blogging and extended video recording which will not damage the CMOS sensor.

ANYWAY your error "E62:20" is a software code for "hard or known" hardware fault (image stabilization issue) and off-topic of original thread,I would suggest a new thread and hope original posters reply,there was incredible work by a few members on this issue ( E91:01) which plagues many RX models for no reason out of the blue.
I understand your frustration,as I did with the error of this thread but I firmly know yours is a faulty lens issue ,there is more inside the lens than you think electronically which can break from any "drop",I have repaired enough lenses to suggest just buy a new lens from EBAY,they are under 100$ or put-up with the issue till camera stops working.

TRY THIS,I found a few posts which say "turn off image stabilization" in movie and photo mode within settings ,your error is coming from faulty stabilization within the lens which is done through a very fragile sliding lens within lens assembly.

You are lucky the used prices of your model have come down greatly the last year,take a peek on ebay,I barely sold 2 M3's 2 months ago for 120$ free shipping each in great working condition

This guy on youtube is very thorough in repairs for the RX line,here is one for your model,look close at the opened lens in the video,you will see the sliding "flap/lens" inside on 2 rails,it has a small ribbon cable held in place by a flimsy piece of metal which pops loose some times and jams the internals,other times something cracks or breaks.
https://youtu.be/wo3bfd3ffmQ

Again ,I had repaired enough of these models to know what is best to do for specific errors,in your case sell it "as is" and buy a warrantied used one on the bay.

Im sorry this chat has wandered very far from "coding" as this site is for?? and going in the direction of mechanical repair,I personally wish there was a way to turn off EVERY error code which locks up the RX line but this happens to all Sony's since the 90's,what would be best is a simple user interfaced app with a list as to what to turn off or on but that is a big undertaking for a factory-only accessible operating system,we are lucky to have had this thread to reset the most common non-hardware error.

ronenliberman, 2023-08-04:
Hi 
 I got error E:91:01 on my RX100 M6. honestly, I don't remember what I did before I received this error, but I don't think it was unordinary. 
 Can you please help me understand what to do to fix it? I tried resetting to the default setting but it didn't work. 

Thank you

Justin2187, 2023-09-17:
Hi there, 
I had the e:91:01 error flashing on my Sony PJ590v handycam. But whenever I tried to run pmca-console-v0.18-win.exe it force quited for no reason. The only thing that I could open was the gui.exe but that didn't solve the problem. Plus this model does not have wifi capabilites, so I don't know what to do. Can someone please assist me? I would really appreciate it! :)

Tetsuya-Nishida, 2023-11-08:
Does someone let me know how to reset E91 01 error code on RX100 m1??
As you know m1 cannot install openmemory App....

akleckas, 2024-02-22:
Has anyone tried to fix this error according to the above instructions on the RX100 M5A or newer camera? Have you guys @magickone or @ronenliberman tried this on your cameras?

Udeson, 2024-03-29:
@peppersass do you remember where you bought the capacitor replacement? Cannot find them...

How can I start examine which part I need to replace? In my understanding the E:91:01 Problem can be caused by broken capacitor, flash unit, flex board (connecting flash unit and capacitor with mainboard) or Mainboard itself. Where and how should I start?

Overwrting the errorcode is working but does not fix the problem since the Error comes back after a few minutes...

Thanks in advance for any reply and support

Udeson, 2024-03-29:
> Hi there, I had the e:91:01 error flashing on my Sony PJ590v handycam. But whenever I tried to run pmca-console-v0.18-win.exe it force quited for no reason. The only thing that I could open was the [...]

You have to run the .exe from a commandlinen/shell

jat-79, 2024-07-06:
Hi! Flash error reset works on my RX100 IV, but apparently I have also hardware issues, because its comming back when I try to use the flash. But this thing that I could now reset the errorcode on my own is very usefull, it allows me to fix this without the Sony. I have tried every factory reset things I found on the internet with no luck, but this thing worked and removed the annoying blinking E:91:01 message! Thank you very much!!!

The instructions worked well otherwise, but I couldn't copy the backup files to my sdcard because "no such directory" error. When I go to /android/mnt and type "ls" I see the sdcard directory, but when i type "cd /sdcard" or "cd /android/mnt/sdcard" it just says can not cd to there. Is this some permission issue?

furkanbasersk8, 2024-07-06:
I have Sony Rx100 M5, i don't have a error like **E91 or  any error code**, guessing it's depending hardware problem, how can i solve my rx100m5 flash problem especially **RX100 M5** Thanks in advance.

RX100V, 2024-10-07:
Hello thanks a lot it's worked well for me. To people with a RX100M5 android changed the directory for the sd card : 
So the command are : 
cp backup.bin /android/storage/sdcard0/backup.bin
cp backup.bak /android/storage/sdcard0/backup.bak

jat-79, 2024-10-08:
My RX100 IV Hardware problem were in the flash driver circuit. I change the ST-1032 flexboard, then reset the errorcode, and flash is now working. Thank you!

ronenliberman, 2024-10-08:
@jat-79 can you please elaborate on how you've changed the flexboard? is it possible to do it without experience or better to ask a lab?  and then how did you reset the errorcode, was it directly from the camera main menu screen?

jat-79, 2024-10-08:
@ronenliberman you have to open up the camera, then take the lens out, and then the view finder.
Speaker, flash capacitor, and NFC-antenna need to be desoldered from the old one and soldered to the new one.
There could be 200-300Volts in the flash cap, you will get a shock if you touch the wrong place. It could be discharged with the 1kohm/1W resistor. 
Reset is a little bit tricky, it needs to be carried out as peppersass say in this post: https://github.com/ma1co/OpenMemories-Tweak/issues/145#issuecomment-280181255

I have vlog video from the reset procedure and the flexboard change, but only in finnish. Mine is MK IV, there are some differences between the models.
Language unlock and flash error reset: https://www.youtube.com/watch?v=7O_CPRWexqY
Flex board change: https://www.youtube.com/watch?v=CkvBDRrOrw8

Good videos from EEVblog about the RX100 MK IV tear down.
EEVblog 1428 - Sony RX100 IV Camera Repair and then WTF!: https://www.youtube.com/watch?v=x0_9GV0QU3A
EEVblog 1429 - Sony RX100 Camera Troubleshooting Part 2: https://www.youtube.com/watch?v=KwWWyOs4KHI

Altimator, 2024-11-09:
Thank you @peppersass! I have a DSC-HX90V that I dropped in a stream and this worked to get rid of the E:91:01 error. I was pleasantly surprised to find the flag and string located at the same addresses as in your RX-100.

It turns out the flash is indeed broken, but the error does not return so long as I keep the flash disabled in the settings. I rarely used it anyway.

madliszka, 2025-01-08:
Hi,
as I have sony DSC-rx100m5a I'm not able to use adb. But seeviceshell is avalible. In serviceshell there are bk w/bk r/bk patch and bk s avalible. Also there is an access to camera storage (so I was able to pull from camera to PC Backup Files). 

I was trying to change the adresses as @peppersass wrote, but for bk w i have following error:
senser adjust control error 129

So I have tried to use bk patch instead with success. Then I have used bk r to read the data, and it seems that it has been changed. But after reboot the flags are set again. 

Do you have any clue what also can I do there? 

This error 127 seems like the shell do not have a permission to write a data to memory. 
BK unlock/lock does not help.

thanks for any clue!

wen-templari, 2025-03-01:
Thank you @peppersass !! I have just removed the `E:91:01` error on my RX1R following your guide. 

A small difference is I am using updatershell, I need to write full 8 digit value or there will be an error: `Error: Unknown command`.
```
> bk w 000e0004 00
Error: Unknown error

>bk w 000e0004 00 00 00 00 00 00 00 00
Success
```

heyjxc413, 2025-04-17:
> Thank you [@peppersass](https://github.com/peppersass) !! I have just removed the `E:91:01` error on my RX1R following your guide.
>
> A small difference is I am using updatershell, I need to write [...]
Hi May I ask how you use updatershell on RX1R? I meet the same question.

wen-templari, 2025-04-18:
> Hi May I ask how you use updatershell on RX1R? I meet the same question.

I followed the instruction here https://github.com/ma1co/Sony-PMCA-RE?tab=readme-ov-file#firmware-updater

heyjxc413, 2025-04-23:
> I followed the instruction here https://github.com/ma1co/Sony-PMCA-RE?tab=readme-ov-file#firmware-updater

Ok I fix the problem, Thanksssssss!

snyggis72, 2026-10-06:
Hello
Can anyone explain how the command for making backup.bin and backup.bak would be on an Win 11 pc and RX100 IV?
This does not work for me:
/# cd /setting
(changes current directory to /setting)
/# cp backup.bin /android/mnt/sdcard/backup.bin
(saves a copy of backup.bin on the sdcard -- just in case)
/# cp backup.bak /android/mnt/sdcard/backup.bak
(saves a copy of backup.bak on the sdcard -- just in case)

My SD card after formatting is called ( i have 2 partitions when connected to my computer) USB-stasjon and PMHOME
Sub under USB-stasjon: DCIM, MP_ROOT and PRIVATE
Sub under PMHOME: INFO and LICENSE

PS i have made the other commands just fine and E:91:01 disappeared but there is a hardware fault and it is showing up again.
But i really would like to make a backup of firmware.