ma1co/OpenMemories-Tweak #147: A6000 - File corrupted after 3h 28m 33s

Taylor0001, 2017-01-11:
I have a dummy battery for my A6000. When recording XAVC-S to a 128 GB SD card the REC counter stops at 3h 28m 33s. After manually stopping the recording many of the buttons no longer functioned, including the playback button (until the camera was restarted). I waited 30 minutes before powering off the camera and waited another 10 minutes before removing the SD card, but the video file was corrupted. I notice with the dummy battery the battery meter still counts down until it indicates the battery is almost dead, even though it's receiving direct power. At this time I am unable to record longer than 3h 28m 33s and any attempt to do so results in a total loss of recorded video.

gamerpaddy, 2017-01-12:
Resolution / Mbps?
Filesize of the corrupted file?
Dummy battery model? (does it just passes trough 7.4v or does it have internal controller)
What power source?

this tweak app just changes 3 numbers in a config*:
0 hours, 29 minutes 50 seconds
to
13 hours, 1 minute, 0 seconds
not more, not less.

*(except on 4k, there its just one value, 300 seconds to 32767 seconds)

Taylor0001, 2017-01-12:
Thanks for replying, your questions are answered in order below:

1080p60 50 Mbps.
So far the filesizes have been between 88 GB and 88.5 GB.
The dummy battery was plugged into an outlet, this is the one I bought: https://www.bhphotovideo.com/c/product/927355-REG/power2000_ac_fw50_dc_coupler_power_supply_for.html

Taylor0001, 2017-01-15:
I did some more testing and this issue only occurs when recording XAVC-S video at 60p. In my tests, when recording XAVC-S at 30p or 24p, or AVCHD at 60p, the camera records until the card is full. I also tried using a different SD card with the same result. I shoot sports events at 60p and I would really prefer to shoot XAVC-S, if there is any way to correct this issue I would greatly appreciate it.