ma1co/OpenMemories-Tweak #148: Sony a6000 temperature warning. How long can you record for.

ntbCreative, 2017-01-13:
Hello there.
I was wondering if you can let me know (if possible) how long I can shoot video on the a6000 without it damaging or blowing up my camera..lol
I did a test and shot 37min and the temperature indicator came on,  so I stopped recording.
The maximum time I need to shoot is 3hrs, but normally around 1hr.
I use the camera to record my band performances and want to be able to record full sets as I am on the drums and can't operate the camera.
Thank you for your input.
NB

Taylor0001, 2017-01-14:
I typically ignore the warning, if the camera gets too hot it shuts itself down. The best way to mitigate the effects of extended use is to avoid direct sunlight, pull the screen away from the body, and use a fan to cool the back of the camera. When recording indoors at a reasonable temperature with a fully charged battery I've managed to record a clip 1h 37m in length before the battery died.

ntbCreative, 2017-01-14:
Thank you very much for the info, Taylor0001.
Cheers
NB

dxnacho, 2017-05-17:
I couldn't get more than 23:23 with the hack and everything else done. Am I missing someting?

lluisgerard, 2017-05-18:
@dxnacho I have a 6500 but, here we go: 
- Be sure to format the SD card using the camera software itself (or format with exFat instead of FAT32 thus allowing files greater than 4GB to be saved).
- Check with different video formats (in my case, mp4 format doesn't work but 4K, XVAC and ACHD seem to work) Might be related  https://github.com/ma1co/OpenMemories-Tweak/issues/150

dxnacho, 2017-05-18:
I did change format of the SD Card, but didn't consider the video format. I'll try and let you know, many thanks!!!

closed 2018-10-26 by ma1co