ma1co/OpenMemories-Tweak #150: Recording limit @ 23:25 on MP4 setting

bloodmage0, 2017-01-18:
Something I came across was that the limit for MP4 recording was greatly reduced with this tweak. Despite having the record limit on or off, it still managed to end at about 23:25 mins. 

I tested this on 1080 but did not try VGA.

XVAC and ACHD seems to work as intended for the tweak.

Thanks!

Gabby16bit, 2017-02-19:
Yeah, the same happens on my Sony A7r II. Ntsc / Pal. On mp4 it ends at 22:32, so is greatly reduced.
Tested on MP4 1080 60 fps NTSC and the 50fps on PAL.

Other formats seems to work fine.
It's a shame because mp4 is the format i use the most for what i have to do with the camera  right now.
So please @ma1co, if you can do something about that, would be great!
Thanks.

@bloodmage0  Have you tried to uninstall the app? It still stops early?

Sam8726, 2017-04-14:
My a6000 stops at 23min. I uninstalled the app and re-installed again, but same result.

dxnacho, 2017-05-18:
Same problem here.

cws127, 2017-06-14:
We've just discovered the same issue on our A7Rii's (have 3 of them).  I normally record in 4K or 1080p XAVC s so never came across the problem, but my work colleagues started shooting .mov files and found they were hitting the 4GB file limit and the camera stops.  In AVCHD with the 2Gb limit the camera keeps recording creating new contiguous files, but the .mov stops once it hits 4Gb.  
In 2 tests we carried out (30p and 25p) that was around 35min & 8 sec in each test.
30p was in the Sony v3.3 software, 25p in the v4.0 software.

@ma1co is this anything you might be able to exert some control over... getting the camera to roll contiguous files as other .mov cameras do (eg GoPro, Zoom Q4...) ??

cheers, and thanks for this wonderful app!!