ma1co/OpenMemories-Tweak #153: Paypal donation option

JohnONolan, 2017-01-31:
You know what sucks about Github issues? You pretty much only ever hear from people who have problems, while everyone who uses your code successfully remains silent and anonymous. So I just wanted to stop by and say: Thank you, from the bottom of my heart, for making this. The NTSC warning on every startup was *actually* killing me, and this has completely solved it! <3

Please consider adding a paypal link (or bitpay or whatever) somewhere in your readme. I would love to send a small token of thanks for writing something which I would happily have paid for in the first place.

alexnesher, 2017-02-08:
I agree! It would be really great ) thanl you for your work!

mistermenear, 2017-06-06:
I would be happy to donate something to the cause as well. I will start with my sincere appreciation of your dedication to the Sony cameras and their flaws. 
Thanks ma1co!

dxsaigon, 2017-06-10:
Me too, thanks so much, I've been stuck with a Japanese camera for two years and finally today I'm able to use it in English (their menu is actually only slightly more usable in English, damn you Sony UX  department!) but I'm still happy to finally be able to parse it quickly instead of looking up Japanese. 

I used pmca-gui v0.11 with an A6000, no problems at all. Though it did take me a while to find out where in the menu the apps were located and again, where the language preferences were. But after a few youtube searches, done. So happy. Thank you!

ma1co, 2020-02-08:
Better late than never: I don't need any money myself, but as a trial, I have installed the following page where you can donate to charity instead. Please consider it, I get notifications about the donations and it greatly boosts my motivation! https://www.globalgiving.org/fundraisers/ma1co/

Zofware, 2020-02-08:
Good idea and done! Thanks for lifting the 30 minute record limit. It's allowed me to record my daughter's soccer games without breaks in the middle.

ma1co, 2020-02-10:
@Zofware Wow, thank you for the generous donation!