ma1co/OpenMemories-Tweak #154: Sony rx1r (not rx1rm2) japan edition language change

electricsharky, 2017-02-12:
Hi! I just wanted to know if on an rx1r bought in Japan can be changed the language? Just because while the rx1rm2 is listed among the cameras the previous model rx1r is not listed.

Thabk you!

ma1co, 2017-10-30:
@electricsharky 
The RX1R does not support Android apps, so in the past, this was impossible.

But there is a new experimental way to tweak settings on older cameras using the firmware update functionality. This should also work on the RX1R (after some more development). If you own such a camera and want to help with the development, please let us know.

closed 2017-10-30 by ma1co

sporre, 2018-02-28:
Hello, I have the same problem here. Is there already a code which can help me change the language?