ma1co/OpenMemories-Tweak #155: I have opened all the languages, but want to revert to the original language menu

a997476356, 2017-02-14:
My English is not very good. This is using Bing translator.
  My Sony camera a6000 is the Chinese version, I disable protect unlock all 35 languages. Then found does not provide an option to restore only 11 languages. I camera warranty, so I'm afraid to go after the warranty time will say that they cannot guarantee the staff see the language menu.
  Could you provide an option to restore the default, I tried to restore factory settings to undo this change, but could not do so. Because my firmware is latest 3.2, so I use the Sony software updates I no when prompted to upgrade, I cannot Flash the firmware again.
  And then I try to use https://github.com/ma1co/Sony-PMCA-RE, a tool that is "PMCA-console-v0.10-win.exe firmware-f FirmwareData_ILCE6000V320.dat" command, but it is wrong. Error message:
 
E:\BaiduYunDownload\sonyapp>pmca-console-v0.10-win firmware -f FirmwareData_ILCE6000V320.dat
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "<string>", line 67, in <module>
  File "<string>", line 61, in main
  File "pmca\commands\usb.py", line 254, in firmwareUpdateCommand
  File "pmca\usb\sony.py", line 319, in checkGuard
  File "pmca\usb\sony.py", line 296, in _sendWriteCommands
Exception: Firmware update error: Invalid version
pmca-console returned -1
[image]

Thanks.

a997476356, 2017-02-14:
使用你的Sony-PMCA-RE，mtp模式和大容量存储模式都试过了也无法更新。

ma1co, 2017-02-18:
Can you please post the TWEAKLOG.TXT from your SD card?

a997476356, 2017-02-18:
Ok, already uploaded.

[TWEAKLOG.TXT](https://github.com/ma1co/OpenMemories-Tweak/files/785180/TWEAKLOG.TXT)

ma1co, 2017-02-18:
You're using an outdated version. Please install the newest version of the app.

closed 2017-02-18 by ma1co

a997476356, 2017-02-18:
Oh, my god. My previous OpenMemories-Tweak was installed at 

> https://sony-pmca.appspot.com/plugin/app/com.github.ma1co.openmemories.tweak

 I thought from sony-pmca.appspot.com installation is the latest it Now I downloaded and installed [OpenMemories-Tweak-release-0.9.apk](https://github.com/ma1co/OpenMemories-Tweak/releases/download/v0.9/OpenMemories-Tweak-release-0.9.apk) The problem has been resolved. Please update the sony-pmca.appspot.com above version, to avoid other people encounter the same problem with me.

ma1co, 2017-02-18:
sony-pmca.appspot.com should install the newest version (0.9). Please try again and let me know if the problem persists.

a997476356, 2017-02-18:
Uh, now I tried again, did not appear this problem. I should be wrong, I downloaded another OpenMemories-Tweak through sony-pmca.appspot.com installed, not check the version number. I'm going to say sorry to you.

这锅我背(The word machine is translated and there is a problem.)

(#°Д°)

ma1co, 2017-02-18:
No worries. I'm glad you got it to work.

a997476356, 2017-02-18:
Thank you so much.