ma1co/OpenMemories-Tweak #157: [REQUEST] Disable display dimming during 4K video mode for a6300 a6500 A7RII

mercier999, 2017-02-25:
On the a6300, a6500, A7RII the screen darkens significantly whenever recording 4K video, even with the monitor set to "Sunny Weather".
I am sure you could disable this, and it would be great news.

pixlwave, 2017-03-06:
I believe this limitation is due to the battery not being able to supply enough power (whether or not that means it can be bypassed I don't know, just thought I'd share).

mercier999, 2017-03-07:
@pixlwave Even if we could shoot a video clip up to 60-80 seconds in 4K with a bright screen, that would be great. We must have an option to choose when needed.

mercier999, 2018-02-07:
Any news on solving the screen brightness problem when 4k recording?

altermind81, 2018-05-18:
Agreed... this would be a great boon :)

mtalpine, 2020-05-08:
I've been using your wonderful hack for a while now! Thanks so much for everything. I am running into issues with the screen dimming as well on my a6300. If there is any way to add this enhancement it would truly make this camera soooo much better. Any news on this possibility?