ma1co/OpenMemories-Tweak #159: ADB connection unreliable

jbieler, 2017-02-27:
Thank Ma1co for adding the ads connection to the open memories tweak. On my A6000 it's very unstable. For example, if I add a break point and want to step through some code, it crashes after a few steps. Is this problem common? Can you think of a fix for that? Maybe I'm not following the correct debugging process. What are the recommended steps to use the debugger?

ma1co, 2017-02-27:
Just to make sure... Did you set the auto power off time of your camera to a high value (30min for example)?

jbieler, 2017-02-28:
Yes I did that. What else is needed for the setup?

ma1co, 2017-02-28:
What do you mean by crashing? Does the camera turn off? I sometimes get a black screen, but the camera stays connected over adb...

jbieler, 2017-02-28:
The debugger gets disattached of the code and the App on the camera continues to run. The screen is not turning dark and the app will continue to run.

ma1co, 2017-03-02:
Does the adb connection terminate? Or can you still see your camera in Android Studio?

jbieler, 2017-03-02:
The connection does terminate. I have to run "adb connect" again.

ma1co, 2017-03-02:
But you don't have to restart adbd on the camera?

Are you sure this isn't just a wifi issue? Can you run an "eternal" ping in the background to see if your camera disconnects from the network?

closed 2018-10-26 by ma1co