ma1co/OpenMemories-Tweak #160: Dead RX10ii

Cookiemovies, 2017-03-05:
Hi, first everything went well, I could use the installer and install the tweak app. Established wifi connection, enabled telnet and I could enter commands from my laptop telnet session. Unfortunately, one time I forgot to close down the wifi connection and switched the camera off. 

I couldn´t switch it on again!
The red light is now always on and nothing happens, black screen and no activity.
When removing battery, the red light goes off and goes immediately on, when the battery is in. Same with microUSB connected to the laptop. Light is on, but no activity and the camera is not recognized.

Any idea of a hard reset or replacing the firmware without any camera activity?
I think, that the firmware was deleted or destroyed, wondering how this can happen.

In the telnet session you can´t delete anything regarding android itself. Otherwise the Sony guys did a bad job.

I´m then about to send the camera to a repair center, wondering whether they can do anything.

Thanks for any answers!

ma1co, 2017-03-07:
I don't think not disabling wifi is the reason for your problem.

What commands have you run in the telnet session? You're running a root session, it's pretty easy to destroy your firmware like that if you don't know what you're doing...

I'm not aware of any easy way to replace your firmware if your camera doesn't connect over USB anymore.

Cookiemovies, 2017-03-07:
Thanks for your reply, I don´t think either, that this can destroy the firmware. 

I was mostly navigating through the directories using ls, cd etc. I´ve also tried to copy with cp to the sd card, but that was denied by the system. So I suppose I was not using a root session, I usually know how to use a Linux bash in a telnet session.

I suppose a repair centre does have the ability to flash the firmware without turning the camera on, as otherwise, all devices, not only cameras will be dead when deleting firmware. Usually there is flash memory to hold the firmware.

denisrus, 2017-03-11:
There was something like this with my A6000 when i installed some Android app... i just left it on for some minutes at it turned on again.

closed 2018-10-26 by ma1co