ma1co/OpenMemories-Tweak #163: Sony A6000 problem with connecting

donmarkon, 2017-03-26:
I followed all steps disabled nod32, drivers are installed properly also but i'm constantly getting message like this http://i.imgur.com/okVoky9.png 
Loading plugin

Plugin loaded

Starting task

Got task key

Cancelling plugin

Plugin cancelled

Starting download to camera

Progress: Uploading 0%

Error: Network connection

I'm under Windows 10 so any help would be great! i also did try under another PC and there i'm getting SSL error ... Thank you in advance :)

donmarkon, 2017-03-29:
Any help please ?

Unexisted, 2017-03-29:
I have the same problem with my a6300! And I need help too! 

Original USB cable, MTP mode on camera, IE 11 browser, Windows 7. But every time I getting message "Network error". 

Help please!

donmarkon, 2017-03-30:
It seems that this is death place :(

ma1co, 2017-03-30:
First of all, does the official [Sony app store](https://www.playmemoriescameraapps.com/) work for you? If it doesn't, there's a problem with Sony's software and I can't help you.

Second, make sure you're using IE. And disable your antivirus software if it messes with SSL connections.

Third, there is an alternative installer you can try: [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases/latest)

donmarkon, 2017-03-30:
Yep app store works normal. I'm using IE (on windows 10) i disabled nod 32 ...

Unexisted, 2017-03-30:
Yeah! Third way (GUI installer) worked perfectly on first try! Thank you so much, ma1co! 👍🏻

closed 2017-03-30 by ma1co