ma1co/OpenMemories-Tweak #166: [a6000] Record limit hack never works for me

Sam8726, 2017-04-14:
Hello
I really needed this function and was about to buy Panasonic model which doesn't have recording limit. So, I was so excited but disappointed soon. I tried almost 10 times (more than 5 hours), changing 3 batteries. But, all the recordings stop at exactly same time 23min 2-30sec and file size 2.3Gb (no exception). If I check tweak app video recording limit is set to 13h 01m. Languge change works well after disabling protection.
I will try any advice, pleas help me. Thanks developer.

Test environment:

-ILCE-6000 (a6000)
-Tweak v 0.10
-Firmware 3.2/3.1 (didn't work on both)
-Tweak installed by installer on Mac OS Sierra
-Sandisk SD class10 32Gb (formatted both FAT and exFAT, both failed)
- Recording setting: MP4, 12M mode
-19mm sigma art lens
-Mode: M 
-iso 400 / S 160 / F7.1
-Cool temperature : 11 F
-Battery full when start

lh8511, 2017-04-14:
I would suggest you to try with a different sd card...
Also do FORMAT card from camera itself.

Sam8726, 2017-04-15:
@lh8511 Thank you for advice. I tried Sandisk 16gb, but failed, and formatted exFAT and FAT both) Thanks again.
I think hack works fine, because AVCHD works fine, it records more than 1 hour. Only mp4 has a limit. And I also found out Panasonic non-EU models work same way(30min limit in mp4) 
I don't know why, but mp4 doesn't support more than 30min limit with camera letting alone camcorder. I am happy with hack, thank you!!

closed 2017-04-15 by Sam8726