ma1co/OpenMemories-Tweak #167: Sony Alpha NEX-5.

SeLaY27, 2017-04-16:
Здравствуйте. Для первой версии nex-5 будет обновление?
Hello. For the first version of the nex-5 will be an update?

SeLaY27, 2017-04-24:
?

ma1co, 2017-04-25:
Not for now. See the [FAQs](https://github.com/ma1co/OpenMemories-Tweak#i-have-an-old-camera-that-doesnt-support-apps).

closed 2017-04-25 by ma1co

ma1co, 2017-10-30:
@SeLaY27 
There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on a NEX-5, so your mileage may vary. If you attempt anything, please let us know!

SeLaY27, 2017-10-31:
Model: NEX-5
Product code: 0032820425
Serial number: 04282402
Firmware version: 5.00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to app install mode
Traceback (most recent call last):
File "C:\projects\sony-pmca-re\pmca-gui.py", line 74, in do
File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 238, in installCommand
File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 34, in switchToAppInstaller
File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 306, in switchToAppInstaller
File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 244, in _sendCommand
File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 67, in sendSonyExtCommand
File "C:\projects\sony-pmca-re\pmca\usb\__init__.py", line 33, in _checkResponse
pmca.usb.InvalidCommandException: Mass storage error: Sense 0x5 0x20 0x0

ma1co, 2017-10-31:
Please read the section I linked to:

> In pmca-gui, go to the _Tweaks_ tab and click the _Start tweaking_ button.

SeLaY27, 2017-11-01:
what it gives me. the camera throws off everything to the factory settings. Clip 29.50 more minutes to write. Sorry, maybe I don't understand the Complexity of the translation.

>
>Please read the section I linked to:
>>In pmca-gui, go to the  Tweaks tab and click the  Start tweaking button.

С уважением,
Сергей Селезнёв
selay27@mail.ru

ma1co, 2017-11-09:
Sorry, my bad. The NEX-5 is an old camera. We can run custom code, but we don't know how to change settings. So you can't modify the video recording limit for now. Sorry.