ma1co/OpenMemories-Tweak #169: [Request] Disabling long exposure noise reduction

stuomas, 2017-04-18:
Since BetterManual (https://github.com/obs1dium/BetterManual) app can already disable this forced functionality on A5100, maybe it is possible to add as a general tweak? I'm not sure how it works in the app. This way the official timelapse app would be usable on A5100.

JulioCVM, 2017-08-13:
Any achievement in Long Exposure Noise Reduction global deactivation?

MSargeNZ, 2017-10-31:
Yes please for this feature!

stuomas, 2017-10-31:
Have you @ma1co (or somebody else) tried to compare the Backup.bin of A5100 and A6000? They have very similar hardware and I believe very similar firmware too. Functionally the two cameras have very limited amount of differences, and that information could be used to narrow down some settings in the file? For example this long exposure noise reduction is an option on the A6000, and I am 100% sure it is artificially hidden by Sony on the A5100.

stuomas, 2017-11-12:
Just wanted to share some happy news, the property has been found, and it is now possible to turn off long exposure noise reduction globally in A5100. All you need is OpenMemories-Tweak to enable telnet, and issue one command through it. 

More in here:
https://github.com/ma1co/OpenMemories-Tweak/issues/205

sspieser, 2018-09-06:
does that work on A5000 also ?

stuomas, 2018-09-06:
@sspieser I have not tried or heard confirmation from anyone, but I'd guess yes. You can start by trying to read that property 0x01070108 and see if it exists with value 01.

sspieser, 2018-09-06:
@stuomas so far I have not tried to "hack" my A5000 with OpenMemories Tweak or any other ;-)
OK I'll take some time to understand the whole process, risks, and maybe try :-)

sspieser, 2018-09-06:
@stuomas It works! I issued "`bk w 01070108 00`" through pcma-console updatershell !

stuomas, 2018-09-07:
@sspieser Great! Thanks for confirmation, there has been many asking if it works on A5000 also.

@pynkstyle Yes it could, I will look into it. But meanwhile, it is pretty simple to do without also.

mariandomanik, 2018-11-15:
Thank you guys, I just want to confirm 0x01070108 set to 00 also disabled NR on my a5100, firmware 3.10.

albertlopes, 2019-10-24:
@stuomas hello, can you help me. I'm trying to send this command trough shell but any time I try to use the "updatershell" in my camera appears the prompt "reset device". I'm using the a5100 firmware 3.10.

I have enabled telnet and pluged trough usb, am I missing something?

stuomas, 2019-10-25:
Hi @albertlopes, I can't help with updatershell, I have only tried via telnet. Enable wifi and telnet from OpenMemories-Tweak app in your camera, connect the camera and PC to same wifi, open telnet connection from PC to camera, and command `bk.elf w 01070108 00`

albertlopes, 2019-10-25:
@stuomas hello, thank you, tried with telnet and worked.

Colonist21212, 2020-02-04:
Hello,

Does the telnet method make it a permanent change? I mean will it still be disabled when you power off/on the camera?
And will it stay disabled also when using Imaging Edge?

BR,
Joachim

albertlopes, 2020-02-04:
Yes, it is permanent and will stay after reboot. Only loose if you update
the firmware

Colonist21212, 2020-02-05:
Thanks for the quick reply. Great!

glundby, 2020-08-15:
Just did in telnet bk w 01070108 00 But testing long exposure I did not get any change. Still process same amount of time that shutter was open. (a5000) what was original value? should I set it back? Any idea as to what to do? Desperately need the hack for my astro modified a5000.

glundby, 2020-08-15:
bk.

glundby, 2020-08-15:
GOOD! I think i found out what's missing in the procedure above:
bk.elf l  01070108         "List current value in backup
bk.elf w 01070108 00    "Writes 00 to this setting in backup
bk.elf s  01070108         "Syncs the value from backup to live.  And then after shut off and on IT WORKS!

ZbynekJirka, 2022-12-09:
THX........ready for a5100