ma1co/OpenMemories-Tweak #173: [ReQuest] Enable Sunny Weather Monitor Brightness in Video Mode

kabauterman, 2017-04-26:
Hello ma1co,

thanks for your great work! I use it on my A7RII and RX100IV and I love it.

I have a request, maybe a simple one. I shoot a lot of video and it pisses me of that I can't choose the sunny weather monitor brightness in Video Mode. I think sony just turned it off to save battery but I think it would be way easier to see what I shoot.

Maybe this is not a big deal for you to bring it to OpenMemories-Tweak, but it will be a big deal for every film maker. :)

swedroid, 2017-07-31:
Yes please! I just bought the A6500 and it's pretty much useless for outside video. I don't get why Sony disabled the sunny weather mode. Just put up a warning telling the user to flip out the screen, so that it doesn't touch the back of the camera. 

It would be amazing if it was possible to enable sunny weather mode :)

loverhd, 2017-08-05:
Yes please! 
My A7R II also suffer from this issue. and the issue is really critical: nearly impossible to shoot 4k video on bright light.

It would be amazing if it was possible to enable sunny weather mode :)

sonyfan, 2017-11-20:
Sony A6500 when shooting video, the lcd screen goes so dark that i can barely see it in outdoor light.  Absolutely killing the video features of this great camera. Please, help!