ma1co/OpenMemories-Tweak #174: Disable 3.2 sec+ and bulb filtering.

ScorpionCarcass, 2017-04-28:
There are a bunch of us that are trying to disable star and detail filtering with the 3.3 firmware for the Sony A7rii, A7, A7ii, A7s, A7sii (Probably others as well). Is there a way to do that through tweak?

Here is a link to the issue and also pictures showing the issue:

http://community.sony.com/t5/Alpha-SLT-DSLR-Cameras/star-eater-algorithm-in-Bulb-mode-shots-please-offer-alternative/td-p/619350

Sony is not responding anyone on this but the problem is obvious.

cws127, 2017-06-14:
Sony released v4.0 a couple of days ago, apparently this does help but is not a proper fix.  Best technical thread I've found so far on it...
https://www.dpreview.com/forums/thread/4167521#forum-post-59681303

azaitsev, 2017-06-14:
It's not a fix:
http://www.fredmiranda.com/forum/topic/1494820/0#14070527

cws127, 2017-06-14:
lot of "noise" in those threads, but basically says they've modded the green channel only.
Noticed a comment on another post for A5100 camera for "BetterManual" that  "Long exposure noise reduction is disabled (normally forced in single shot mode for exposures >= 1" and in bulb mode on the A5100)."
To ask a stupid question... anything in the code (I'm not a programmer) that could be used on the A7 cameras to get rid of the "star eater" issue?

astrobuoy, 2017-07-31:
Please sign the petition to get Sony to fix this http://www.change.org/p/sony-remove-the-star-eater-on-sony-a7s-r-mk-i-ii-and-a9-cameras

mungewell, 2017-12-27:
Has anyone identified the area of firmware which actually implements the 'spacial filter'? Is this done within the (normal) firmware, or embedded on the sensor itself?

Poking around the code for the 'Smart Remote' on camera app (which seems to be a complete re-implementation of the menu-ing and controls), there is mention of two different raw modes. Are both 'compressed' and 'uncompressed' raw afflicted by the 'star eater'?

In "com.sony.imaging.app.srctrl-1_source_from_JADX/com/sony/imaging/app/base/shooting/camera/RawRecModeController.java"
```
public class RawRecModeController extends ShootingModeController {
    private static final String API_NAME_RAW_REC_MODE = "setRawRecMode";
    public static final String DEFAULT_SETTING = "compress";
    public static final List<String> KNOWN_SETTINGS = Arrays.asList(new String[]{DEFAULT_SETTING, "uncompress"});
    private static final String TAG = "RawCompressController";
    public static final String TAG_RAW_REC_MODE = "RawRecMode";
```

mungewell, 2017-12-28:
Did some poking around the firmware for the A5000; it appears that there are a couple of 'filters' within 'firmware.tar_unpacked/0700_part_image/dev/nflasha16_unpacked_unpacked/framework/sony.cameraex.odex' 
```
RawImgAdditionNRDevelopFilter
RawRawAdditionNRDevelopFilter
```

These appear to be from the source "CameraSequence.java", and extend 'BaseDevelopFilter'. I'm not an Android dev, so not 100% sure what these actually do.

There appears to be some conditional setting of 'path' within the 'setSource' function, which looks (to me) to be a little suspicious.
```
.method public setSource(ILcom/sony/scalar/hardware/CameraSequence$RawData;Z)V
    .registers 6
    .param p1, "index"    # I
    .param p2, "raw"    # Lcom/sony/scalar/hardware/CameraSequence$RawData;
    .param p3, "releaseSource"    # Z
```

Anyone more experienced want to take a look?

azaitsev, 2018-01-01:
>Are both 'compressed' and 'uncompressed' raw afflicted by the 'star eater'?

Yes, both

jkbsahner, 2025-02-12:
Hey Guys, do you know if there has been any progress in finding out how to fix the star eater issues via firmware?

Philippea78, 2025-02-12:
Hello, no sorry