ma1co/OpenMemories-Tweak #176: website down

Denny69, 2017-06-03:
https://sony-pmca.appspot.com/apps down for two days.
does the website reset the quota, because I was try to enter the web many times but it seem always over quota? Is there any way to install openmemories beside from the website? thank you.

rigrx, 2017-06-04:
There is a windows installer as well. But keep in mind it is beta.

azaitsev, 2017-06-08:
@rigrx could you give me a link to APK so i'll install it via local installer

azaitsev, 2017-06-08:
oh, stupid me
i found it in releases oh github

ma1co, 2017-06-08:
There was a lot of traffic during the past days and the site exceeded the free quota. However, the quota is reset every 24h. As an alternative, you can use pmca-gui, the newest version also works when the site is down: https://github.com/ma1co/Sony-PMCA-RE/releases/latest

closed 2017-06-08 by ma1co