ma1co/OpenMemories-Tweak #178: tweak not work on a6500 fw 1.03

zgmladen, 2017-06-08:
web app:

Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection
----------------------------------------------------------
pmca-gui:

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
Querying MTP device
Sony Corporation ILCE-6500 is a camera in app install mode
Using remote server sony-pmca.appspot.com
Starting task
Starting communication
Traceback (most recent call last):
  File "<string>", line 74, in do
  File "pmca\commands\usb.py", line 224, in installCommand
  File "pmca\commands\usb.py", line 70, in installApp
  File "pmca\installer\__init__.py", line 56, in install
Exception: Response error Result(code=300, message='URL is invalid')

ma1co, 2017-06-08:
@zgmladen Please try the newest build of pmca-gui and let me know if it works: https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts

zgmladen, 2017-06-08:
Thanks, it works. Installed and removed 30min limit. verified

ma1co, 2017-06-08:
Thank you for your feedback, the version was released as v0.11: https://github.com/ma1co/Sony-PMCA-RE/releases

closed 2017-06-08 by ma1co

amirsfairf, 2017-07-06:
didn't work for me. i had this msn 

Loading app list
Using local server
Found 7 apps
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
No devices found. Ensure your camera is connected.

ma1co, 2017-07-06:
@amirsfairf Which camera? Is it connected in MTP or mass storage mode?

amirsfairf, 2017-07-06:
a6500 FW 1.03 
MTP
I saw all the vídeos on youtube and did all methods but none work. I used several usb cable too.
I used pmca-gui-v0.10-win, v0.11, v0.12 and failed.
No devices found. Ensure your camera is connected.
at internet explorer : network problem
but in mass sotorage I can dowloud all my images. So, i have comunication pc and a6500.