ma1co/OpenMemories-Tweak #179: open memories-tweak Error: Network connection

nardydaido, 2017-06-08:
Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection

please help me 
error network connection

ELouisYoung, 2017-06-09:
Same problem for me.

ma1co, 2017-06-09:
@ELouisYoung Use [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases/tag/v0.11)

ELouisYoung, 2017-06-09:
[image]

antonpoll, 2017-06-11:
The same problem =\ a6500 v 1.03

antonpoll, 2017-06-11:
Solved by pmca-gui

ELouisYoung, 2017-06-11:
The Pmca gui for mac doesn't work.  @antonpoll did you use the windows version?

ma1co, 2017-06-11:
@ELouisYoung Please try pmca-gui v0.10 (older version). Does it give you the same error?

nardydaido, 2017-06-11:
you can give me step use the pmca-gui ? what the version ?
please help me
my Sony a7ii on 4.0 version
thanks

antonpoll, 2017-06-11:
@ELouisYoung  Windows 7

nardydaido, 2017-06-11:
i tried with pcma-gui 0.10 and doesnt work 
you can help me 

and i tried with pcma console 0.10  with cmd and dosnt work

nardydaido, 2017-06-11:
thank you @ma1co new version 0.11 solved for my a7ii  and with tweak o.10.apk
you have link tweak 0.11 .apk
thanks for reply

ma1co, 2017-06-14:
@ELouisYoung The issue should be fixed in v0.12

ELouisYoung, 2017-06-14:
thanks, i'll try today and report back.

kris35, 2017-06-14:
Can someone please explain this to me? i don't know how to code at all.

When i try to 'get camera info' it gives me the "network connection" error. 

i would be eternally gratefull.

ELouisYoung, 2017-06-14:
worked!

bazsikadam, 2017-07-02:
Hi All!
I have the same issue on an A7II please help. How can I install the app?
On IE the log display Network connection error. 

I tried all I could.

bazsikadam, 2017-07-02:
Ohh Yeah!
PMCA-GUI worked!!!!
Full Happyness!
https://github.com/ma1co/Sony-PMCA-RE/releases/tag/v0.12

daviddingnz, 2017-07-05:
Hi folks,

I have the same issue on Sony a6500 using Mac. (Error: Network connection)

I have tried all options including using pmca-gui

I'll be elated if anyone can help

daviddingnz, 2017-07-05:
I got it sorted! The file pmca-gui-v0.12-osx.dmg did the trick

chog117, 2017-07-07:
@ma1co  hey man sorry i'm very new at this. I've installed open memories on my a7sii via pmca-gui v0.12 and I have made sure that the locked settings are unlocked and left the pal/ntsc waring unchecked to keep it disabled but the annoying message is still there :(. I'm running the latest firmware on the camera. would that be to cause?

thanks

Nick

zeroxino, 2017-07-24:
Hello everyone. Im having an issue with installing open memories. I get a NETWORK ERROR on my a6300 ver. I believe 2.00. Weird thing is I was able to install it on another a6300 ver.2.00. I've read GUI mentioned. What do I need to do to be able to use GUI. 
daviddingnz: I got it sorted! The file pmca-gui-v0.12-osx.dmg did the trick

Please help. Tnanks

zeroxino, 2017-07-24:
sorry, I ment to type I was able to install on ver. 1.10

remainzzz, 2017-08-01:
Same issue here. A7rii version 4. 
using ie win 7.

after i start Get Camera Info or install Tweak
I get Error: Network connection

My camera changes to 
Application Download...

Tried pmca-console-v0.12-win.exe
console just opens and closes(?correct)

ma1co, 2017-08-01:
@remainzzz You don't seem to be using pmca-gui v0.12. Make sure you download the newest release.

remainzzz, 2017-08-01:
Quite right. Sorry to waste your time. I thought the web gui was 0.12
https://sony-pmca.appspot.com/plugin/app/com.github.ma1co.openmemories.tweak
But the pmca-gui v0.12 worked fine.
https://github.com/ma1co/Sony-PMCA-RE/releases/tag/v0.12

silvermoon23, 2017-08-19:
Just Use Safari Browser Dude. even in Windows safari browser can solved the problem

ma1co, 2017-10-30:
Closing this. Please open a new issue if anybody has more problems.

closed 2017-10-30 by ma1co

hazerto, 2021-08-01:
i'll try it