ma1co/OpenMemories-Tweak #184: a7sii firmware 3.0 issue

findamir, 2017-06-14:
I have not been successful for my win 10 x64 desktop and laptop to recognize the camera for this new firmware update. I put everything back to "normal" (factory default), uninstalled open memories and "initialized" the camera and still the same results.
Any suggestions?

ma1co, 2017-10-30:
@findamir 
Please use [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases/latest).

closed 2017-10-30 by ma1co

findamir, 2017-11-05:
Thank you for this, sir. 
I went through mcafee and windows 10 security and finally got to run as administrator. Finally go through Windows defender and it says "insufficient system resources exist to complete the requested service." and the app gets deleted. 
And I am also not sure if this is supposed to install so I can further connect the camera in order to get it working again.
Cany you please advise? Thank you!