ma1co/OpenMemories-Tweak #190: [Request] Higher mbit 1080p Recording for NEX-5R?

Joulutonttu, 2017-07-04:
Hello,

A while back I saw someone hack or presumably hack the bitrate for NEX-5N camera. Here is the article for that [http://www.eoshd.com/2012/08/sony-nex-5n-hack-is-go-46mbit-1080p-tested/](url) 

They stated that 46mbit 1080p would have been achieved. I wonder if this is still possible for other Sony cameras like the old NEX-5R?

ma1co, 2017-07-05:
Just for the record: There was never a working bitrate hack. "Someone1.00" found the decryption keys for the firmware updates (nobody knows how) and managed to reverse engineer the firmware file format. They then raised some money and disappeared.

Joulutonttu, 2017-07-05:
@ma1co Ah yes this is what I found, but I didn't know about the fundraising scam. I'm sorry to hear that.

ma1co, 2017-07-05:
I wouldn't accuse them of scamming, apparently some money was indeed used to buy cameras. But the whole story was way over-hyped. Obtaining the firmware is an important first step. But understanding it, and then extending it is a whole different story.