ma1co/OpenMemories-Tweak #191: [Request] Default Exif for Manual Lenses

pqhf5kd, 2017-07-06:
Currently, when shooting with a mania lens 'Sony 0mm F0.0' is recorded in the Exif.
Would it be possible for a tweak or app that could change that?

fotovideoguy, 2018-05-23:
i too am looking for a solution to this issue, which the manufacterer has letft out.
Please.

lh8511, 2018-06-06:
[Here](https://github.com/vlousada/LegacyLenses22)'s a walkarrond from @vlousada - #legacylenses
(should work with latest models that support playmemories)

fotovideoguy, 2018-06-07:
Thanks for this.
Now, how to install this App.

lh8511, 2018-06-07:
Hi
You must download the apk in that repository and then install it with ma1co/pmca-gui

fotovideoguy, 2018-06-07:
It took me all day to find it.
Not sure why.
I will install and test it

fotovideoguy, 2018-06-08:
installed successfully on my test NEX 5R.
the camera now does nothing when powered on after running the App.
screen shows image but not Menu or other controls from buttons.
as another poster mentioned with a different camera, leaving the camera in on mode, the camera eventually starts working again. tried running the App again, same result.
similar issue on A7R2. App is not working on this camera.

fotovideoguy, 2018-06-08:
[image]

on both cameras when installing it says Invalid APK file, then continues to install.

vlousada, 2018-06-10:
Hi @fotovideoguy ,
Yes, the apk is not signed but can be installed anyway :)

I developed this app innitially for nex6 and later adapted it for A6300.. I do not have nex6 (which nex5r is similar) anymore to test  against the latest version, but can confirm it works on A6300! 
It should work on A7x series too, but I did not test properly.
Your issues may due some wrong key buttons mapping or some procedure that it is NOT present at those cameras.  Please check at your SDCARDS if having a LOG.TXT and post it here, so I can try to find what is wrong...

1 final note: The app is intended only for PHOTO (not video) and RAW only..

Regards

fotovideoguy, 2018-06-11:
Hello Vitor,

These are the only log texts on the card, one for each camera.

hopefully it helps.

I only shoot Raw and seldomly do video, so it should not pose any issues.

Thank you for your time on this.

sam

[INFO][MainActivity] application start
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: ILCE-7RM2
[INFO][InfoAdapter] Backup region: CX79500_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 4.1.2 (SDK 16)
[INFO][InfoAdapter] Java API version: 2.14
[INFO][onPause] .InfoActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .ProtectionActivity
[INFO][writeProtection] setting protection unlock to true
[INFO][writeProtection] success
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .DeveloperActivity
[INFO][onPause] .DeveloperActivity
[INFO][onResume] .ProtectionActivity
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: ILCE-7RM2
[INFO][InfoAdapter] Backup region: CX79500_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 4.1.2 (SDK 16)
[INFO][InfoAdapter] Java API version: 2.14
[INFO][onPause] .InfoActivity
[INFO][MainActivity] application end

[INFO][MainActivity] application start
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: NEX-5R
[INFO][InfoAdapter] Backup region: CX75800_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 2.3.7 (SDK 10)
[INFO][InfoAdapter] Java API version: 1.1
[INFO][onPause] .InfoActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .ProtectionActivity
[INFO][writeProtection] setting protection unlock to true
[INFO][writeProtection] success
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: NEX-5R
[INFO][InfoAdapter] Backup region: CX75800_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 2.3.7 (SDK 10)
[INFO][InfoAdapter] Java API version: 1.1
[INFO][onPause] .InfoActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .ProtectionActivity
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .DeveloperActivity
[INFO][setWifiEnabled] setting wifi to true
[INFO][onPause] .DeveloperActivity
[INFO][onResume] .ProtectionActivity
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: NEX-5R
[INFO][InfoAdapter] Backup region: CX75800_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 2.3.7 (SDK 10)
[INFO][InfoAdapter] Java API version: 1.1
[INFO][onPause] .InfoActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .ProtectionActivity
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .DeveloperActivity
[INFO][onPause] .DeveloperActivity
[INFO][onResume] .ProtectionActivity
[INFO][onPause] .ProtectionActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: NEX-5R
[INFO][InfoAdapter] Backup region: CX75800_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 2.3.7 (SDK 10)
[INFO][InfoAdapter] Java API version: 1.1
[INFO][onPause] .InfoActivity
[INFO][MainActivity] application end

vlousada, 2018-06-11:
Hi Sam,

Those are NOT the logs was looking for... ;)
Try to locate a folder in yours sd cards named 'LLEGACY' and the 'LOG.TXT' inside.

Anyway just checked that I am not putting too much information from app itself into log, so will try to compile another .apk to properly debug errors in your cameras!

Vitor

fotovideoguy, 2018-06-11:
Thank you.
I fid look through all the folders on both cards from both cameras, but did not see the folder or the log text.
I have formatted the cards, run the apps in the hope it would create the folders and files but to no avail.
I will attempt this again with a more thurough search
Thank you.

Sam

From: Vitor
Sent: Monday, June 11, 5:32 PM
Subject: Re: [ma1co/OpenMemories-Tweak] [Request] Default Exif for Manual Lenses (#191)
To: ma1co/OpenMemories-Tweak
Cc: fotovideoguy, Mention

Hi Sam,
Those are NOT the logs was looking for... ;)
Try to locate a folder in yours sd cards named 'LLEGACY' and the 'LOG.TXT' inside.
Anyway just checked that I am not putting too much information from app itself into log, so will try to compile another .apk to properly debug errors in your cameras!
Vitor

vlousada, 2018-06-11:
Hi again Sam,
Just released new [apk](https://github.com/vlousada/LegacyLenses22/releases/download/0.94d/LegacyLensExif-debug_0.95.apk) version with some extended debug.

Please follow this instructions:
1) go to camera app menu and uninstall LegacyLenses app
2) install the new apk with pmca-gui
3) after install put the camera in "M" dial
4) de-attach any electronic lens or put a manual lens with simple adpater
5) run the LegacyLens application

Let me know the results :)

Note: just added info for nex-5R which was ot present at the app

regards
Vitor

fotovideoguy, 2018-06-12:
I have followed up on your LegacyLenses22 repository.

fotovideoguy, 2018-06-12:
I was not able to attach this to the repository when I commented.

Sam

fotovideoguy, 2018-06-17:
Still looking for how to attach my lens list for both the NEX 5R and now the A7R2, now that both cameras have this app installed

vlousada, 2018-06-24:
Added some debugging info in order to check why your lens list (profiles.xml) not loaded.
Check the most recent release in the repo LegacyLenses -> https://github.com/vlousada/LegacyLenses/releases