ma1co/OpenMemories-Tweak #195: Some problems with my ax40

funnykind, 2017-07-12:
hello
I got a japan only (J1) sony fdr-ax40,so you know the language problem upsets me a lot.
Knowing that  your app can not support my device without an app launcher,i tried the fwtool.exe.But the modified firmware didn't work at all.
My steps:
unpack the firmware,
swap the bin files in/0110_backup,
swap the second column values in/0111_backup_sum/backup file,
increase the version,
pack.
So is there anything i forgot to do?

ma1co, 2017-10-30:
@funnykind 
You don't have to patch firmware files anymore. There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on an AX40, so your mileage may vary. If you attempt anything, please let us know!

closed 2017-10-30 by ma1co

semitehsu, 2018-05-19:
I have the same machine.  Unfortuntly, I can open the language option but I can pagedown for 2 two page.
As soon as I do it, AX40 restart.  Do you have any idea about it?