ma1co/OpenMemories-Tweak #198: a5100 disable fake shutter sound function

antlvk, 2017-07-26:
Can add this "a5100 disable fake shutter sound" function. cause it sound so fake

mungewell, 2017-12-10:
On AS100 menu item "Beep" seems to be controlled via the key
```
id=0x01070026, size=0x0004, attr=0x40:
  01 00 00 00

'00 00 00 00' shutter is noisy, no chime on start-up
'01 00 00 00' shutter is noisy, chime on start-up
'02 00 00 00' shutter is silent, no chime on start-up
```

Gelisob, 2018-11-26:
Thanks for the hint, found 03 00 00 00 value at that ID on my A5100 (fw 3.10) and changed it to 02 00 00 00, dont seem to have the silly extra sound from speaker now, only mechanical that happens due to shutter flicking.

[image]

viksou, 2019-02-02:
Could you please let me know how to start the USB debug shell? Thanks.

Gelisob, 2019-02-02:
First line on the screenshot, you start it in cmd (command prompt). If these things confuse you, you might not want to mess around with it. Don't ruin your camera.

nodeOfCode, 2019-03-18:
> Thanks, Gelisob.
> Unfortunately my Windows 10 Home PC doesn't seem to be compatible with pmca-console-v0.17-win. The app opens but shuts down quickly.
> pmca-gui v0.17 is working fine.

It must be opened in command prompt. Otherwise it has nothing to execute and closes. I'll write up a guide below.

nodeOfCode, 2019-03-18:
1. Set camera USB Connection to MTP or Mass Storage. Connect to PC.
`Setup > USB Connection > MTP`

2. Open Command Prompt and navigate to directory containing pmca-console application.

3. Launch console in updatershell code. Use the filename appropriate for your version.
`pmca-console-v0.17-win.exe updatershell`

4. Press OK on the camera when prompted. Console will read:
`Waiting for camera to switch`

5. [Optional] Read original block setting in case you wish to change back
`bk r 01070026`
     You may notice we are reading the block noted in mungewell's post.

6. Write new block
`br w 01070026 02 00 00 00`

7. You can re-read the block as explained in step5, or just exit.
`exit`

I didn't notice a change in shutter sound. It definitely isn't silent. However, turning off chime stops the half-shutter beep which I found useful. So I reverted to the default setting.

albertlopes, 2019-10-24:
@nodeOfCode hi, i have a problem on step 4. When I press ok, my camera goes black and nothing happens on the shell. After some time appears Operation timed out. did you have this problem.
 a5100 firmware 3.10

alex-ong, 2020-04-15:
I didn't notice a change in shutter sound either, so i changed to to `02 00 00 00` and back to `03 00 00 00`

xVxM4tthewxVx, 2020-04-17:
Changing to `02 00 00 00` also appears to remove the auto focus beep. Is there a way to keep that while removing noisy shutter?

avelytchko, 2020-05-27:
I got error **Failed to execute script pmca-console**  with command `pmca-console-v0.17-win.exe updatershell` on A5100 with firmware 3.10, so I run `pmca-console-v0.17-win.exe updatershell -m ILCE-5100` and it works fine, no any fake shutter sound after changing 01070026 from 03 00 00 00 to 02 00 00 00.

edykim, 2021-11-19:
I tested it with my A5000 using the same address, and it is working perfectly without the fake sound. Now I understand why they decide to put the fake sound. :laughing:

islisis, 2023-09-08:
Just as a note as many report the fake shutter sound to not change, I tested it working on a5100 and in fact a NEX-3N as well - instructions often leave out syncing to disk using the "bk s" command, so it is worth checking whether it saves properly typing that command after writing and before exiting.

bk w 01070026 02 00 00 00
bk s
exit