ma1co/OpenMemories-Tweak #203: A99II Support

frank2266, 2017-08-04:
Hi guys,
please, add the support of the A99II.
That will be a tremendous step for the faithful A-Mount community.
PLEASE!
Cheers Frank

ma1co, 2017-10-30:
@frank2266 
There is a new experimental way to tweak settings on cameras without the Android subsystem using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on an A99M2, so your mileage may vary. If you attempt anything, please let us know!

closed 2017-10-30 by ma1co

frank2266, 2017-11-17:
I just applied the proposed procedure to unlock all the languages on my A99M2.
And... it worked flawlessly! Thank you very much.
You're my personal hero!