ma1co/OpenMemories-Tweak #205: Where to find Long Exposure Noise Reduction property in Backup.bin?

JulioCVM, 2017-08-13:


KillerInk, 2017-08-26:
Longexposure NoiseReduction works like a Darkframe. After the image has been taken, the shutter is closed for the same time. This serves to prevent the electronic noise.
I do not believe it is what you are looking for

Thats the method from BetterManual to disable it:

```
private void disableLENR()
    {
        // Disable long exposure noise reduction
        final Camera.Parameters params = m_camera.createEmptyParameters();
        final CameraEx.ParametersModifier paramsModifier = m_camera.createParametersModifier(m_camera.getNormalCamera().getParameters());
        final CameraEx.ParametersModifier modifier = m_camera.createParametersModifier(params);
        if (paramsModifier.isSupportedLongExposureNR())
            modifier.setLongExposureNR(false);
        m_camera.getNormalCamera().setParameters(params);
    }
```

stuomas, 2017-11-05:
@ma1co do you have any tips on how to begin searching this property? I tested with my A5100 to compare firmware dumps before and after a single on/off setting change (red eye comp) and the differences are like tens of lines. How to narrow it down? 

I thought if this works, the same could be done on A6000 for the LENR setting which is not hidden on that model, and tried to change same property on A5100.

ma1co, 2017-11-09:
Property 0x01070108 might be it.

stuomas, 2017-11-10:
@ma1co I took a leap of faith and tried it without trying to confirm first, how could you be wrong, right? The property existed in the backup.bin of my firmware disassembly with value 01, so it seemed probable enough.

**That is it!**

I set this property to 00 via telnet and my forced noise reduction was finally gone! I'm so excited. May I ask how you tracked down the correct property? I tried to disassemble the libObj.so as someone suggested and look for strings, and found many LONG_EXP_NR strings but nothing seemingly related to that property. Well I used radare2 and it's not very....user friendly (oh irony).

Probaly spending the weekend taking long exposure timelapses with that app when it's finally usable!!

BasicObject, 2017-11-24:
Can this be used on any other cameras? I'd like this to work for my A7S Mark I.

ma1co, 2017-11-25:
@BasicObject On the A7S, this option is available in the [menu](http://helpguide.sony.net/ilc/1420/v1/en/contents/TP0000226624.html).

farhi, 2019-03-20:
Many thanks for the great work you did. This is just wonderful. Tested on a a5100 to remove LENR. Installed the OM Tweak, then connected to local wiki and opened a telnet connection. Then a simple telnet to camera IP from any system (Linux/Mac) does the job, and command 'bk.elf w 01070108 00' works fully.
I think you can close that issue.

1crem1, 2023-12-03:
Worked for me. thanks very much - was at the end of my a5100 tether.