ma1co/OpenMemories-Tweak #206: Installed - But options grayed out?

graphicism, 2017-08-20:
I installed the app just fine via internet explorer on windows 10 to my Sony a6000. - No problem, the app installed and is visible in the applications list, but when I enter I am stuck on a grayed out info screen. I deleted the app and this time installed it via the cameras wifi. It also installed with a problem but is giving me the very same issue.

I have the latest 3.20 firmware installed.

Any ideas?

graphicism, 2017-08-20:
Of course 3-minutes after I register and post I figure out my several-hour long issue.

I just had to click up-twice on the D pad. Weird but it's working.

ma1co, 2017-08-29:
Yes, the initial focus is a bit off. I'll fix this in the next version.

closed 2017-08-29 by ma1co