ma1co/OpenMemories-Tweak #207: Error: Network connection

paraisosaudiovisuais, 2017-08-26:
Hi, 

I follow your instructions step by step, but I can´t install the app. I´m using IE. Here´s the log file:

Loading plugin

Plugin loaded

Starting task

Got task key

Cancelling plugin

Plugin cancelled

Starting download to camera

Error: Network connection

Starting task

Got task key

Cancelling plugin

Plugin cancelled

Starting download to camera

Error: Network connection

Any ideas?

Tks

ma1co, 2017-08-29:
Which camera? Which firmware version? Some of the newest firmwares are not compatible with the web installer, use [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases/latest) instead.

paraisosaudiovisuais, 2017-08-29:
Sony A6300. Latest firmware 2.0. I downloaded pcma-gui, but where do I download the apps?

ma1co, 2017-08-30:
@paraisosaudiovisuais Go to the "Install" tab, select the app and click "Install".

julienzgraggen, 2017-09-18:
For me it's always the same. Error: Network connection ???

ma1co, 2017-10-30:
@julienzgraggen Use pmca-gui.

closed 2017-10-30 by ma1co