ma1co/OpenMemories-Tweak #210: Sony RX100 Mark 5 Issue

Pippen78, 2017-08-30:
Installed on RX100 Mark V with firmware version 2.0 and kind of screwed up the system. It looks like now the 5 minute limit is actually applied to all resolution settings in 100M and 60M including the HD formats, not just the 4k ones. The HD settings using 50M record without a time limit. This problem also persists when the Tweak is uninstalled and the camera is fully reset. Any chance of an update that fixes this and also unlocks 4k for Mark v? The menu says that it is only available for Mark IV.

peppersass, 2017-09-01:
That's normal for the stock RX100M5 -- i.e., the 5-minute limit also applies to the 120p formats.

Pippen78, 2017-09-03:
It seems to be related to the bit rate. Basically anything above 50 has the 5 minute limit. 50 in whatever framerate does not. Not sure whether there is a discussion or whether this would be really difficult but can OpenMemories Twerk be updated for the 2.0 firmware and/or RX100 M5? It would be fantastic to open this capability. I also have an A6300 which works well with the Tweak (installed before the firmware was updated) but if you travel a lot the smaller the devices you need to carry the better)...

ma1co, 2018-10-26:
Duplicate of #130.

closed 2018-10-26 by ma1co