ma1co/OpenMemories-Tweak #214: A7RII Open Memories Tweak Black screen

Magnet8, 2017-09-30:
I installed the Open Memories Tweak App which worked great on my Sony A7RII but I have a big problem now. I plan to sell the camera so I removed the app again which worked well so far. Then I was resetting the camera in the menu. (not only a settings reset but a complete initialization) After this the camera did not turn on again. Its completely dead with a black screen (very little flickering on the screen is visible) No menu shows up, nothing. I did put the battery also out over night without success. I know it’s my risk but wondered if you can help me in any way to get the camera back to life. Thanks for any help!

Magnet8, 2017-09-30:
Is there a hard reset for the A7RII and if yes, how can I do it?

GeertRooY, 2017-10-01:
I hope you get a solution for your problem. I wanted to do the same. After your message, it seems smart for me to wait a while to avoid the same problems.

Magnet8, 2017-10-01:
Got the camera running over USB again but unable to install the Tweak again (Network error) Also when initializing the camera again the camera won't start up (black screen again) Only over the mentioned USB-connection the camera comes to life again. Any idea if the initializing should only be done when the tweak is still installed or any other hint?

Magnet8, 2017-10-01:
Although the tweak is installed all languages show up and the PAL/NTSC warning also doesn't show up. 

@ma1co Any idea why this happens and what I could do?

jinziqi, 2017-10-02:
did you try the suggestion here? https://github.com/ma1co/OpenMemories-Tweak/issues/179
basically use the pmca-gui instead of web interface

ma1co, 2017-10-08:
@Magnet8 First, sorry for the late answer, it's been a busy week. I'm sorry to hear about your camera. This should not happen for sure. How is the status now? Once you connect it over USB, does the camera work normally again? Do you have to connect it over USB everytime you turn it on, or just the first time?

A few things I want to mention:
1. Neither uninstalling the app nor factory resetting the camera will reset the tweaked settings. The only way to remove the tweaks is to disable them in the app.
2. The reason you cannot install the app anymore is probably because you updated the firmware in the meantime. Please use [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases/latest) to re-install the app.

To summarize, here is what I would try: Re-install the app using pmca-gui, disable all tweaks (don't forget to re-enable protection) and re-do the factory reset (this should also uninstall the app). Please keep us updated.

Magnet8, 2017-10-08:
Many thanks ma1co! Got the camera running again but it was tricky. Connected over USB and language only showed Chinese language. Luckily I got another A7RII so I was able to go through the menu. I followed your instructions and was able to do a reset to the factory settings. Also the pmca-gui app helped vs. the web-version. Here I had the connection errors. Many thanks, great app!

GeertRooY, 2017-10-11:
Thanks a lot ma1co for your information. I also downloaded the app for unlocking the languages. It works perfect.

closed 2017-10-30 by ma1co