ma1co/OpenMemories-Tweak #224: Feature: Streaming to other than uStream

mungewell, 2017-11-02:
Some of the Action Cams support streaming to the uStream service, via connecting to camera to a WiFi hotspot. It seems _many_ on the internet would like to be able to stream high quality images to difference services, or even their own PCs.

The streaming seems to be done via the 'ustream.apk' installed on the camera, for as100:
fw_as100\firmware.tar_unpacked\0700_part_image\dev\nflasha16_unpacked_unpacked\app\ustream.apk

Note: This is different from Liveview/Playback streaming (over Sony Remote APK), which is low quality/low frame rate.

Looks like the 'keys' have already been identified in the sister project:
(https://github.com/ma1co/Sony-PMCA-RE/blob/master/pmca/usb/sony.py#L196)

Is it possible to have a page to set these to point at other services (ie. YouTube)? Would it actually work?

mungewell, 2017-11-02:
Seems the 'normal' setup procedure is to log into uStream with username to get a camera-id/auth-token from uStream via the 'Network Settings Tool' (Sony's desktop app).
https://youtu.be/PWATV9sqTxA?t=333

I assume that these are these keys:
```
  ('macId', Struct.STR % 41),
  ('macSecret', Struct.STR % 41),
```

Resolutions offered appear to be 640x360 or 1280x720. The other strings shown in the video clip seem to make sense against the keys too. With 'LiveStreamingServiceInfo3' keys being related to 'recording' (saving) a copy of the video at uStream. Social media settings line up with "LiveStreamingSNSInfo".

The APK has a library 'libUstreamLib.so' filled with functions referencing 'RTMP' :-)

mungewell, 2017-11-12:
Added some code to read all the Streaming settings, dump them to file and write the 'adjusted' social settings back to the camera. WIP code here:
https://github.com/mungewell/Sony-PMCA-RE/tree/201701112-streaming

Need to think more how to write the other settings....

It seem that ustream keys (set by 'Network Settings Tool') are plain ascii strings, and that they are NOT able to be revoked (so be careful what you post!)
```
Service:            0
Enabled:            1
macId:              27aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa77
macSecret:          ceaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa0d
macIssueTime:       fc8f075a00000000 <--- binascii coded hex
shortURL:           http://ustre.am/1Aaaa
videoFormat:        3
enableRecordMode:   0
videoDescription:   Shot 100% with Sony's Action Cam #SonyActionCam #ProveYourself
videoTag:           
twitterEnabled:     0
twitterConsumerKey: o9aaaaaaaaaaaaaaaaaqg <--- I didn't set this!
twitterConsumerSecret: TTaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaasA
twitterAccessToken1: 
twitterAccessTokenSecret: 
twitterMessage:     Live Streaming from Action Cam by PONY
facebookEnabled:    0
facebookAccessToken: 
facebookMessage:    Live Streaming from Action Cam by PONY
```

mungewell, 2017-11-13:
The API also returns 'channel' and 'supportedFormats'. Don't know why these aren't stored within the 'info[1..3]' data blocks.

The channel number is a INT32 which correlates to the Ustream RTMP URL from website "Settings/Broadcast/Encoder", ie
`rtmp://1.23xxxxxx.fme.ustream.tv/ustreamVideo/23xxxxxx`

'videoFormat' is 1=640x360 or 3=1280x720, these are the only 'supportedFormats' that my camera issues (HDR-AS100).

'macId' and 'macSecret' do not match anything on the website settings. These seem to be issued as/with a separate scheme. Maybe there is some magic correlation of these to the channel key...

mungewell, 2017-11-13:
Another thing to note is that 'macId' and 'macSecret' are both 40 hex characters (limited to 0..9, a..f), whereas the uStream 'channel key' is 32 alpha-numeric characters (upper and lower case).

mungewell, 2017-11-13:
uStream documentation confirms that they are using OAuth2, with 'Hmac-sha1'. Fits with the observed 40 character id/secret.

'libUstreamLib.so' from the on camera APK contains hard-coded URLs:
```
ustreamVideo/%s
%d.fs_%s.1.%s.fme.ustre
%d.1.%s.fme.ustream.tv
```

mungewell, 2017-11-14:
Can now read/write the WiFi settings ('MultiWiFi' does not work with AS100, even though the camera supports it).
Also cleaned up the code and formated the output/input files to be more readable and easier to edit.
https://github.com/mungewell/Sony-PMCA-RE/tree/201701112-streaming

Novex, 2018-03-11:
I managed to convince my FDR-X3000 to send the rtmp stream to a custom server instead of ustream. From what I found it may not be possible for it to talk natively to rtmp servers without some translation in between.

It looks like sony/ustream implement some custom extensions to rtmp? Basically, the x3000 just sat there spinning after it established the connection with the custom server and sent the `publish` rtmp command. I was testing with nginx-rtmp-module, which has worked well for everything else I've used it for.

In the network dump I have with the x3000 talking to ustream, the publish command was followed by the ustream server sending back `broadcastStarted` and `recordStarted` rtmp messages, then the camera started sending what looks like data. I couldn't find any mention of these messages in the rtmp spec though.

I also say "what looks like data" because Wireshark couldn't seem to decode the audio/video packets sent to ustream (it shows them as `Unknown`). This also makes me feel like a normal rtmp implementation may not be able to understand them either. I'm happy to be wrong here though.

[Here are the pcap's for both sessions to ustream and to custom servers](https://github.com/ma1co/OpenMemories-Tweak/files/1800141/fdrx3000-rtmp-pcap.zip). I've deleted the ustream account that I used so hopefully there isn't anything sensitive in them (except maybe the wall that I recorded ;).

mungewell, 2018-05-16:
Looks like good progress has been made. Can anyone recomend a good way of building a local RMTP server for testing, for example a LiveCD distro that can be modified.

It's noted above that the APK uses a fixed URL to send the video. There are a couple of parameters in the uStream configuration, did anyone try adjusting these to see if they change the behavior?

```
    [
        "service",
        0
    ],
```
and
```
    [
        "unknown",
        1
    ],
```

[ustream_fake.cfg.txt](https://github.com/ma1co/OpenMemories-Tweak/files/2010407/ustream_fake.cfg.txt)

mungewell, 2018-05-18:
Mentioned above there is a hardcoded URL, digging into the 'libUstreamLib.so' file I found this:
`http://api.ustream.tv/users/self/channels.json?detail_level=broadcaster`

Attempting to 'wget' this from Ustream results in an auth error. I am assuming the contents of this needs to be spoofed to get us going, but what should it have?

In preparing to build my own RTMP server I rebuilt the 'libnginx-mod-rtmp' Ubuntu18.04 package with @transelement 'force-ping' patch included ([force-ping.zip](https://github.com/ma1co/OpenMemories-Tweak/files/2015293/force-ping.zip) ). This can be installed on a live-cd, virtual machine, or full system.

But how should you configure nginx? Any guidance would be appreciated...

Novex, 2018-05-19:
@transelement holy moly that's awesome! Thank you so much for the detailed write-up and sorry about the late reply (apparently I missed the gh notification then it was gone to the nether :( )

Yeah, [I hijacked dns and just had a python server returning my dummy `channels.json` data for](https://gist.github.com/Novex/cfa41eceb55b238d325afaab7bb271ba). @mungewell, those scrips might help you with what you need to return, eg. the `channels.json` response I used looks like this (I'm not sure if that channel id needs to actually match):

```
{  
   "channels":{  
      "23548279":{  
         "broadcast_urls":[  
            "rtmp://192.168.0.5/stream"
         ]
      }
   }
}
```

Also I don't remember fiddling with the `unknown` or `service` parameters in the config.

I'm still holding out hope to get a stream direct to any RTMP server, I was going to start investigating the [custom app launcher mentioned here](https://github.com/ma1co/Sony-PMCA-RE/issues/65#issuecomment-384427628) (which overwrites the ustream application, I'm not sure if that can be reversed).

I think you're possibly on the money with the changing publishing endpoints since I don't remember dealing with those - thank you for letting me know! I might take another crack at this now that you've confirmed it's possible :) 

If you've got any other info to share as well I'm all ears! I'll keep you posted if I find out anything else when I get a chance to take another look.

mungewell, 2018-05-19:
@Novex thanks for the info.

As you are just using a python script to MIM and fake up the initial web server, any reason why the camera itself could serve this file to itself?

BTW: overwriting the uSteam APK on that over bug is just as those camera have a limited UI and this just gives us a point of launching our code... it could be another RTMP stream I guess.

mungewell, 2018-05-21:
I was able to configure a Live-CD Ubuntu (Xubuntu really) with NetManager faking up an AP and nginx with the patched rtmp module. My AS100V camera will connect the AP and pull the 'channels.json' file, but then it does not seem to send any actual data. 

Nginx logs only show when the RTMP is stopped (ie, by pressing stop on camera). 

Configuration is included in this 'tar'. 
[settings.tar.gz](https://github.com/ma1co/OpenMemories-Tweak/files/2021263/settings.tar.gz)

Without the patched module it reconnects every 40s.
```
10.42.0.143 - - [21/May/2018:01:00:10 +0000] "GET /users/self/channels.json?detail_level=broadcaster HTTP/1.1" 200 135 "-" "Ustream-110-1/1.30dd916"
10.42.0.143 [21/May/2018:01:00:50 +0000] PUBLISH "live" "broadcaster/live3024" "" - 898 409 "http://www.ustream.tv" "LNX 9,0,124,0" (40s)
10.42.0.143 [21/May/2018:01:01:36 +0000] PUBLISH "live" "broadcaster/live9459" "" - 898 409 "http://www.ustream.tv" "LNX 9,0,124,0" (40s)
```

Is it just me, or my camera...?

Novex, 2018-05-22:
@mungewell I tried this tonight and I'm seeing similar results to you - my x3000 connects and has the spinner on screen infinitely (but no flashing red light). Wireshark doesn't show it sending any data other than the ping responses and the initial connection. nginx-rtmp-server shows the device as connected and i can see it publish to the `broadcaster/live1111` endpoint, but it doesn't show any I/O traffic and lists the state as `idle` instead of `active`.

@transelement did you change anything else with your config? The only thing I can think is I set my camera to livestream in HD if you perhaps left it as SD? If not, could I bother you for a .pcap of your working setup so I can try to trace where I'm going wrong?

mungewell, 2018-05-22:
@Novex thanks again for testing, seems that our two cameras are doing exactly the same :-(. Maybe @transelement has a different camera and/or setup.

Looking again at the code on the camera, I see that the ustream lib has more hardcoded URLs. The one with '%' suggests that the code has variables for these, and that URL matches the format of the link sent from UStream.
```
$ strings libUstreamLib.so  | grep "ustream.tv"
http://rgw.ustream.tv/gateway.php
http://api.ustream.tv/users/self/channels.json?detail_level=broadcaster
%d.fs_%s.1.%s.fme.ustream.tv
%d.1.%s.fme.ustream.tv
http://rgw.ustream.tv/gateway.php
http://cgw.ustream.tv/rgw
http://www.ustream.tv/mobile.swf
http://www.ustream.tv
```

I wondered whether the camera is double checking the 'channels.json' that it receives, and refusing to play (or just getting confused).

I copied the _official_ 'channels.json' file (pulled from your PCAP), and attached a sanatized version here ( [channels_full.json.txt](https://github.com/ma1co/OpenMemories-Tweak/files/2027675/channels_full.json.txt) ). This requires a few more DNS spoofs... Unfortunately playback still fails to start, but this might be helpful to others.

Can you confirm how you are configuring the camera for WiFi/Ustream? If not using pcma-console, can you use the 'wifi' and 'stream' commands to see if there's a setting we have missed.

Novex, 2018-05-23:
Yeah I have a feeling they might have a different version of the ustream.apk which behaves differently to ours.

For our version, I managed to get mine working! We need to respond to the initial `startPublish` and `callModule` messages from the camera with `broadcastStarted` and `recordStarted` messages respectively.

I've [hacked them into nginx-rtmp-module](https://github.com/Novex/nginx-rtmp-module/tree/sony-ustream-extensions) and they seem to work, though stopping the stream still hangs the camera. I suspect it's waiting on `recordReady` and `recordSaved` messages.

Also in case anyone was wondering, these are the video format details (which can be set with the [pmca-console](https://github.com/ma1co/Sony-PMCA-RE)):

| videoFormat | resolution | video bitrate | audio bitrate | fps |
| ------------- | --------- | ----- | ---- | ---- |
| 1                  | 640x360 | ~565 kbps | 128 kbps | 30 |
| 3                  | 1280x720 | ~2.7 mbps | ~140 kbps | 30 |

Novex, 2018-05-23:
@mungewell after all that I noticed the `enableRecordMode` field in the settings returned from `pmca-console stream` - is that set to 0 or 1 for you?

Mine is set to 1, and I have a sneaky feeling that maybe if you set it to 0 the camera might not wait for those messages and this might work with @transelement's build.

mungewell, 2018-05-23:
Grrr.... GitHub. How do you pull a patch/diff file from the WebUI (ie. without pulling the whole repo via git)????

mungewell, 2018-05-24:
Stepping aside for a question; which cameras that Tweak could support actually support the UStream live streaming.

I know we have:
FDR-X1000V / FDR-X3000 /HDR-AS50 / HDR-AS100V / HDR-AS200V / HDR-AS300 / HDR-AZ1 - http://support.d-imaging.sony.co.jp/www/ac_portal/features/networksettingtool/en.html

Any others? Can the "UStream.apk" be loaded onto others (a5000 for example)?

mungewell, 2018-05-24:
@Novex Your guess about 'record' was correct. When I disabled this I got RTMP video/audio packets, but I couldn't view them (probably me not knowing what I doing).
[working.zip](https://github.com/ma1co/OpenMemories-Tweak/files/2033641/working.zip)

I also built the rtmp module with both patches, and that worked with 'record' enabled.
[libnginx-mod-rtmp-patched2.zip](https://github.com/ma1co/OpenMemories-Tweak/files/2033640/libnginx-mod-rtmp-patched2.zip)

mungewell, 2018-05-24:
That diff trick is neat; I just pulled the repo and built the modified '.deb' package (above).

Were you using my 'nginx.conf' file? Maybe it's broken, Totally willing to admit I'm a Noob wrt nginx :-)

I was using 'rtmpdump' and 'vlc', both with 'rtmp://10.42.0.1/sony/live' - neither gave any data, VLC just showed the progress bar bouncing left/right. Wondering whether RTMP/video stream can be extracted from the PCAP (above).

mungewell, 2018-05-25:
I tried again to view stream with rtmpdump and vlc - still just sits with no data delivered.

I was able to decode something out of the pcap file.... but not video
```
simon@thevoid:~$ tcpdump tcp dst port 1935 -w out.pcap -r working_record_tcpdump.pcap 
reading from file working_record_tcpdump.pcap, link-type EN10MB (Ethernet)
simon@thevoid:~$ tcpflow -T out.rtmp -r out.pcap
simon@thevoid:~$ python3 ~/Downloads/rtmp2flv.py out.rtmp
[INFO] Reading from 'out.rtmp'
[DEBUG] Server uptime: 21d 19h 50m 10.419s, version: 80.83.141.53
[DEBUG] New chunk stream 3
[INFO] Stream 0 AMF0 command: ['connect', 1.0, {'app': 'ustreamVideo/12345678', 'flashVer': 'LNX 9,0,124,0', 'swfUrl': 'http://www.ustream.tv/mobile.swf', 'tcUrl': 'rtmp://111.1.12345678.fme.ustream.tv/ustreamVideo/12345678', 'capabilities': 1.0, 'audioCodecs': 1143.0, 'videoCodecs': 252.0, 'videoFunction': 1.0, 'pageUrl': 'http://www.ustream.tv', 'objectEncoding': 0.0, 'fpad': False}, {'deviceType': 'camcorder', 'phoneType': 'HDR-AS100V', 'deviceOS': 'Linux', 'appId': 110.0, 'appFlavor': 1.0, 'appVersion': 1.0, 'brandId': 1.0, 'buildNumber': '30dd916', 'rpin': '0.5142c1a531d9ba74bc85a5f20ff55c8ed4fa68cd', 'type': 'broadcaster', 'streamNamePrefix': '6498', 'macId': 'aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa', 'macNonce': '0:x14dautkfl', 'macSig': '9SmAEGLXn9KqyXBcF86hT97O8eA='}]
[DEBUG] New chunk stream 2
[WARNING] Missing timestamp, assuming 0
[WARNING] Missing stream id, assuming 1
[INFO] Set chunk size 1024
[INFO] Stream 0 AMF0 command: ['createStream', 2.0, None]
[ERROR] Error processing 'out.rtmp'
Traceback (most recent call last):
  File "/home/simon/Downloads/rtmp2flv.py", line 22, in main
    convert_file(fn, args)
  File "/home/simon/Downloads/rtmp2flv.py", line 59, in convert_file
    for m in read_rtmp_messages(f, args.chunksize):
  File "/home/simon/Downloads/rtmp2flv.py", line 159, in read_rtmp_messages
    eventtype, val = struct.unpack('>HI', data)
struct.error: unpack requires a buffer of 6 bytes
```

mungewell, 2018-06-12:
Ping. Anyone else having success with streaming to their own PC?

This might become a hotter topic, as uStream is stopping its 'free to use' accounts:
https://video.ibm.com/blog/streaming-video-news/enhanced-professional-plans-and-sunsetting-free-streaming/

wsop, 2018-06-13:
can i try with a7s ?

MakarkinPRO, 2018-08-21:
Is it any step by step to steaming to:
1) Any rtmp like youtube, facebook via WiFi
2) To my local nginx\obs server via WiFi
thx

kirashi3, 2018-08-22:
@mungewell and @Novex -- you guys are both doing GREAT work at reverse engineering the uStreaming functionality for Sony cameras, which is especially important now that uStream is owned by IBM and thus won't have a free or even paid consumer-facing tier. (They're really going all corporate...)

I'm tech savvy enough to unlock smartphone bootloaders and flash custom ROM's / pull logcats on Android phones, but get a similar error when trying to install the OpenMemories-Tweak on my HDR-AS50 ActionCam. That being said, I'm not entirely sure what the AS50 supports compared to other Sony Cameras.

I'd be more than happy to help test streaming and provide feedback / logs as you guys continue development on this though - just need a little pointing in the right direction as to what you need from me to test streaming from my camera, if you want the assistance, that is. 😃

Seb0PL, 2018-09-12:
I'll be happy to try this with **FDR-X1000V** :sunglasses: 

I would love to stream to [YouTobe](https://www.youtube.com/channel/UCFKURYra_hgoupgR_j_Qs6A) or [Facebook](https://www.facebook.com/Leczyzwierz/) directly :movie_camera: 

Is it in your scope to **provide user friendly guide on how to set this up**?
If you wish I might help you with testing :nerd_face:

LucaLanziani, 2018-11-03:
If you guys are still looking at this I've created a sort of "all in one" solution for this problem

https://github.com/LucaLanziani/sony-camera-stream-receiver

NOTE: still work in progress, it works for me ;)

Dehtiarov, 2018-12-29:
> 
> 
> If you guys are still looking at this I've created a sort of "all in one" solution for this problem
> 
> https://github.com/LucaLanziani/sony-camera-stream-receiver
> 
> NOTE: still work in progress, it works for me ;)

2018-12-30 00:25:54 10396 [INFO] Node Media Server v1.4.7
  dnsproxy:info we are up and listening at 192.168.1.147 on 53 +0ms
2018-12-30 00:25:54 10396 [INFO] Node Media Rtmp Server started on port: 1935
Server listening at http://0.0.0.0:80/
2018-12-30 00:25:54 10396 [INFO] There is a new version 1.4.9 that can be updated
2018-12-30 00:26:24 10396 [INFO] [rtmp connect] id=0LCOQ0AA ip=::ffff:192.168.1.62 app=23614434 args={"app":"23614434","flashVer":"LNX 9,0,124,0","swfUrl":"http://www.ustream.tv/mobile.swf","tcUrl":"rtmp://api.ustream.tv/23614434","capabilities":1,"audioCodecs":1143,"videoCodecs":252,"videoFunction":1,"pageUrl":"http://www.ustream.tv","objectEncoding":0,"fpad":false}
2018-12-30 00:26:24 10396 [INFO] [rtmp publish] New stream. id=0LCOQ0AA streamPath=/23614434/broadcaster/live7659 streamId=1
2018-12-30 00:26:24 10396 [ERROR] Unknown command { cmd: 'startPublish' }
2018-12-30 00:26:24 10396 [ERROR] Unknown command { cmd: 'callModule' }

LucaLanziani, 2018-12-30:
> @LucaLanziani I assume that you already had your camera set up with uStream... Did you confirm that a user can connect with 'fake' credentials?
> 
> Earlier in this bug I noted I had patched the tools to write streaming settings to camera (ie without using Sony's tools or needing a real uStream account).
> https://github.com/mungewell/Sony-PMCA-RE/tree/201701112-streaming
> 
> If so, I can rebase and try a pull request...

@mungewell yes, I've created a temporary account on uStream to run that. 
It would be awesome if it would be possible to change the stream endpoint so we can forget about the DNS spoofing part, let me know if I can help somehow on that.

LucaLanziani, 2018-12-30:
@Dehtiarov sorry but I'll need more details about the problem if you want me to address it, I've added the code to handle those commands here and I've got confirmation that it is working for someone else too. https://github.com/LucaLanziani/Node-Media-Server/commit/764920d88757be7faf8ca5236b148ef8538c0032

Dehtiarov, 2019-01-11:
[image]
Hi

> 
> 
> @Dehtiarov sorry but I'll need more details about the problem if you want me to address it, I've added the code to handle those commands here and I've got confirmation that it is working for someone else too. [LucaLanziani/Node-Media-Server@764920d](https://github.com/LucaLanziani/Node-Media-Server/commit/764920d88757be7faf8ca5236b148ef8538c0032)

[image]
Hi, in your Sony x3000 can you select in Action Cam Movie Creator properties USTREAM resolution higher than 1280*720 ?

Dehtiarov, 2019-01-11:
[image]

Hi, in your Sony x3000 can you select in Action Cam Movie Creator properties USTREAM resolution higher than 1280*720 ?

Memorystack16, 2019-01-24:
Hi, is there a way to get a guide with steps by steps? i'm just asking cause i got a bit lost in the guide posted at https://github.com/LucaLanziani/sony-camera-stream-receiver anyone got a video or more in depth guide?

thanks

dloprios97, 2019-02-02:
@LucaLanziani Hi man. My question is how do I turn my computer into a DNS server?

LucaLanziani, 2019-02-02:
@dloprios97 did you check this section of the readme? https://github.com/LucaLanziani/sony-camera-stream-receiver#dns-spoofing

dloprios97, 2019-02-02:
@LucaLanziani  yeah. But that is the only step? What about the second one `set your router to use your local computer as dns server`?
I've done everything but have no idea how to get the live number.
Sorry, I'm a newbie on this, but willing to test it out!

Novex, 2019-04-03:
I finally got around to taking a look at the ustream app - after a bit more assembly than I would have liked I've got a super buggy, basic, app that can work with the launcher to stream to (somewhat) arbitrary RTMP endpoints for 20 seconds at a time.

You can find it at https://github.com/Novex/pmca-livestream

That's the good news. The bad news is that the quality is [still shocking](https://gfycat.com/improbablethunderouschimneyswift) and I haven't found a way to improve it. Everything in the java code that looks like it will change things either has no effect or causes a crash (and I'm still pretty green reversing the lower-level binaries).

Mostly, I'm hoping this makes it easier for others to start experimenting so we can determine if it's possible to get usable video quality out of the camera, then put the time into making it stable and user friendly.

wesley9946, 2019-06-01:
Will it be possible to live stream from a HDR-CX450 to Youtube or something like an RTSP URL?

Remboooo, 2020-09-11:
You guys are geniuses, using the different elements in this thread I got my (new, never connected to uStream) FDR-AX53 streaming to Twitch over WiFi.
First order of business: getting the 'live stream' option to be enabled in the cam. The tools from Sony to log in to uStream don't work any more; the login dialog redirects you to IBMid login and from there it appears to work but doesn't log in. And the 'live stream' option in the cam stays disabled. And, of course Sony's support directed me to IBM support, and IBM support directed me back to Sony support (as I 100% expected). 

So, I used @mungewell 's modified pmca-console.py at  https://github.com/mungewell/Sony-PMCA-RE/tree/201701112-streaming to get some (dummy) credentials in the camera. My stream.cfg:
```
[
    [
        "twitterEnabled",
        0
    ],
    [
        "twitterConsumerKey",
        ""
    ],
    [
        "twitterConsumerSecret",
        ""
    ],
    [
        "twitterAccessToken1",
        ""
    ],
    [
        "twitterAccessTokenSecret",
        ""
    ],
    [
        "twitterMessage",
        "Live Streaming from Handycam by Sony"
    ],
    [
        "facebookEnabled",
        0
    ],
    [
        "facebookAccessToken",
        ""
    ],
    [
        "facebookMessage",
        "Live Streaming from Handycam by Sony"
    ],
    [
        "service",
        0
    ],
    [
        "enabled",
        1
    ],
    [
        "macId",
        "1234567890123456789012345678901234567890"
    ],
    [
        "macSecret",
        "123456789012345678901234567890123467890"
    ],
    [
        "macIssueTime",
		"a9315a5f00000000"
    ],
    [
        "unknown",
        1
    ],
    [
        "channels",
        [
            12345678
        ]
    ],
    [
        "shortURL",
        "http://ustre.am/1AAAA"
    ],
    [
        "videoFormat",
        3
    ],
    [
        "supportedFormats",
        [
            1,
            3
        ]
    ],
    [
        "enableRecordMode",
        0
    ],
    [
        "videoTitle",
        "Test"
    ],
    [
        "videoDescription",
        "Test"
    ],
    [
        "videoTag",
        "Test"
    ]
]
```
of course first install the python dependencies using `pip install -r requirements.txt`, then `python pmca-console.py stream -w stream.cfg`. This errored out 9/10 times but it worked once, which is all I needed. I'm not 100% sure but I think it's important that the `macIssueTime` is fairly recent; a timestamp from 3 years ago didn't seem to work. This is just a little-endian unix timestamp in hex format.

With that, the 'live streaming' option in the cam was enabled, but obviously it didn't work yet.
I got to work setting up dnsmasq on my network with an entry in `/etc/hosts` that pointed `api.ustream.tv` to my server. Now I could see in my nginx access logs that the camera was requesting `/users/self/channels.json?detail_level=broadcaster`. So I got to work setting up a webserver serving `api.ustream.tv` with a static `/users/self/channels.json`: `{"channels":{"12345678":{"broadcast_urls":["rtmp://api.ustream.tv/mystream"]}}}`
Adding the following nginx configuration to serve it:
```
server {
	listen 80;
	server_name api.ustream.tv;

	error_log /var/log/nginx/ustream-error.log;
	access_log /var/log/nginx/ustream-access.log;

	location / {
		root /var/www/ustream/;
	}
}
```
and placing the `channels.json` in `/var/www/ustream/users/self` (making sure user `www-data` can access it).

Next up is the RTMP part. First I tried the Debian package `libnginx-mod-rtmp`, but this only works for ~40 second intervals at a time before the camera reconnects. This is the behaviour described earlier in this thread and @transelement fixed with his mod-rtmp patch -- you are a genius and a scholar!
So I got to work de-installing libnginx-mod-rtmp and in stead compiling https://github.com/transelement/nginx-rtmp-module/commit/162163903da077553b378186cf7dba025d2e5c04 using the guide at https://gorails.com/blog/how-to-compile-dynamic-nginx-modules . I copied the resulting `ngx_rtmp_module.so` to `/etc/nginx/modules-available` (I'm sure that's not the proper place for it, but it's fine..)
Then I plopped in the following config in `/etc/nginx/modules-enabled` (NOT sites-enabled, it will complain about the `rtmp` directive not being allowed there, which is because all the files inside sites-enabled end up inside an `http` section in the config!):
```
load_module /etc/nginx/modules-available/ngx_rtmp_module.so;

rtmp {
	access_log /var/log/nginx/ustream-rtmp-access.log;

	server {
		listen 1935;
		chunk_size 65536;
		max_message 5M;
		ping 20s;
		ping_timeout 36000s;
		force_ping on;

		application remenmarleen {
			live on;
			meta off;

			push rtmp://live-ams.twitch.tv/app/live_...;

			recorder all {
				record all;
				record_path /var/rtmp;
				record_unique on;
			}
		}
	}
}
```
I'm not sure what combination of magic values gave me a glitch-free stream in the end, but this config works to get a perfectly stable stream!

sysadminpower2019, 2020-10-03:
Hello,

has there been a solution to this?

SirYogiWan, 2021-01-24:
So now I just need to find someone who knows what they're doing with coding, to tell me wtf I do with all this lol.

gaston908, 2021-01-24:
> transelement/nginx-rtmp-module@1621639

Congrats, wish there was a thread for noobs. Been trying for a year to get
mine streaming with no luck !

Memorystack16, 2021-01-24:
Is there any video on the different steps? Man im'a try to do it but this is way above my lvl of magical coding power

Palmeza, 2022-06-02:
Hello, this solution can be used with OSB Studio? Thank

chiburek, 2022-08-24:
Can someone help me to configure the access point for live streaming? I get the error "NO AP". Streaming doesn't seem to work on the Multi Wifi configuration and Sony's software is no longer available for download.

SirYogiWan, 2023-03-14:
> Can someone help me to configure the access point for live streaming? I get the error "NO AP". Streaming doesn't seem to work on the Multi Wifi configuration and Sony's software is no longer available for download.

if you use the wayback machine (site for archived pages), you can get the download by pasting the sony page address (the page that tells you it's no longer available to download). 

as for multi wifi, you can't stream on multiple camera's, to the one connection, if that's what you are referring to

wrynerson, 2023-03-25:
Has anyone figured out how to get the GoPro Max to Livestream in 360? I can livestream in hero to FB but not 360. I want to use it for VR.

dodolia907, 2024-09-20:
Thank you all for your deep knowledge. I packed your knowledge into a container and succeeded in streaming on our HDR-CX680. 
https://github.com/TechnoTUT/sony-camera-rtmp-relay
We ran this container on Kubernetes and streamed the 9-hour DJ event over our local network. docker and podman will probably work too.
We also achieved multicam streaming by using VLANs in our Cisco APs.

dimadesu, 2026-09-29:
Thanks everyone for you hard work reverse engineering this stuff!
I got it to work with Sony FDR-X3000.
I filmed a video of how I did it where I also explain in general how the hack works for less technical people.
I might have gotten a few things slightly wrong, but most of it is right.
Hopefully, more people can get live streaming to work on the Sony cameras with this information.
https://www.youtube.com/watch?v=e8huIP4TBB0

Memorystack16, 2026-09-29:
> Thanks everyone for you hard work reverse engineering this stuff! I got it to work with Sony FDR-X3000. I filmed a video of how I did it where I also explain in general how the hack works for less [...]

You sir are a legend i've bought and followed this tread for years getting occasionnal updates and wondering if ever something would work. This has been a long trip ! thank you

dimadesu, 2026-09-29:
@Memorystack16 Happy to hear that! This is definitely for enthusiasts.
I'm also very happy to get RTMP out of my old Sony cam. This was a dream of mine for a long time. So cool to see it working!