ma1co/OpenMemories-Tweak #225: UpsideDown mode  .  This mode is useful if you want to mount your camera upside-down.

giannisss, 2017-11-05:
Hi guys.I am wonder if it is easy to make for the screen of sony 6300 an upside down mode.I don t have thumb in my right hand and is is better if i rotate the camera to the left hand.This is a video i ade o explane my problem.Sorry for my bad english,if you see the video you will exacly understand...thank you!
https://www.youtube.com/watch?v=gRM-qcu-ZZM&t=76s

mungewell, 2017-11-17:
The 'flip' mode is supported on some action cams (ie. HDR-AS100V). I have pulled a 'Backup-bin' with/without the flip enabled, but need some help identifying appropriate keys.

mungewell, 2017-11-24:
So if you have telnet access the camera (developer tab of 'tweak' app), you could try the following keys. When the AS100 flip option is enabled all three are set to 1 (from 0).
```
id=0x01070cc3, size=0x0001, attr=0x40:
id=0x01070cc4, size=0x0001, attr=0x40:
id=0x01070cc5, size=0x0001, attr=0x40:

```
Unfortunately these don't work on my HX60, and I haven't figured out how to change keys manually on the AS100 to confirm that it's really these ones that cause the flip.

Commands would look something like:
```
$ telnet 192.168.1.101
Trying 192.168.1.101...
Connected to 192.168.1.101.
Escape character is '^]'.

BusyBox v1.13.4 (2012-07-17 15:11:04 JST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/ # /usr/bin/bk.elf w 0x01070cc3 1
id:01070cc3 size:1 data:01, 
/ # /usr/bin/bk.elf w 0x01070cc4 1
id:01070cc4 size:1 data:01, 
/ # /usr/bin/bk.elf w 0x01070cc5 1
id:01070cc5 size:1 data:01,
```

mungewell, 2017-11-24:
Figured out how to set keys on the AS100.. these key _DO_ control the flip. But the first 2 have to be set together, or the camera just flashes 'wait' continuously. When both are set to 1 the recorded video is rotated 180' (or probably mirrored on both axes which is the same).

The third key control a left/right audio flip, and can be set independently of the others.