ma1co/OpenMemories-Tweak #227: Mass storage error: Sense 0x2 0xff 0xff

yisding, 2017-11-07:
I'm trying to use pmca-gui with my Mac and NEX-6.

I get the error:

pmca.usb.MscException: Mass storage error: Sense 0x2 0xff 0xff

This is using the latest 0.15

ma1co, 2017-11-09:
Should be fixed with ma1co/Sony-PMCA-RE@a81e2ce7c9fb592e9b9be217722d0710ce809275. If you use the binaries, you'll have to wait for v0.16.

closed 2017-11-09 by ma1co

ma1co, 2017-11-28:
@yisding The v0.16 binaries have been released.

judy, 2017-12-13:
I'm getting the same issue. v0.16 gui, Mac OSX High Sierra, Sony a6000. I was able to replicate with the console command too, which lets me copy-and-paste.

```
Using drivers OS-X-MSC, libusb-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Traceback (most recent call last):
  File "pmca-console.py", line 91, in <module>
    main()
  File "pmca-console.py", line 61, in main
    infoCommand(args.driver)
  File "/nevermindmyprojectfolder/Sony-PMCA-RE/pmca/commands/usb.py", line 209, in infoCommand
    info = dev.getCameraInfo()
  File "/nevermindmyprojectfolder/Sony-PMCA-RE/pmca/usb/sony.py", line 251, in getCameraInfo
    data = BytesIO(self._sendCommand(self.SONY_CMD_DevInfoSender_GetModelInfo))
  File "/nevermindmyprojectfolder/Sony-PMCA-RE/pmca/usb/sony.py", line 244, in _sendCommand
    ) + data).ljust(writeBufferSize, b'\0'), readBufferSize)
  File "/nevermindmyprojectfolder/Sony-PMCA-RE/pmca/usb/sony.py", line 67, in sendSonyExtCommand
    self._checkResponse(response)
  File "/nevermindmyprojectfolder/Sony-PMCA-RE/pmca/usb/__init__.py", line 38, in _checkResponse
    raise UnknownMscException(msg)
pmca.usb.UnknownMscException: Mass storage error: Sense 0x2 0xff 0xff
```

judy, 2017-12-13:
Found a workaround for now. It seems to be fine when I switch the a6000's USB setting to MTP mode.

judy, 2017-12-13:
I tried to update the firmware via the gui. Unfortunately, that appears to kick the A6000 back into Mass Storage mode, and the same problem happens. Digging into the code base now (though I'm no Python programmer...)

judy, 2017-12-13:
One last comment before I'm AFK for the day. I stuck a debug statement after this line in osx.py:

```
res = iokit.IOConnectCallStructMethod(self.dev, kextSendMemoryBlock, byref(desc), descSize, byref(output), byref(outputSize))
```

The code I'm getting back is `3758097084`, or `0xE00002BC` in hex. Looks like a `kIOReturnError`? Don't know much about it.

judy, 2017-12-14:
Alright, back at it again today. 😄 Last night, tried specifying `libusb` driver as a workaround. No luck, it told me `usb.core.USBError: [Errno 13] Access denied (insufficient permissions)`. Maybe that's a new issue, but I don't want that to derail this one.

judy, 2017-12-14:
One more thing I noticed. When it fails (back on the native osx driver), it kicks all of my other USB _and_ Bluetooth devices off. Both the camera and a usb drive are plugged in currently, and I get a notification that they were disconnected. They reconnect pretty quickly though. I wonder if some subsystem is being rebooted?

melmdb, 2018-10-07:
I have the same problem with my Sony RX100 V

JMY1000, 2018-10-19:
Switching the USB LUN Setting (the toolbox -> 4) solved the issue for me on an a7R.

MuffintopSpeedos, 2019-12-18:
> Switching the USB LUN Setting (the toolbox -> 4) solved the issue for me on an a7R.

Switched it to what?

JMY1000, 2019-12-18:
@MuffintopSpeedos Can’t quite remember; I think there were only two options? If you’re still running into trouble send me a reminder and I can check in a few hours.