ma1co/OpenMemories-Tweak #229: Setting keys in 'Backup.bin' on Action Cams

mungewell, 2017-11-24:
The 'tweak' app is nice, beautify even. Once run you can enable WiFi and telnet, then telnet onto the camera to inspect and change individual keys using the 'bk.elf' command.

However it can't be run unless you have a decent screen and some buttons to control it, so it's not much use on an action cam or a QX10 :-( You can use the 'updatershell' to get a shell on the camera, but unfortunately this doesn't have 'bk.elf' installed.

The attached script can convert (a previous snagged) copy of 'bk.elf' into bash echo commands, which when run will (magically) re-create the file. Have the resultant output in a clipboard buffer and...
```
$ sudo python3 ~/Sony-PMCA-RE-git/pmca-console.py updatershell
No native drivers available
Using drivers libusb-MSC, libusb-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Getting device info
Using firmware for model HDR-AS15

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...

Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>info
Model:              HDR-AS15
Product code:       2412130
Serial number:      1165240
Backup region:      CX40600_UC2
Firmware version:   3.00
>shell
sh: can't access tty; job control turned off

BusyBox v1.13.4 (2010-07-15 10:39:07 JST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/ # cd /ramdsk
```
Now paste the output of the script in the terminal and
```
/ramdsk # ./bk2.elf r 0x003c0373
id:003c0373 size:1 data:
0D,
/ramdsk # ./bk2.elf r 0x003c0374
id:003c0374 size:1 data:
01,
/ramdsk # ./bk2.elf r 0x003c0375
id:003c0375 size:1 data:
00,
/ramdsk # ./bk2.elf w 0x003c0373 0
id:003c0373 size:1 data:00, 
/ramdsk # ./bk2.elf w 0x003c0374 0
id:003c0374 size:1 data:00, 
/ramdsk # ./bk2.elf w 0x003c0375 1e
id:003c0375 size:1 data:1E, 
/ramdsk # ./bk2.elf r 0x003c0373
id:003c0373 size:1 data:
00,
/ramdsk # ./bk2.elf r 0x003c0374
id:003c0374 size:1 data:
00,
/ramdsk # ./bk2.elf r 0x003c0375
id:003c0375 size:1 data:
1E,
/ramdsk #
```
Setting these particular keys limit the record time for each video to 30s, so you probably don't want to do that - but it show that the 'Backup.bin' can be altered on a per key basis.

[bk2echo.py.txt](https://github.com/ma1co/OpenMemories-Tweak/files/1500797/bk2echo.py.txt)

ma1co, 2017-11-25:
I agree, I should integrate this functionality directly in updatershell.

mungewell, 2017-11-25:
Regarding 'bk.elf', the AS15 and AS100 have different versions (presumably as they're different gen cameras). They are not cross compatible. Also as '/setting/' is mounted rw, it's easier to write 'bk2.elf' there and it's persistent across multiple sessions.

mungewell, 2017-11-25:
Retract that last comment.

DO NOT write 'bk2.elf' into the '/setting' directory, this might have been related to a problem I had where my camera would not factory reset, and was giving weird power on/off behaviour. I think this may been as it ('bk2.elf') had occupied too much space in that partition.

I was able (finally) to get back in, delete it and then be able to factory reset... not 100% certain that this was the cause. But playing it safe...

ma1co, 2017-11-28:
@mungewell You can now use the `bk r` and `bk w` commands directly in updatershell. It's pretty basic, but it should be enough for now...

closed 2017-11-28 by ma1co