ma1co/OpenMemories-Tweak #230: USB Mic Update for  Sony RX 100 IV / V

Zoidbioz, 2017-11-24:
Hey there,
I was wondering if there is a way to make an external mic run with the USB input for one of those cams.
I realy like  them but was hoping for better audio.
If you can help me this would be great.

Greetings from Germany

zoidbioz

mungewell, 2018-01-30:
Not an answer, but it might be interesting to look how the cameras with intelligent hot-shoe detect and activate mics connected there...

mikeytown2, 2018-01-30:
In theory it should be possible https://source.android.com/devices/audio/usb

palomojavi, 2018-11-30:
I'm new to this forum, but I purchased a Sony RX100 V that is shipping from japan and its arriving tomorrow. 
I want to install the OpenMemories-Tweak to unlock the english language. 
This USB mic update would be great as well. I don't understand the flag "enhancement". Does this mean that the new version has this option? 
Any suggestions for a good USB mic for this camera?

mungewell, 2019-09-23:
Thread digging quite a bit...
Did anyone investigate this further?