ma1co/OpenMemories-Tweak #233: Initializing A7RII to factory settings

Magnet8, 2017-12-03:
Every time I am initializing the A7RII to factory settings with or without the installed Open Memories App the screen stays completely black after turning it on. Need to connect the camera over USB and navigate through Chinese menu. Any idea how I can come back to the very basic settings where you can dial in country / date time aso?

Magnet8, 2017-12-03:
I am kind of desperate in the moment...This problem comes up only when initializing, when I only reset the camera settings its OK.

Magnet8, 2017-12-03:
I would like to see the camera but if the new owner initializes there will be a huge problem with the black screen. What can I do?

ma1co, 2017-12-03:
Uninstalling the app is not enough to clear the settings. You have to disable them manually.

Try this:
- Re-install OpenMemories-Tweak
- Set all tweak settings back to the default value ("Disable video recording limit": off, "Unlock languages": off, "Unlock protected settings": off)
- Initialize the camera

Magnet8, 2017-12-04:
Ah, many thanks this helps!

rxs0, 2017-12-28:
I had the same problem with the Sony A7RII firmware 4.00.  Reinstalling the OpenMemories-Tweak worked for me!  Thanks ma1co!

rxs0, 2017-12-28:
I had the same problem with my Grey Market A7RII firmware version 4.00 updated over OpenMemories-Tweak v0.10.  I reinstalled the openmemories tweak and this solved the problem.  Thanks ma1co!

closed 2018-10-26 by ma1co