ma1co/OpenMemories-Tweak #234: would the a7r3 be supported?

muttong, 2017-12-04:
just curious, if a7r3 will be supported then I will trade my r2 for an r3. but if it is hard(like the a9 being impossible) then I don't see the point of having a sony camera without open memories tweak.

recording time limit is just a b*tch

ma1co, 2018-01-17:
The A9 and the A7RM3 use a new firmware format and we don't know the encryption keys. So currently we have no way inside those cameras. I am not aware of a way to dump those keys in the near future.

closed 2018-01-17 by ma1co

BruceKilmister, 2018-04-14:
i just bought a a7iii from japan, so that means i have to use japanese for the life? what a shame.

solomatovroman, 2018-09-07:
Please advise if there is any progress

FCS7iii, 2018-11-24:
Any update on that ? No chance to have the a7iii supported ?