ma1co/OpenMemories-Tweak #236: Dec 2017 Sony a6000

knowledgeiseverything, 2017-12-09:
Hello I have an issue using pmca-gui v.016
Camera is connected via USB (MTP Active) - USB connected...
Here is the issue:
-----------------
Using drivers OS-X-MSC, libusb-MTP
Looking for Sony devices
No devices found. Ensure your camera is connected. 
-----------------
MacOs Sierra 10.12.6

judy, 2017-12-13:
You have to install the driver from Sony. Get the "PlayMemories Camera Apps Downloader" here:

https://sony-pmca.appspot.com/plugin/install

knowledgeiseverything, 2017-12-20:
Hi Judy! Thanks for your reply! I tried. Installed...now it sees the camera as memory mass but as soon as it connects bluetooth switches off...leaving me with out keyboard or trackpad (I'm using a mac) - I try to connect a USB mouse...nothing it doesn't read. Looks like it switched off all the peripheries USB/BLUET. etc. Don't know if the camera has something wrong at this point.. ?

judy, 2017-12-21:
Whelp, now you're having the issue I was having over in [Issue 227](https://github.com/ma1co/OpenMemories-Tweak/issues/227). Hopefully your ticket provides some visibility to this issue. I'm not sure if @ma1co sees the other issue since it's been closed.