ma1co/OpenMemories-Tweak #238: Reverse Engineering

mturkdfeldt91, 2017-12-12:
I've started to reverse engineer the settings. I have no experience whatsoever but work with rooting Androids and messing with ROMs so I am looking to give this a go.

I have the a6000 and found some entries in this file that seem to show it is possible to have 4K and 150 MBPS..

I'm working on setting up a way to put various scripts on the SD card to be ran at start up automatically. 

Currently jobless and waiting for school to start so I will be giving this my all. Let me know if you have something to add or want to help.

[image]

williamkapke, 2017-12-13:
> I'm working on setting up a way to put various scripts on the SD card to be ran at start up automatically.

... yes please! 🎉  That'd be really useful

mungewell, 2017-12-14:
Welcome, hacking these cameras is fun (and addictive). I have also seen those strings (libObj.so right?), both in AS100 and A5000 firmware.

I am somewhat stumbling around trying to understand the connection(s) between the Android apps, and how they 'talk' through frameworks, to Linux libs, to 'Backup.bin'. This may help us (automagically) identify appropriate keys and experiment how the features of the cameras can be enhanced.

I assume that you have tweak to get a live telnet session, 'bk.elf' can read and write keys in real time for experimenting.

I identified some of the keys, documented here:
https://github.com/ma1co/fwtool.py/files/1503349/video_settings.zip

mturkdfeldt91, 2017-12-14:
Yes, those strings are from libObj.so on the a6000. 

At three this morning, I was experimenting with sdmon.elf. Are you familiar with it? It seems to allow you to register and set parameters that are available to Scalar Daemon, which is the Android app side of things I believe. Running it through Telnet allows you to change settings and what is available per setting, such as the levels of ISO sensitivity or DRO levels, etc. Although I am unsure where such changes would take effect, as I did not notice the changes in any of the apps.

I put in about an hour early this morning in a half sleep state so I will get back to it and look into it some more.

mungewell, 2017-12-15:
Funnily enough I was looking a 'sdmon.elf' last night too, on my new a5000 (open box bargain!). Didn't make much sense to me...

You can see the changes with 'bk.elf' appear immediately on the OSD display or in menus, for example:
```
Codec (confirmed HX60)
id=0x01070087, size=0x0001, attr=0x68:
3 = AVCHD
4 = MP4
a = XAVC S
```
For example by typing the following to change to AVCHD
`bk.elf w 0x01070087 3`

The menus seem to be limited to what the camera officially supports, maybe this is stored in another key. For example setting 'XAVC S' or setting '60p' make the menu setting appear blank (at least on HX60 or A5000).

mikeytown2, 2017-12-15:
[USB audio](https://source.android.com/devices/audio/usb) would be a huge win for these cameras.

mturkdfeldt91, 2017-12-16:
I've got a little side project trying to simplify the on screen display. The only issue is I can't get the file to transfer back to the camera no matter how I try. Any suggestions? 

I've modified the framework-res.apk file. Need to get it back to /android/system/framework/ but everything is read only.

alreadysabbir, 2017-12-17:
@mturkdfeldt91 try https://android.stackexchange.com/questions/110927/how-to-mount-system-rewritable-or-read-only-rw-ro

mturkdfeldt91, 2017-12-18:
Thanks @alreadysabbir 
Using that link as a starting point, I was able to find the correct answer online. The solution provided at the link did not work. What ended up working was the following command through Telnet which carries over to ADB:

# mount -t vfat /dev/dnflasha3 /system -o remount,rw

mungewell, 2017-12-20:
Interesting, although somewhat scary if you make the screen unusable.... how would you re-init the camera?

Seen mention of people wanting "Clean HDMI" outputs, is that in-scope of what you're doing? Maybe as one of the 'disp' modes...

mturkdfeldt91, 2017-12-20:
Exactly what I am trying to do, remove all the on screen information as a mode.

Anyhow, I've got another issue somehow related to the SD card now. Without the SD card installed I am able to use the camera normally, taking one photo, view it, change settings, etc. When I insert an SD card, it no longer wants to cooperate, after viewing a single photo it will have a black screen and the camera does not respond to anything. I'd like to totally reset the firmware, but Sony didn't think things through and made it so you can't reapply a firware if it is the same one. 

How do I get a new Firmware image on this camera to start completely over?

Also, 1070a03 is a new OSAL ID I found if you want to add it to your list. It tells the camera if it has one or two SD card slots, 00 being one, 01 being two SD cards. It adds slot settings to Format and the media info settings. Also will add on screen sdcard image with a 1 in it.

mturkdfeldt91, 2017-12-22:
Any idea how to read .UXC files? I've tried many ways and googled with no results.

motioncamera, 2018-02-21:
If you run into the audio settings or any key as to where I can activate the XLR-K2M through the multi-interface shoe, please post.  I am convinced that it is a setting that has been disabled since other audio devices will work with the multi-interface shoe. Let me know what if anything I might be able to do to contribute and help this process. Thanks

mungewell, 2018-02-21:
The XLR-K2M looks nice, albeit rather expensive. Sony's page says it's only compatible with the RX10, RX10M2, RX10M3 and RX10M4.
http://support.d-imaging.sony.co.jp/www/cscs/accessories/compatibility.php?area=ca&lang=en&mdl=XLR-K2M&cat=1&loc=en_CA

My wife has the HX60 with MultiShoe, which I believe supports external MICs. Might take a trip down the to camera store...

motioncamera, 2018-02-21:
It is compatible with the a6300, a6500, a7_ series etc.  As well as those you mentioned. Just not enabled on the a600 and was only recently enabled onthe a6300.  Seems stupid to me but then Sony do what they want to do I guess.  I have three cameras, a6500, and two a6000  that I want it to work on.
As I said if there is anything I might do, willing to learn just let me know.

crirus, 2018-03-17:
Any chance to set OSD over HDMI so I know what I am doing by looking at a external monitor while recording 4k?
Apparently they cut out all that info when outputing 4k.

thegallivanter, 2018-06-18:
mturkdfeldt91, do you know if anyone ever tried enabling the video format of the lowest bitrate 4K setting possible on a camera that does not officially support it? (say 4k 24fps 60mbps xavcs on a a7s or a6300) A little like HD video is enabled on older bodies that did not support HD. Or is it not as simple as changing the setting in the code and seeing if it works? (I`m new to this repo) Similar to [this ](https://github.com/ma1co/OpenMemories-Tweak/issues/23) and [this](https://github.com/ma1co/OpenMemories-Tweak/issues/217) issue I guess.

mwlczk, 2019-07-24:
> I have a7ii. And I am wondering is it any success of enabling 4k in a7ii? Looks like it has the same processor as a7sii. So I guess there is a big change to enable it. I am even ready to contribute some donation for the project if it is exist...

secondig this. 
@ma1co: but what would be the way to find out where to look for the "hidden" switch?
Can I compare an a7III dump to a7II in that matter?

zbgn, 2021-02-15:
Hello, have you made any progress on this matter so far ? 

Is there anything I can do to help on the subject of 4K and a7II ?

I'm linking other issues there:
#375 #360

Pegacy, 2022-06-05:
> Hello, have you made any progress on this matter so far ?
> 
> Is there anything I can do to help on the subject of 4K and a7II ?
> 
> I'm linking other issues there: #375 #360

I don't know if this information will help. But a7m2 allows display pictures in 4k in playback mode on hdmi, as well as a very interesting behavior at a ratio of 3: 2 in manual (photo) mode, when zooming in, you can see characteric of 4k rolling shutter.

ma3rian, 2024-08-04:
I'm currentrly working on my ILCE7M3 and for now i managed to unpack the firmware file (4.01)
my purpose is to find the code that is assigned for the limited recording and  "Touch operation" setting in the camera in order to manipulate it (if possible) so that can be globally enabled.
Currently i'm using BinaryNinja tool in windows10, but i wonder if it's  better on Linux?

any help from you LADS is blessed