ma1co/OpenMemories-Tweak #239: Open UDP socket

Dev1an, 2017-12-16:
Does anyone know if it is possible to open a UDP socket on these cameras? I would like to make a remote follow focus controller and I was thinking to communicate the focus position over UDP.
Does anyone know if this is possible or an alternative way to send focus instructions to the camera from a remote device?

alreadysabbir, 2017-12-17:
@Dev1an you can use playmemories app for that.

Or look at their https://developer.sony.com/develop/cameras/ if you need rest api.
It's already got a udp socket for SSDP and a TCP socket for rest api. 
You have to connect it  via wifif with remote app open on the camera.
Example ssdp discovery script that actually works https://gist.github.com/alreadysabbir/37570f1fb63d3249c5e4532f05ff7006

Dev1an, 2017-12-18:
Hey @alreadysabbir, thanks for your quick response!
I checked the playmemories app and the SDK but I could not find any way to manually set the focus distance using a remote device (am I overlooking something?).
It is possible to autofocus on a specific (x,y) point on the image, but this is not what I am looking for though.

I am trying to make a device like depicted below where you have a remote focus wheel. It would work just like you do manual focus (using the wheel on the lens), but now from a remote device.
[image]
Have a look at this video if it's still not clear: https://www.youtube.com/watch?v=UhE0ArjzJdI

mungewell, 2017-12-20:
Interesting project.

I don't think you can use the Sony Remote API to set a specific focus distance, only to set a focus position in the frame or enable tracking focus. It might be worth poking around in their (on camera) APK to see how they access the focus behaviours of the camera, you can use the Remote API to set MF... maybe something is hidden in there.

BTW:
https://github.com/Bloodevil/sony_camera_api

mungewell, 2017-12-20:
Looking at the 'Smart Remote' app it seems that you can't set a particular value, just focus in/out normally driven by keys/levels on camera. But if you've an app running perhaps you can adjust and read back current value to 'home in' on desired value.

From "com/sony/imaging/app/base/shooting/trigger/FocusAdjustmentKeyHandler.java"
```
    public int turnedMainDialNext() {
        FocusMagnificationController controller = FocusMagnificationController.getInstance();
        controller.moveFocusDriveFar();
        controller.rescheduleTimeout();
        return 1;
    }
```

From "com/sony/imaging/app/base/shooting/camera/CameraSetting.java"
```
    public void moveFocusDrive(int direction) {
        moveFocusDrive(direction, FOCUS_DRIVE_SPEED);
    }

    public void moveFocusDrive(int direction, int speed) {
        if (Environment.isAvailableFocusDrive()) {
            this.mCameraEx.startOneShotFocusDrive(direction, speed);
        }
    }

    public int getFocusCurrentPosition() {
        return this.mFocusCurrentPosition;
    }

    public int getFocusMaxPosition() {
        return this.mFocusMaxPosition;
    }
```

Dev1an, 2017-12-20:
Hey @mungewell, thanks for all the info!
I indeed already saw this `startOneShotFocusDrive` API being used in the [focus bracket](https://sony-pmca.appspot.com/plugin/app/com.obsidium.focusbracket) application to focus to a specific position.

But I was not able to find someone who made an app with a TCP/UDP server, hence my question...

mungewell, 2017-12-20:
The Sony "Smart Remote" manages the WiFi as a WiFi-Direct link and brings up it's own Orb (http) server.

alreadysabbir, 2018-01-04:
@mungewell exactly.
It has an SSDP protocol for rest API server discovery. I was able to connect my mac via wifi (Smart remote app) and call the API.
SSDP example: https://gist.github.com/alreadysabbir/37570f1fb63d3249c5e4532f05ff7006