ma1co/OpenMemories-Tweak #242: Development environment help?

abena13, 2017-12-20:
I'm interested in helping with the project and have an idea for it, but struggling to get started.

I have Android Studio and cloned recursively this repository. Opened it with the Gradle option. But I am getting errors (not warnings) about deprecated ABIs (armeabi) and lots of errors about ERANGE and EINVAL not being declared in this scope (from the JNI C++ parts).

Any tips how to get the development environment set up more painlessly?
I had to change Gradle version from 3.0.0 to 3.0.1 and arm-linux-gnueabi-g++ to arm-linux-gbueabihf-g++ because that's how they were on my system, but other settings I haven't touched. Using Linux.

ma1co, 2018-01-17:
@abena13 Here is what I have installed on my machine:

* The newest version of Android Studio
* [Android NDK 14b](https://developer.android.com/ndk/downloads/older_releases.html) (this is the last version which supports the android-9 platform). This is used for the JNI and adbd binaries.

ma1co, 2018-01-17:
@abena13 Here is what I have installed on my machine:

* The newest version of Android Studio
* [Android NDK 14b](https://developer.android.com/ndk/downloads/older_releases.html) (this is the last version which supports the android-9 platform). This is used for the JNI and adbd binaries.
* [arm-linux-gnueabi](https://releases.linaro.org/components/toolchain/binaries/4.9-2017.01/arm-linux-gnueabi/) for the protectiontweak binary (which runs outside the android environment). This binary is only used for the protection tweak on Japanese cameras and you can probably get around installing an additional toolchain by commenting out the `runMake` task in build.gradle

Please let me know if you need more help.

closed 2018-10-29 by ma1co